framps/raspiBackup #224: Option für Fortschrittsanzeige bei dd Backup und manuellen Aufruf

Hofei90, 2020-05-08:
Hallo,
habe den Vorschlag, dass man eine Fortschrittsanzeige bei einem manuellen Aufruf von raspiBackup.sh anzeigen könnte, wenn der Backupmodus auf dd steht.
Das ganze könnte man über einen Parameter ermöglichen wie z.B:
`./raspiBackup.sh --show_dd_progress`
Aber vielleicht fällt dir auch was besseres ein.

Wie man die Fortschrittsanzeige selbst macht, finde ich ist hier gut beschrieben:
https://wiki.ubuntuusers.de/dd/#Fortschrittsanzeige

framps, 2020-05-08:
Es gibt die Option -g die bei dd, tar und rsync eine Fortschirttsanzeige einschaltet. Hast Du die Option benutzt?

Hofei90, 2020-05-08:
Feature schon vorhanden, nur konnte ich es nicht anschauen

closed 2020-05-08 by Hofei90