framps/raspiBackup #226: Two identical systems, rsync works on one, but not on the other, forces to dd

rjsears, 2020-05-10:
Version: 0.6.5 CommitSHA: 7752af8 CommitDate: 2020-05-07 CommitTime: 14:18:31

PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Hello - 

I have two identical Raspberry Pi systems with the exception that one is using a 64GB SD card and one is using a 128GB SD card. They both have the same identical mounts and only two partitions (/ and /boot). Both are running the identical version of Raspian.

One works perfectly with raspiBackup and rsync while the other keeps giving me an error saying that there are more than two partitions and dd must be used. 

The only difference that I can see is on the one that is saying that it needs to use DD, the filesystems are identified by their specific devices:

`
/dev/mmcblk0p6  /boot           vfat    defaults          0       2
/dev/mmcblk0p7  /               ext4    defaults,noatime  0       1
`

While on the system that works correctly they are identified by UUIDs:

`
PARTUUID=5e3da3da-01  /boot           vfat    defaults          0       2
PARTUUID=5e3da3da-02  /               ext4    defaults,noatime  0       1
`

I know I can override with -P but I think there is some other reason that this error is happening on two identical systems. I would *much* rather use rsync than having to DD each time a backup is done.

Thank You

framps, 2020-05-10:
Would you please provide the output of following command
```
sudo lsblk
```
executed on both systems?

rjsears, 2020-05-10:
Thank you for pointing me in the right direction, it would appear that NOOBS creates a ton of extra partitions that you do not see from within the OS but you can when you run lsblk. I must have manually installed the first one and then used NOOBS on the second one.

Sorry to have taken your time.

framps, 2020-05-10:
No problem. Please close the issue if you're fine now.

rjsears, 2020-05-10:
Solved and Closed!

closed 2020-05-10 by rjsears