framps/raspiBackup #227: RBI0027E: Erforderliches Verzeichnis /usr/local/etc existiert nicht

bismosa, 2020-05-10:
Hello,

i tried to install with:
` curl -sSLO https://www.linux-tips-and-tricks.de/raspiBackupInstallUI.sh && sudo bash ./raspiBackupInstallUI.sh`

I get this error:
```
20200510-080955: DBG: 3132 -  raspiBackupInstallUI.sh 0.4.3.2, 2020-05-06/19:59:23 - 750042a
20200510-080955: DBG: 3133 -  whiptail version: whiptail (newt): 0.52.19
20200510-080955: DBG: 0615 -  Checking for /usr/local/bin
20200510-080955: DBG: 0615 -  Checking for /usr/local/etc
20200510-080955: DBG: 1375 -  -> unrecoverableError 27
/usr/local/etc: DBG: 0000 -

  File "./raspiBackupInstallUI.sh", line 3135, in main
    checkRequiredDirectories
  File "./raspiBackupInstallUI.sh", line 617, in checkRequiredDirectories
                        unrecoverableError $MSG_MISSING_DIRECTORY "$d"
  File "./raspiBackupInstallUI.sh", line 1386, in unrecoverableError
        logStack
20200510-080955: DBG: 1393 -     Id: 27
20200510-080955: DBG: 1394 -     Msg: /usr/local/etc
20200510-080955: DBG: 1403 -     RBI0027E: Erforderliches Verzeichnis /usr/local/etc existiert nicht.

Ein nicht behebbarer Fehler ist aufgetreten. Siehe Logdatei raspiBackupInstallUI.log.
20200510-081133: DBG: 1407 -  <- unrecoverableError
20200510-081133: DBG: 1511 -  -> cleanup
20200510-081133: DBG: 1545 -  <- cleanup
```

Version:
```
./raspiBackupInstallUI.sh --version
./raspiBackupInstallUI.sh: Ungültige Option -- -
raspiBackupInstallUI.sh 0.4.3.2, 2020-05-06/19:59:23 - 750042a
raspiBackupInstallUI.sh -i [ -e ]? | -u
-i: unattended install of raspiBackup
-u: unattended uninstall of raspiBackup
-e: unattended install of raspiBackup extensions
```
 ```
cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```
Im always logged in as root on this Pi. 

Should i only create this directory?

Regards
Bismosa

framps, 2020-05-10:
> Should i only create this directory?

Yes. Please make sure it has following attributes:
```
ls -la /usr/local/etc -d
drwxr-xr-x 1 root root 32 May  8 11:38 /usr/local/etc
```
Frankly I have no idea why this directory is missing on your system. That's a standard directory on Stretch.

bismosa, 2020-05-13:
Hello!
Many thanks. I have created the directory and set the permissions. Installation successfully.

Regards
Bismosa

closed 2020-05-13 by bismosa