framps/raspiBackup #228: Problem with mail colouring

WAdama, 2020-05-10:
Version: 0.6.5 CommitSHA: 750042a CommitDate: 2020-05-06 CommitTime: 19:59:23

Mail colouring seems not work correct.

Content type is shown in subject: "Backup erfolgreich beendet. Content-Type: text/html"
The html tags are then shown in mail body.

I'm using postfix as mail program. The mail command must contain a parameter: -a 'Content-Type: text/html'
I've tested it with the mail text and then colouring is correct

framps, 2020-05-10:
I use exim and everything works fine for me. I frankly thought this means it works for all mail programs and never tested with postifx. Thank you very much you already provided a fix for postifx. I now have to check whether it still works for exim. Will come back to you and I will ask you to verify a fix on postfix :wink: .

WAdama, 2020-05-10:
Say the word and I will test... :)

framps, 2020-05-13:
I'm not an eMail expert and I was glad when I managed to get eMail clored with exim. I found [this page](https://unix.stackexchange.com/questions/15405/how-do-i-send-html-email-using-linux-mail-command) which makes me nervous because I don't want to fiddle with all the different eMail clients and their different options. 

Which setting do you use for
```
DEFAULT_MAIL_PROGRAM=
```
when you use postfix? 

Maybe the easiest fix for your issue is to add a new MAIL_PROGRAM parameter `SENDMAIL`.

WAdama, 2020-05-13:
DEFAULT_MAIL_PROGRAM= is set to **mail**, postfix seems to install a link to **mailx**, that's why the -a also works in mail for me.

I agree, it may easier than to play with existing part.

framps, 2020-05-13:
I think I found a solution without a new MAIL_PROGRAM parameter :wink: 

I created a fix in [this branch](https://github.com/framps/raspiBackup/tree/issue228_postfixColoring). Please use this version of raspiBackup.sh and test whether it now works with postfix. You don't have to update any configuration option.

WAdama, 2020-05-13:
I just tested it and it works not as hoped. 

I found this in the mail source code:
`To: test@test.de, text/html@server.service.test.de
Subject: Server: Backup erfolgreich beendet.
Content-Type:
MIME-Version: 1.0
`
Maybe this is of help for you...

framps, 2020-05-13:
Great it works now with postfix. 

But what do you want to tell me with the additional comment? Reading this it looks like the mail was sent to two recipients and one recipient eMail is somehow mangled with the html content type tail.  Which target eMail did you specify?

WAdama, 2020-05-13:
It seems you read it wrong: I just tested it and it works **not** as hoped...

framps, 2020-05-14:
There is new code available :wink: . There is now a new option `--eMailColoring` which is `subject` per default. You should use `option` to use the `-a` option instead of the subject line. Corresponding config option is `DEFAULT_EMAIL_COLORING`. Please test again with your postfix setup.

WAdama, 2020-05-14:
Have I only to set `DEFAULT_EMAIL_COLORING="OPTION"` in config file?

Because only setting it made no difference, the mail is send with content type in subject.

framps, 2020-05-14:
Are you sure you use the latest fix? Revision should be `669382b` (check with --version). I checked on my environment and when I use `DEFAULT_EMAIL_COLORING="OPTION"` it works for me. Actually finally email sending failed because exim option -a has another semantic then in postifx. I also can see in the debug log there is no Content type attached to the subject.

WAdama, 2020-05-14:
I checked it to be sure and it is the right version. According to debug log the option is used but the mail looks the same as with subject.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/4630653/raspiBackup.log)

framps, 2020-05-14:
> I've tested it with the mail text and then colouring is correct

Would you please show exactly the command you used to send a ContentType html eMail with postfix? As you see in the log the -a option is used now and I have no clue why it's still displayed in the subject :disappointed:

WAdama, 2020-05-14:
Of course...

The command is `mail -a 'Content-Type: text/html' -s "Test" test@test.de < ./mail.txt`

(I saved the text from a mail of raspiBackup in mail.txt)

framps, 2020-05-15:
Hm ... that's what is used by raspiBackup with one difference: There is no file used but a string wich is piped in. But that shouldn't cause th issue. 

I enhanced the logging in the eMail code section. That's what I get when I use OPTION with exim
```
20200515-113425 DBG 3035:      --- eMail: p@@@@@@@@@@@@e
20200515-113425 DBG 3036:      --- eMail Program: mail
20200515-113425 DBG 3037:      --- Subject: @HOSTNAME@: Backup started.
20200515-113425 DBG 3038:      --- ColoringOption: -a 'Content-Type: text/html'
20200515-113425 DBG 3039:      --- ContentType: 
20200515-113425 DBG 3040:      --- Parms: 
20200515-113425 DBG 3045:      --- mail -a 'Content-Type: text/html'  -s "@HOSTNAME@: Backup started."  p@@@@@@@@@@@@e <<< "</br> --- RBK0009I: @HOSTNAME@: raspiBackup.sh V0.6.5-hotfix (27ac20b) started at Fri May 15 11:34:25 CEST 2020.</br> --- RBK0167I: Sending email.</br> </br> </br>"
mail: cannot send message: Process exited with a non-zero status
20200515-113425 DBG 3048:      --- mail: RC: 36
```
Please check what's your result in the log.

vsc55, 2020-05-15:
Hi,

I have updated today to version 0.6.5 and now the emails reach me blank if the option "DEFAULT_COLORING" is defined with "CM", if I remove "M" I already receive emails correctly.

Checking the email that came in white I see that in the content of the message it is in the email header.

I am not using postfix, I use sSMTP 2.64.

Email header:
```
Return-Path: <root@dominio>
X-Envelope-To: to@dominio
Message-ID: <***@dominio>
X-Spam-Status: No, hits=0.9 required=5.0
	tests=AWL: -0.291, BAYES_00: -1.665, HTML_EXTRA_CLOSE: 0.001,
	HTML_MESSAGE: 0.001, HTML_MIME_NO_HTML_TAG: 0.377, MIME_HEADER_CTYPE_ONLY: 0.717,
	MIME_HTML_ONLY: 0.723, MISSING_MID: 0.497, TO_NO_BRKTS_HTML_ONLY: 0.591,
	URIBL_BLOCKED: 0.001, TOTAL_SCORE: 0.952,autolearn=no
X-Spam-Level: 
X-Footer: Y2VyZWJlbHVtLm5ldA==
Received: from host
	by **** with SMTP
	for *****;
	Fri, 15 May 2020 11:24:52 +0200
Received: by *****:25 (sSMTP sendmail emulation); Fri, 15 May 2020 11:24:52 +0200
From: "root" <****>
Date: Fri, 15 May 2020 11:24:52 +0200
To: *******
Subject: 8-) pdc02: Backup finished successfully.
Content-Type: text/html
</br>
--- RBK0009I: pdc02: raspiBackup.sh V0.6.5 (7752af8) started at vie may 15 11:14:46 CEST 2020.</br>
--- RBK0128I: Using logfile /backup//pdc02/pdc02-rsync-backup-20200515-111445/raspiBackup.log.</br>
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.</br>
<span style=color:#FF8000>!!! RBK0202W: 8-) Friendly reminder: Execute now a restore test. You will be reminded 0 times again.</span></br>
--- RBK0200I: /boot and / located on same partition .</br>
<span style=color:#FF8000>!!! RBK0236W: Partition oriented backup will not be maintained any more and disable somewhere in the future.</span></br>
--- RBK0151I: Using backuppath /backup.</br>
<span style=color:#FF8000>!!! RBK0157W: No services to stop.</span></br>
--- RBK0081I: Creating backup of type rsync in /backup//pdc02/pdc02-rsync-backup-20200515-111445.</br>
--- RBK0085I: Backup of type rsync started. Please be patient.</br>
--- RBK0036I: Saving partition layout.</br>
--- RBK0045I: Creating backup of partition layout in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.sfdisk.</br>
--- RBK0045I: Creating backup of partition layout in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.blkid.</br>
--- RBK0045I: Creating backup of partition layout in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.parted.</br>
--- RBK0045I: Creating backup of partition layout in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.fdisk.</br>
--- RBK0046I: Creating backup of master boot record in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.mbr.</br>
--- RBK0092I: Saving partition mmcblk0p1 (268MB) ...</br>
--- RBK0133I: Using directory /backup//pdc02/pdc02-rsync-backup-20200515-102110 for hardlinks.</br>
--- RBK0158I: Creating native rsync backup "/backup//pdc02/pdc02-rsync-backup-20200515-111445".</br>
--- RBK0085I: Backup of type rsync started. Please be patient.</br>
--- RBK0112I: Partition mmcblk0p1 was saved.</br>
--- RBK0092I: Saving partition mmcblk0p2 (62,2GB) ...</br>
--- RBK0133I: Using directory /backup//pdc02/pdc02-rsync-backup-20200515-102110 for hardlinks.</br>
--- RBK0158I: Creating native rsync backup "/backup//pdc02/pdc02-rsync-backup-20200515-111445".</br>
--- RBK0085I: Backup of type rsync started. Please be patient.</br>
--- RBK0112I: Partition mmcblk0p2 was saved.</br>
--- RBK0078I: Backup time: 00:05:08.</br>
<span style=color:#FF8000>!!! RBK0156W: No services to start.</span></br>
--- RBK0159I: 6 backups kept for rsync backup type.</br>
--- RBK0205I: Deleting oldest backup in /backup/pdc02. This may take some time. Please be patient.</br>
--- RBK0033I: Please wait until cleanup has finished.</br>
--- RBK0049I: Messages saved in /backup//pdc02/pdc02-rsync-backup-20200515-111445/raspiBackup.msg.</br>
--- RBK0026I: Debug logfile saved in /backup//pdc02/pdc02-rsync-backup-20200515-111445/raspiBackup.log.</br>
--- RBK0010I: pdc02: raspiBackup.sh V0.6.5 (7752af8) stopped at vie may 15 11:24:51 CEST 2020 with rc 0.</br>
--- RBK0017I: Backup finished successfully.</br>
--- RBK0229I: Telegram notified.</br>
--- RBK0167I: Sending email.</br>
</br>
</br>
```

framps, 2020-05-15:
@vsc55 Thank you very much you reported this issue. I created a new issue on behalf of you because it's related to sSMTP and not postfix. I frankly didn't test message coloring with sSMTP because I don't use sSMTP.

... looks like I fall into the trap of the different eMail client implementations by adding this coloring feature ...

WAdama, 2020-05-15:
I've just tested with the enhanced logging:
`20200515-135020 DBG 3035:      --- eMail: p@@@@e(16)
20200515-135020 DBG 3036:      --- eMail Program: mail
20200515-135020 DBG 3037:      --- Subject: @HOSTNAME@: Backup gestarted.
20200515-135020 DBG 3038:      --- ColoringOption: -a 'Content-Type: text/html'
20200515-135020 DBG 3039:      --- ContentType: 
20200515-135020 DBG 3040:      --- Parms: 
20200515-135020 DBG 3045:      --- mail -a 'Content-Type: text/html'  -s "@HOSTNAME@: Backup gestarted."  p@@@@e(16) <<< "</br> --- RBK0009I: @HOSTNAME@: raspiBackup.sh V0.6.5-hotfix (27ac20b) Fr 15. Mai 13:50:20 CEST 2020 gestartet.</br> --- RBK0167I: Eine eMail wird versendet.</br> </br> </br>"
20200515-135020 DBG 3048:      --- mail: RC: 0
`
Looks just as in your example. The funny thing is if I'm use this command munually then it works. So the command itself is correct...

As to in German: "Da hab' ich ja ein Faß aufgemacht"

framps, 2020-05-15:
> Looks just as in your example. The funny thing is if I'm use this command munually then it works. So the command itself is correct...

Well, I'm struggling with the space bashism ... so it may be the command executed on your side is different than the command executed by raspiBackup.

Would you mind to enable some addtl debugging and change

```
				$EMAIL_MAILX_PROGRAM)
					logItem "$EMAIL_PROGRAM $coloringOption $EMAIL_PARMS -s "\"$subject\"" $attach $EMAIL <<< "\"$content\"
					"$EMAIL_PROGRAM" "$coloringOption" $EMAIL_PARMS -s "$subject" $attach "$EMAIL" <<< "$content"
					rc=$?
```
to
```
				$EMAIL_MAILX_PROGRAM)
					logItem "$EMAIL_PROGRAM $coloringOption $EMAIL_PARMS -s "\"$subject\"" $attach $EMAIL <<< "\"$content\"
set -x
					"$EMAIL_PROGRAM" "$coloringOption" $EMAIL_PARMS -s "$subject" $attach "$EMAIL" <<< "$content"
					rc=$?
					set +x
```
?
Then you will get a debug line on your screen which displays the executed command. Please show this output.

For me it's 
```
--- RBK0167I: Sending email.
+ mail '' -s '@HOST@: Backup finished successfully.
MIME-Version: 1.0
Content-Type: text/html; charset=utf-8' @@EMAIL@@
+ rc=0
+ set +x
```

WAdama, 2020-05-15:
I've just done it, this is the output:

```
--- RBK0167I: Eine eMail wird versendet.
+ mail '-a '\''Content-Type: text/html'\''' -s '@HOST@: Backup gestarted.' @@EMAIL@@
+ rc=0
+ set +x logItem 'mail: RC: 0'

--- RBK0167I: Eine eMail wird versendet.
+ mail '-a '\''Content-Type: text/html'\''' -s '@HOST@: Backup erfolgreich beendet.' @@EMAIL@@
+ rc=0
+ set +x logItem 'mail: RC: 0'
```

framps, 2020-05-15:
As you can see and I thought - it's not exactly what you typed in on your system :cry: . That's one of the ugly features of bash - dealing with spaces in parametern. But I finally fixed it :smiley:  Try the latest version and it should work now :crossed_fingers:

WAdama, 2020-05-15:
Wouldn't life be boring without such fun.... ;)

But I'm afraid it didn't work as expected, sorry. I have following output on the console:

```
--- RBK0167I: Eine eMail wird versendet.
+ mail -a ''\''Content-Type: text/html'\''' -s 'GalacticaDC2: Backup gestarted.' push@asche-rz.de
+ rc=0

--- RBK0167I: Eine eMail wird versendet.
+ mail -a ''\''Content-Type: text/html'\''' -s 'GalacticaDC2: Backup erfolgreich beendet.' push@asche-rz.de
+ rc=0
```

Am I wrong or is the `'\'''` after text/html correct?

framps, 2020-05-15:
:angry: You're right. I didn't verify the fix with `set -x`. This bash space handling is very annoying. You will see the debug statement missed single quotes but when you use `set -x` they are there :flushed: . 

I think it now should work. :crossed_fingers: :crossed_fingers: :crossed_fingers: 
 
I'm glad you test this stuff. Sorry for the inconvenience.

WAdama, 2020-05-15:
Glad to be of help...

And now it works, I've got a colourful mail from raspiBackup... :thumbsup:

framps, 2020-05-15:
:upside_down_face: 

Thank you very much for your help on this issue. I'm glad whenever I managed to configure an eMail client/server. Whenever there is an eMail issue with raspiBackup with a client other than exim I need help from people like you running this eMail client/server :wink: 

And I'm glad the issue is solved. I frankly don't want to clonse an eMail issue wuth  `Your eMail client/server is not supported by raspiBackup`

Please close the issue if you're fine now.

WAdama, 2020-05-15:
You're quite welcome... :smiley:

It's the least I can do to help. And closing it now, I'm fine... :wink:

closed 2020-05-15 by WAdama