framps/raspiBackup #229: eMail colouring doesn't work with sSMTP

framps, 2020-05-15:
Hi,

I have updated today to version 0.6.5 and now the emails reach me blank if the option "DEFAULT_COLORING" is defined with "CM", if I remove "M" I already receive emails correctly.

Checking the email that came in white I see that in the content of the message it is in the email header.

I am not using postfix, I use sSMTP 2.64.

Email header:
```
Return-Path: <root@dominio>
X-Envelope-To: to@dominio
Message-ID: <***@dominio>
X-Spam-Status: No, hits=0.9 required=5.0
	tests=AWL: -0.291, BAYES_00: -1.665, HTML_EXTRA_CLOSE: 0.001,
	HTML_MESSAGE: 0.001, HTML_MIME_NO_HTML_TAG: 0.377, MIME_HEADER_CTYPE_ONLY: 0.717,
	MIME_HTML_ONLY: 0.723, MISSING_MID: 0.497, TO_NO_BRKTS_HTML_ONLY: 0.591,
	URIBL_BLOCKED: 0.001, TOTAL_SCORE: 0.952,autolearn=no
X-Spam-Level: 
X-Footer: Y2VyZWJlbHVtLm5ldA==
Received: from host
	by **** with SMTP
	for *****;
	Fri, 15 May 2020 11:24:52 +0200
Received: by *****:25 (sSMTP sendmail emulation); Fri, 15 May 2020 11:24:52 +0200
From: "root" <****>
Date: Fri, 15 May 2020 11:24:52 +0200
To: *******
Subject: 8-) pdc02: Backup finished successfully.
Content-Type: text/html
</br>
--- RBK0009I: pdc02: raspiBackup.sh V0.6.5 (7752af8) started at vie may 15 11:14:46 CEST 2020.</br>
--- RBK0128I: Using logfile /backup//pdc02/pdc02-rsync-backup-20200515-111445/raspiBackup.log.</br>
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.</br>
<span style=color:#FF8000>!!! RBK0202W: 8-) Friendly reminder: Execute now a restore test. You will be reminded 0 times again.</span></br>
--- RBK0200I: /boot and / located on same partition .</br>
<span style=color:#FF8000>!!! RBK0236W: Partition oriented backup will not be maintained any more and disable somewhere in the future.</span></br>
--- RBK0151I: Using backuppath /backup.</br>
<span style=color:#FF8000>!!! RBK0157W: No services to stop.</span></br>
--- RBK0081I: Creating backup of type rsync in /backup//pdc02/pdc02-rsync-backup-20200515-111445.</br>
--- RBK0085I: Backup of type rsync started. Please be patient.</br>
--- RBK0036I: Saving partition layout.</br>
--- RBK0045I: Creating backup of partition layout in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.sfdisk.</br>
--- RBK0045I: Creating backup of partition layout in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.blkid.</br>
--- RBK0045I: Creating backup of partition layout in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.parted.</br>
--- RBK0045I: Creating backup of partition layout in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.fdisk.</br>
--- RBK0046I: Creating backup of master boot record in /backup//pdc02/pdc02-rsync-backup-20200515-111445/pdc02-backup.mbr.</br>
--- RBK0092I: Saving partition mmcblk0p1 (268MB) ...</br>
--- RBK0133I: Using directory /backup//pdc02/pdc02-rsync-backup-20200515-102110 for hardlinks.</br>
--- RBK0158I: Creating native rsync backup "/backup//pdc02/pdc02-rsync-backup-20200515-111445".</br>
--- RBK0085I: Backup of type rsync started. Please be patient.</br>
--- RBK0112I: Partition mmcblk0p1 was saved.</br>
--- RBK0092I: Saving partition mmcblk0p2 (62,2GB) ...</br>
--- RBK0133I: Using directory /backup//pdc02/pdc02-rsync-backup-20200515-102110 for hardlinks.</br>
--- RBK0158I: Creating native rsync backup "/backup//pdc02/pdc02-rsync-backup-20200515-111445".</br>
--- RBK0085I: Backup of type rsync started. Please be patient.</br>
--- RBK0112I: Partition mmcblk0p2 was saved.</br>
--- RBK0078I: Backup time: 00:05:08.</br>
<span style=color:#FF8000>!!! RBK0156W: No services to start.</span></br>
--- RBK0159I: 6 backups kept for rsync backup type.</br>
--- RBK0205I: Deleting oldest backup in /backup/pdc02. This may take some time. Please be patient.</br>
--- RBK0033I: Please wait until cleanup has finished.</br>
--- RBK0049I: Messages saved in /backup//pdc02/pdc02-rsync-backup-20200515-111445/raspiBackup.msg.</br>
--- RBK0026I: Debug logfile saved in /backup//pdc02/pdc02-rsync-backup-20200515-111445/raspiBackup.log.</br>
--- RBK0010I: pdc02: raspiBackup.sh V0.6.5 (7752af8) stopped at vie may 15 11:24:51 CEST 2020 with rc 0.</br>
--- RBK0017I: Backup finished successfully.</br>
--- RBK0229I: Telegram notified.</br>
--- RBK0167I: Sending email.</br>
</br>
</br>
[... 117 lines cut ...]
20200515-105808 DBG 3039:          --- --- RBK0112I: Partition mmcblk0p1 was saved.</br>
20200515-105808 DBG 3039:          --- --- RBK0092I: Saving partition mmcblk0p2 (62,2GB) ...</br>
20200515-105808 DBG 3039:          --- --- RBK0133I: Using directory /backup//@HOSTNAME@/@HOSTNAME@-rsync-backup-20200515-050001 for hardlinks.</br>
20200515-105808 DBG 3039:          --- --- RBK0158I: Creating native rsync backup "/backup//@HOSTNAME@/@HOSTNAME@-rsync-backup-20200515-102110".</br>
20200515-105808 DBG 3039:          --- --- RBK0085I: Backup of type rsync started. Please be patient.</br>
20200515-105808 DBG 3039:          --- --- RBK0112I: Partition mmcblk0p2 was saved.</br>
20200515-105808 DBG 3039:          --- --- RBK0078I: Backup time: 00:29:09.</br>
20200515-105808 DBG 3039:          --- <span style=color:#FF8000>!!! RBK0156W: No services to start.</span></br>
20200515-105808 DBG 3039:          --- --- RBK0159I: 6 backups kept for rsync backup type.</br>
20200515-105808 DBG 3039:          --- --- RBK0205I: Deleting oldest backup in /backup/@HOSTNAME@. This may take some time. Please be patient.</br>
20200515-105808 DBG 3039:          --- --- RBK0033I: Please wait until cleanup has finished.</br>
20200515-105808 DBG 3039:          --- --- RBK0049I: Messages saved in /backup//@HOSTNAME@/@HOSTNAME@-rsync-backup-20200515-102110/raspiBackup.msg.</br>
20200515-105808 DBG 3039:          --- --- RBK0026I: Debug logfile saved in /backup//@HOSTNAME@/@HOSTNAME@-rsync-backup-20200515-102110/raspiBackup.log.</br>
20200515-105808 DBG 3039:          --- --- RBK0010I: @HOSTNAME@: raspiBackup.sh V0.6.5 (7752af8) stopped at vie may 15 10:58:07 CEST 2020 with rc 0.</br>
20200515-105808 DBG 3039:          --- --- RBK0017I: Backup finished successfully.</br>
20200515-105808 DBG 3039:          --- --- RBK0229I: Telegram notified.</br>
20200515-105808 DBG 3039:          --- --- RBK0167I: Sending email.</br>
20200515-105808 DBG 3039:          --- </br>
20200515-105808 DBG 3039:          --- </br> | ssmtp  j@@@@t(21)
20200515-105808 DBG 3042:          --- ssmtp: RC: 0
20200515-105808 DBG 3061:          <-- sendEMail
20200515-105808 DBG 3170:      --- Masquerading eMail
```

_Originally posted by @vsc55 in https://github.com/framps/raspiBackup/issues/228#issuecomment-629157181_

framps, 2020-05-15:
Thank you very much for reporting the issue. I don't use sSMTP thus I didn't test this coloring feature. 

I now have two possibilities to close this issue:
1. I close it with `coloring not supported for sSMTP` :disappointed: 
2. @vcs55 You help me to figure out how sSMTP allows to send html eMails :wink:

vsc55, 2020-05-15:
I'm doing tests, as soon as I know something I'll let you know.

vsc55, 2020-05-15:
This works correctly:
echo -e "To: ***@***\nFrom: **@***\nSubject: 8 pdc02: Backup finished successfully.\nContent-Type: text/html\n\n <span style=color:#FF8000> Hola Mundo</span>" | ssmtp  ***@****

vsc55, 2020-05-15:
Now the message is seen, but there is a new problem and that is that the line breaks (</br>) are not applied.

framps, 2020-05-15:
Thank you very much for your investigations and the patch. I created a new version in https://github.com/framps/raspiBackup/tree/issue228_postfixColoring which should fix the linebreak issue. Please test whether it works now for you.

vsc55, 2020-05-15:
El problema del salto de linea es que a Outlook no le gusta </br>, hay que usar <br>.

framps, 2020-05-15:
Good catch :+1: . That's incorrect html. I changed </br> to <br/> now in the new version. Please check whether it's OK now. :crossed_fingers:

vsc55, 2020-05-15:
I just tried the raspiBackup.sh V0.6.5-hotfix (377a483) and except for the line break problem the rest works fine.

vsc55, 2020-05-15:
And a suggestion.
Emojis could be added to telegram messages.
If copying failed ❎ and 🆗 when copying was successful.

framps, 2020-05-15:
Sorry - I forgot to push my local fix. Please try the latest version which uses `<br/`.

Re emojis in telegram: Thx for this info. Will keep this in my mind :smiley:

vsc55, 2020-05-15:
> 
> 
> Sorry - I forgot to push my local fix. Please try the latest version which uses `<br/>`.
> 
> Re emojis in telegram: Thx for this info. Will keep this in my mind 😃

I have not tested it with your latest version, but it should work since the tests I have done are with the same modifications and work correctly.

framps, 2020-05-15:
Great. Thank you very much for your help on solving this issue. Please execute a final test with the latest version and the close this issue. I will then roll in this fix in next release 0.6.5.1.

vsc55, 2020-05-15:
I have tried the latest version and everything works correctly. 👍
Glad to be of help.😃

PR #232 (2020-05-15): Update raspiBackup.sh
Pull Request to branch correly

It would need a double line break between the message header and the body.
Fix #229
commit 109ccc16 (2020-05-15): Update raspiBackup.sh (#232)

* Update problem-report.md

* Update question-template.md

* Update raspiBackup.sh

Fix  #229

fix br, change </br> to <br/>
closed 2020-05-15 by framps