framps/raspiBackup #233: Systemd Timer Unit deaktivieren

Hofei90, 2020-05-19:
Das stoppen und starten der Service Units funktioniert bisher recht gut. Freut mich, dass das Feature letztens aufgenommen worden ist.
Hierzu habe ich nun noch einen Erweiterungsvorschlag.
Service Units können auch in regelmäßigen Abständen von zugehörigen Timer Units angesprochen werden, wodurch unter Umständen während des Backups Dienste wieder aktiviert werden, welche man eigentlich über die Service Units gestoppt haben wollte.
Gibts es entsprechend zugehörige Timer Units erhält man in raspiBackup schon die Rückmeldung des Systems, dass es eine Timer Unit gibt. Die Meldung sieht wie folgt aus:

```
--- RBK0008I: Services werden gestoppt: 'systemctl stop sps_backup && systemctl stop redis-server && systemctl stop gitea && systemctl stop apache2 && systemctl stop mariadb'.
Warning: Stopping sps_backup.service, but it can still be activated by:
  sps_backup.timer
```
Wie in diesem Beispiel, könnte die sps_backup.service während des Backups durch die zugehörige Timer Unit, in diesem Fall, sps_backup.timer gestartet werden. 

Man könnte doch die Rückgabe nach dem die Service gestoppt worden sind auswerten, und entsprechende Timer Units auch stoppen
`systemctl stop <name_des_timers.timer> ` und noch dem Backup wieder starten `systemctl start name_des_timers.timer `

framps, 2020-05-19:
Ist das nicht gefaehrlich wenn timer units einfach so gestoppt werden und nicht laufen? Ihre Definition und Laufzeiten haben ja einen bestimmten Zweck.

Lass beispielsweite die Timerunit ein LetsEncrypt Certificate updaten. Wenn der Update unbemerkt nicht stattfindet weil raspiBackup ihn gestoppt hat wird eine Webseite die dann ein expired Certificate benutzt nicht mehr laufen. Es lassen sich sicherlich beliebige andere Beispiele finden wo diese Funktionalitaet kontraproduktiv ist.

Es ist aber kein Problem das Stoppen der Timerunits in die -a und -o Optionen aufzunehmen. Dann definiert man das explizit und weiss von dieser Einschraenkung.

Hofei90, 2020-05-19:
Deine Bedenken verstehe ich, bin aber der Auffassung, dass dann die Timer Unit falsch programmiert/erstellt wurde. Denn dann dürfte ich das System auch nie neu starten.

Auszug aus wiki.ubuntuusers: https://wiki.ubuntuusers.de/systemd/Timer_Units/
`OnActiveSec | relative Zeit bezogen auf den Zeitpunkt, als die Timer Unit zuletzt aktiviert wurde` <<versucht Probleme 
`OnBootSec | relative Zeit bezogen auf den Zeitpunkt, als der Rechner gestartet wurde` <<keine Probleme durch Backup
`OnStartupSec | relative Zeit bezogen auf den Zeitpunkt, als systemd gestartet wurde` <<keine Probleme durch Backup
`OnUnitActiveSec | relative Zeit bezogen auf den Zeitpunkt, als die Service Unit zuletzt aktiviert wurde` <<keine Probleme durch Backup (außer durch das vorherige stop der Service Unit verändert sich auch schon die Zeit, dann wär das Problem aber bereits vorhanden)
`OnUnitInactiveSec | relative Zeit bezogen auf den Zeitpunkt, als die Service Unit zuletzt deaktiviert wurde` <<keine Probleme durch Backup (außer durch das vorherige stop der Service Unit verändert sich auch schon die Zeit, dann wär das Problem aber bereits vorhanden)

Da die Wiki Seite @noisefloor erstellt hat, wär interssant wie er darüber denkt.

framps, 2020-05-22:
I'm frankly not conviced it makes sense to spend any efford on this issue ...

framps, 2020-06-01:
@Hofei90  Any updates on this issue from your side :wink: ?

Hofei90, 2020-06-02:
Gute Frage, ich bin der Meinung, dass es eine sinnvolle Erweiterung wäre.
Vielleicht würden wir mehr Feedback bekommen, wenn man das Thema im Forum nochmals anspricht.

framps, 2020-06-02:
Ich bin weiterhin der Meinung wie [oben](https://github.com/framps/raspiBackup/issues/233#issuecomment-630724377) geschrieben :wink: . Willst du das Thema mal im Forum ansprechen?

framps, 2020-06-27:
So wie ich es sehe gibt es keinen Fortschritt zu diesem Thema. Ich schlage vor Du schliesst den Issue.

closed 2020-06-29 by Hofei90