framps/raspiBackup #234: raspiBackupRestore2Image.sh: Error sending the email after pishrink

STW65, 2020-05-19:
pishrink.sh: Shrunk /media/pi/Backup/pihole/pihole-tgz-backup-20200519-043001/pihole-backup.dd from 30G to 6,1G ...

/dev/stdin: Zeile 17: extractVersionFromFile: Kommando nicht gefunden.

-- Send email succeeded!

pishrink.sh: Shrunk /…/pihole-tgz-backup-20200519-043001/pihole-backup.dd from 30G to 6.1G ...
/ dev / stdin: line 17: extractVersionFromFile: command not found.
- Send email succeeded!

no email was sent.

framps, 2020-05-19:
Would you please read [this page](https://github.com/framps/raspiBackup/blob/master/.github/ISSUE_TEMPLATE/problem-report.md) :wink:

STW65, 2020-05-19:
rasbian Buster 10

raspiBackup
Version: 0.6.5 CommitSHA: 7752af8 CommitDate: 2020-05-07 CommitTime: 14:18:31

Log file gibt es keines. Das Ganze passiert zwischen dem fertigem pishrink und dem fdisk

Sorry! Wie geht das mit der Sprache? Mein erstes Ticket.

framps, 2020-05-19:
Kein Problem mit Deutsch. Wenn es ein allemeines Problem ist erstelle ich am Ende eine Summary in English und aendere dann auch den Titel.

framps, 2020-05-19:
Du benutzt noch eine Betaversion von 0.6.5. Upgrade die mal auf die aktuelle Version mit
```
sudo raspiBackup.sh -U -S
```
und probiere noch mal.

STW65, 2020-05-19:
Ok. Daran war ich mir nicht bewusst. Habe keine Beta runtergeladen und installiert.
Aber gut habe ich herunter geladen und gestartet. 

The same.

`pishrink.sh: Shrinking image ...
pishrink.sh: Shrunk /media/pi/Backup/pihole/pihole-tgz-backup-20200519-043001/pihole-backup-full.dd from 30G to 6,1G ...
/dev/stdin: Zeile 17: extractVersionFromFile: Kommando nicht gefunden.
-- Send email succeeded!

Welcome to fdisk (util-linux 2.33.1).
Changes will remain in memory only, until you decide to write them.
Be careful before using the write command.

Command (m for help): 
Expert command (m for help): 
Enter the new disk identifier: 
Disk identifier changed from 0x68dcc2e3 to 0xea7d04d6.

Expert command (m for help): 
Command (m for help): The partition table has been altered.
Calling ioctl() to re-read partition table.
Syncing disks.
`

Nur, ich vermute den Fehler nicht im rasiBackup selbst, denn dass läuft ohne Probleme durch.
Ach das eigentliche Restore2Image auch. Selbst pishrink und fdisk.
Das Script raspiImageMail in Zusammenhang mit dem Script raspiBackupRestore2Image und der MSG Variablen im /tmp Verzeichnis macht Probleme, denke ich. Mail vom raspiBackup selbst funktioniert.

framps, 2020-05-19:
Ach so. Das ist eine wichtige Info. Das Script wurde von kmbach erstellt und hat offensichtlich mit der 0.6.5 Probleme da es sehr von der Implementierung von raspiBackup abhängt.

Brauchst Du das unbedingt? Sonst nimmes doch raus.

STW65, 2020-05-19:
Nein, dass ist nicht wirklich kriegsentscheidend. ;-) 

Danke! Für die schnelle Reaktion.

framps, 2020-05-19:
Ich denke ich habe die Ursache gefunden und gefixed. Dazu gibt es jetzt [diesen Branch](https://github.com/framps/raspiBackup/tree/issue234_raspiImageMail/helper). Wenn Du Lust hast zieh Dir die neuen imageMail runter und probiere es mal. Der Fix ist aber nicht von mir getestet ... Du bist also das Guineapig :wink:

STW65, 2020-05-19:
Also ...
Fehlermeldung ist weg, aber trotz Meldung
"-- Send email succeeded!" keine eMail.

Ich hatte dass schnell und schmutzig so gelöst:

bei pishrink den Parameter d angefügt (erzeugt pishrink.log)
 
sudo mail -s "$HOST_NAME - $IMAGE_FILENAME - $status" email@adresse < pishrink.log

framps, 2020-05-19:
Ok. D.h. Du hast eine Loesung gefunden und ich brauche nicht mehr weiter reinzusehen :+1: Closed Du den Issue?

STW65, 2020-05-19:
Jupp. Alles klar.

closed 2020-05-19 by STW65