framps/raspiBackup #239: Backup fehlerhaft beendet nach Update auf v0.6.5 - RBK0225E: Ungültige eMail "root"

git-olicat, 2020-05-31:
Hallo!

Nach jahrelangem problemlosen Betrieb (danke dafuer!) und auch einer Vielzahl von Updates gab es nun erstmals ein Problem nach einem Update.
Seit ich von v0.6.4.3 auf v0.6.5 aktualisiert habe, klappt das automatische Backup nicht mehr: 
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.5 (f822b75) Son Mai 31 22:48:38 CEST 2020 gestartet.
--- RBK0128I: Logdatei ist /root/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
!!! RBK0124W: Simulationsmodus an.
??? RBK0225E: Ungültige eMail "root".
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0049I: Meldungen wurden in /root/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.log gesichert.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.5 (f822b75) Son Mai 31 22:48:41 CEST 2020 beendet mit Returncode 107.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0167I: Eine eMail wird versendet.

Leider findet Google die Fehlernummer RBK0225E nirgends - daher lege ich mir etas die Karten.
Ein Hinzufuegen von @localhost bei der Emailadresse brachte keine Aenderungen.
Ein Downgrade auf die v0.6.4.3 klappt problemlos. Wie auch damit das Backup. Ein neuerliches Update auf v0.6.5 bringt wiederden gleichen Fehler.

Ist der Fehler bekannt?
Was kann ich tun?
Soll ich noch etwas zur Fehlersuche beistellen?

Viele Gruesse, Oliver

framps, 2020-05-31:
Das ist ein Fehler der schon - leider nicht im github - reported wurde. In 0.6.5 wird geprueft ob die eMail eine valide eMail ist - aber leider habe ich uebersehen dass es auch lokale eMails gibt wie `root` die nicht in das Muster von email@dummy.com fallen. Es ist schon gefixed und wird in der naechsten Version 0.6.5.1, die demnaechst published wird, gefixed sein.

Je nachdem wie dringend Dein Problem ist kannst Du auf 0.6.5.1 warten oder auch den folgenden Code einfach in raspiBackup.sh loeschen:
```
		if ! [[ "$EMAIL" =~ ^[a-zA-Z0-9_.+-]+\@[a-zA-Z0-9-]+\.[a-zA-Z0-9.-]+$ ]]; then
			writeToConsole $MSG_LEVEL_MINIMAL $MSG_INVALID_EMAIL "$EMAIL"
			exitError $RC_PARAMETER_ERROR
		fi
```

git-olicat, 2020-05-31:
Danke!
Ich habe die Zeile 5028-5031 auskommentiert.
Fake-Backup funktioniert.

Oliver

framps, 2020-05-31:
Perfekt. Dann schliesse ich mal diesen Issue fuer Dich :wink:

closed 2020-05-31 by framps