framps/raspiBackup #241: v0.6.5.1: problems with mail/nullmailer

git-olicat, 2020-06-03:
Hi!

Das Problem liegt moeglicherweise an meiner "speziellen" Konfiguration - das hatte aber bis zur v6.5.0 problemlos funktioniert. Beim fake-Backup kommt es zu einer Fehlermeldung beim Versand der Status-Mail. Bitte lass mich wissen, ob ich noch etwas dazu beitragen kann.

Gruss, Oliver

root@raspberrypi:~# raspiBackup.sh -F
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.5.1 (d66749e) Mit Jun  3 10:41:36 CEST 2020 gestartet.
--- RBK0128I: Logdatei ist /root/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
!!! RBK0124W: Simulationsmodus an.
--- RBK0141I: Nur der von den definierten Partitionen belegte Speicherplatz wird gesichert.
--- RBK0151I: Backuppfad /srv/oli-ubuntu2/raspi wird benutzt.
--- RBK0008I: Services werden gestoppt: 'service samba stop;service mysql stop;service cron stop'.
Stopping Samba daemons: nmbd smbd.
Stopping MySQL database server: mysqld.
Stopping periodic command scheduler: cron.
--- RBK0081I: Backup vom Typ dd wird in /srv/oli-ubuntu2/raspi/raspberrypi/raspberrypi-dd-backup-20200603-104135 erstellt.
--- RBK0003I: Backupgröße wird von 15.06 GiB auf 7.36 GiB reduziert.
--- RBK0085I: Backuperstellung vom Typ dd gestartet. Bitte Geduld.
--- RBK0078I: Backupzeit: 00:00:00.
--- RBK0007I: Services werden gestartet: 'service samba start;service mysql start;service cron start'.
Starting Samba daemons: nmbd smbd.
Starting MySQL database server: mysqld . ..
Checking for tables which need an upgrade, are corrupt or were
not closed cleanly..
Starting periodic command scheduler: cron.
--- RBK0159I: 5 Backups werden für den Backuptyp dd aufbewahrt.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0049I: Meldungen wurden in /root/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup.log gesichert.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.5.1 (d66749e) Mit Jun  3 10:41:51 CEST 2020 beendet mit Returncode 0.
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0167I: Eine eMail wird versendet.
raspberrypi: contains invalid character ':'
beendet.
MIME-Version: contains invalid character '\012'
1.0
Content-Type: contains invalid character '\012'
usage: nullmailer-inject [flags] [recipients] <message
Reformat and inject a message into the nullmailer queue
  -a, --use-args       Use only command-line arguments for recipients
  -b, --use-both       Use both command-line and message header for recipients
  -e, --use-either     Use either command-line and message header for recipients
  -h, --use-header     Use only message header for recipients
  -f, --from=VALUE     Set the sender address
  -n, --no-queue       Send the formatted message to standard output
  -v, --show-envelope  Show the envelope with the message

  -h, --help           Display this help and exit
nullmailer-inject: unknown option letter -s

framps, 2020-06-03:
Die Ursache liegt am eMail Coloring und der unterschiedlichen Art bei den eMail Clients wie man den ContentType text/html setzt. Wenn Du das ausschaltest wirst Du den Fehler nicht bekommen (COLORING nur auf "C". setzen).

Ich hatte laenger gezoegert diesen Suppport einzubauen weil ich schon so etwas befuerchtet hatte :cry: . 

Kannst Du das Debuglog welches in dem Backupverzeichnis abgelegt wird hier posten damit ich genauere Infos bekomme?

git-olicat, 2020-06-03:
Hi!

Waere es bei erwartbaren Problemen dann nicht sinnvoller, dieses neue Feature optional zu setzen?

>  Wenn Du das ausschaltest wirst Du den Fehler nicht bekommen (COLORING nur auf "C". setzen).

Das reicht offenbar noch nicht, um wieder fehlerfrei zu arbeiten. Siehe anderes Log.

Logfiles anbei:
raspiBackup.log im CM-Mode
raspiBackup-nurC.log im C-Mode

[raspiBackup-nurC.log](https://github.com/framps/raspiBackup/files/4722572/raspiBackup-nurC.log)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4722522/raspiBackup.log)

Gruss, Oliver

framps, 2020-06-03:
Danke fuer die Logs. 

> das hatte aber bis zur v6.5.0 problemlos funktioniert.

D.h. also - nur um sicherzustelle ich verstehe Dich richtig - mit 0.6.4.3 hat es noch funktioniert und ab 0.6.5 sowie der 0.6.5.1 gibt es das Problem? 

> Waere es bei erwartbaren Problemen dann nicht sinnvoller, dieses neue Feature optional zu setzen?

Sagen wir mal so :smiley: . raspiBackup unterstuetzt exim, postfix, ssmtp und msmtp. Die drei eMailClients sind getestet worden. Der nullmailer wird nicht unterstuetzt (ich wusse bis dato nicht das es den ueberhaupt gibt).  Dass es bislang funktioniert hat ist Dein Glueck :wink: . Aber ich versuche gerne auch den nullmailer zu unterstuetzen. Ich habe ihn eben bei mir auf einer Spielraspi installiert - nur bekomme ich bislang Fehler beim Senden. Irgendwas stimmt bei der Konfiguration bei mir noch nicht. 

Hast Du vielleicht einen besseren Link als [diesen hier](https://wiki.debian.org/nullmailer) fuer mich?

git-olicat, 2020-06-03:
> mit 0.6.4.3 hat es noch funktioniert und ab 0.6.5 sowie der 0.6.5.1 gibt es das Problem?

Mit der "gepatchten" Version v0.6.5 (#239) gab es keine Probleme. Erst die Aenderungen der v0.6.5.1 erzeugen bei mir hier die Email-Probleme.

Sorry, ich wusste nicht, dass der nullmailer so exotisch ist.
Meine nullmail-Config ist offenbar sehr uebersichtlich. In /etc/nullmailer/adminaddr und defaultdomain steht nix drin. In defaulthost nur localhost und bei remotes die IP-Adresse des tatsaechlichen internen Mailservers. Alle Dateien sind seit 12/2015 unberuehrt. ;-)

Gruss, Oliver

[raspiBackup-v065.log](https://github.com/framps/raspiBackup/files/4722880/raspiBackup-v065.log)

framps, 2020-06-03:
Gut dass ich nachgefragt habe. Ja in der 0.6.5.1 sind noch ein paar Aenderungen wg Coloring drin. Ich sehe mir das mal genauer an. Waere doch gelacht wenn wir den nomailer nicht hinbekommen :-)

framps, 2020-06-03:
Ich habe jetzt mehrere Stunden versucht nullmailer beizubringen eine eMail ueber meinen Provider zu schicken. Klappt nicht. Dann habe ich mich mal ueber nullmailer eingelesen. Wenn man die nullmailer speziellen Tools benutzt und nicht `mail` sieht es schon anders aus. Da klappte dann auch mein Emailversenden über meinen Provider.

Ich habe mal die v0.6.5 mit der 0.6.5.1 verglichen und eine Aenderung in einem [dedizierten branch](https://github.com/framps/raspiBackup/tree/nullmailer) zurueckgedreht. Teste mal ob es damit wieder geht.

git-olicat, 2020-06-03:
Leider nicht.
:-(
Anbei zum Vergleich das Logfile der "Spezialversion" und der v0.6.4.3 bei das Senden der Mail noch immer klappt.
Aber investiere bitte nicht mehr soviel Zeit fuer "mein" Spezialproblem. Ich kann auch gut bei der funktionierenden Version von raspiBackup bleiben.
Offenbar bin ich ja weltweit der einzige Nutzer ...
;-)
Schoenen Abend, Oliver

[raspiBackup-0643.log](https://github.com/framps/raspiBackup/files/4725610/raspiBackup-0643.log)
[raspiBackup-0652devel.log](https://github.com/framps/raspiBackup/files/4725611/raspiBackup-0652devel.log)

framps, 2020-06-04:
Vielen Dank fuer das Log. Das ist sehr merkwuerdig:

In 0.6.5.1 steht
```
20200603-203601 DBG 3057:          --- mail  -s @HOSTNAME@: Backup erfolgreich beendet. @@@@4
@HOSTNAME@: contains invalid character ':'
```

und in 0.6.5 log steht
```
20200603-131952 DBG 3010:          --- mail  -s @HOSTNAME@: Backup erfolgreich beendet.  @@@@4
20200603-131952 DBG 3018:          --- mail: RC: 0
```

Es ist also identisch. Kann sein dass noch irgendwelche nicht darstellbaren Zeichen darin stehen.

kannst Du mal um die zu sehen folgendes machen:

Ändere mal in der 0.6.5.1
```
$EMAIL_MAILX_PROGRAM)
					logItem "$EMAIL_PROGRAM" "${coloringOption[@]}" $EMAIL_PARMS -s "\"$subject\"" $attach $EMAIL <<< "\"$content\""
					"$EMAIL_PROGRAM" "${coloringOption[@]}" $EMAIL_PARMS -s "$subject" $attach "$EMAIL" <<< "$content"
					rc=$?
					logItem "$EMAIL_PROGRAM: RC: $rc"
					;;
```
in
```
$EMAIL_MAILX_PROGRAM)
					logItem "$EMAIL_PROGRAM" "${coloringOption[@]}" $EMAIL_PARMS -s "\"$subject\"" $attach $EMAIL <<< "\"$content\""
set -x
					"$EMAIL_PROGRAM" "${coloringOption[@]}" $EMAIL_PARMS -s "$subject" $attach "$EMAIL" <<< "$content"
					rc=$?
set +x
					logItem "$EMAIL_PROGRAM: RC: $rc"
					;;
```
und zeige mal die zusaetzlichen Zeilen die dann auf den Bildschirm geschrieben werden.

Wenn Du das dann auch noch in der 0.6.5 Version machen koenntest damit ich das vergleichen kann waere gut.

git-olicat, 2020-06-04:
Guten Morgen!

Das Problem liegt tatsaechlich in der coloringOption begruendet.
Starte ich lokal den mail/nullmailer so wie es aus dem raspiBackup-Script gestartet wird mit:

mail '' -s 'raspberrypi: Backup erfolgreich beendet.' root </dev/null

erfolgt die gleiche Fehlermeldung wie im raspiBackup.sh - der nullmailer mag die '' vor dem -s nicht.

Verzichte ich auf die coloringOption (hier ''):

mail -s 'raspberrypi: Backup erfolgreich beendet.' root </dev/null

klappt das tadellos.

Dein Aufruf des mailers in der funktionierenden v0.6.5:
"$EMAIL_PROGRAM" $EMAIL_PARMS -s "$subject" $attach "$EMAIL" <<< "$content"

und in v0.6.5.1ff:
"$EMAIL_PROGRAM" "${coloringOption[@]}" $EMAIL_PARMS -s "$subject" $attach "$EMAIL" <<< "$content"

Hinzugekommen ist "${coloringOption[@]}" - verzichtest Du auf die Anfuehrungszeichen - machst also nur ${coloringOption[@]} - klappt das Backup.
Ich habe jetzt aber nicht geprueft, inwiefern die Anfuehrungszeichen bei tatsaechlich befuellter coloringOption wichtig sind.

ohne Anfuehrungszeichen:
-->
+ mail -s 'raspberrypi: Backup erfolgreich beendet.' root
+ rc=0

mit Anfuehrungszeichen - also Original 0.6.5.1:
-->
+ mail '' -s 'raspberrypi: Backup erfolgreich beendet.' root
raspberrypi: contains invalid character ':'
+ rc=0

Gruss, Oliver

framps, 2020-06-04:
Moin Oliver,

sehr gute Analyse des Problems und sehr hilfreich fuer mich :+1: . 

`${coloringOption[@]}` ist leer wenn kein coloring benutzt wird. Aber die `""` tauchen natuerlich trotzdem im Aufruf auf. D.h. Du hast die Ursache gefunden. Jetzt muss ich nur noch rausfinden wie ich die `""` wenn nichts in den Optionen steht unterduecken kann. Da muss ich mich noch tiefer in bash und Quoting bzw space handling einlesen und probieren. Aber nachdem Du jetzt die Ursache gefunden hast wird sich da auch noch eine Loesung fuer finden :smiley: 

keep you posted ...

git-olicat, 2020-06-04:
Ich verstehe gar nicht, was Du mit dem [@] dabei willst.
Ginge nicht einfach das Setzen
statt:
coloringOption=(-a "Content-Type: text/html")
einfach:
coloringOption='-a "Content-Type: text/html"'
(als die doppelten Anfuehrungszeichen innen und einfache aussen herum)

Und bei der Abfrage dann statt:
"${coloringOption[@]}"
ein:
${coloringOption}

ohne jegliche Anfuehrungszeichen?

Oliver

framps, 2020-06-04:
Perfekt. Du kennst Dich mit bash aus :+1:.

bash ist tricky wenn es mit Strings arbeiten muss wenn es durch Leerzeichen getrennte Teilstrings hat. Teilstring1: `-a`. Teilstring2:  ` "Content-Type: text/html"`.
Die uebliche Trick ist dazu bash Arrays zu benutzen und mit `"${coloringOption[@]}"` wird aus dem Array wieder ein mit Leerzeichen getrennte Stringansammlung. Und als ordentlicher basher habe ich `"` um den wieder zusammengebauten String gemacht. Ich denke wenn man die `"` weglaesst ist Dein Problem geloest. Nur muss ich auch testen ob dann noch alles funktioniert wenn dann doch mal was in coloringOption steht. 

Aber Dein Vorschlag ist auch gut. Vielleicht bin ich da etwas ueber das Ziel hinausgeschossen mit dem Array. Ich werde mir auch den Vorschlag genauer ansehen.

framps, 2020-06-05:
I created a fix in https://github.com/framps/raspiBackup/commits/nullmailer and used your proposal. Please test.

git-olicat, 2020-06-05:
Hi!

Fuer mich sieht das jetzt soweit gut aus. Das Fake-Backup war erfolgreich.
Ohne Fehlermeldung.
Mit Mail. 
Danke!

Oliver

[raspiBackup-200605.log](https://github.com/framps/raspiBackup/files/4736377/raspiBackup-200605.log)

--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.5.2-dev (350a607) Fre Jun  5 15:26:57 CEST 2020 gestartet.
!!! RBK0192W: =========> HINWEIS <=========
!!! RBK0192W: Dieses ist eine Entwicklerversion welcher nicht in Produktion benutzt werden sollte.
!!! RBK0192W: =========> HINWEIS <=========
--- RBK0128I: Logdatei ist /root/raspiBackup-cede1d7b5792ac326bfd90ca443d27b58ea249dd/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
!!! RBK0124W: Simulationsmodus an.
--- RBK0141I: Nur der von den definierten Partitionen belegte Speicherplatz wird gesichert.
--- RBK0151I: Backuppfad /srv/oli-ubuntu2/raspi wird benutzt.
--- RBK0008I: Services werden gestoppt: 'service samba stop;service mysql stop;service cron stop'.
--- RBK0081I: Backup vom Typ dd wird in /srv/oli-ubuntu2/raspi/raspberrypi/raspberrypi-dd-backup-20200605-152655 erstellt.
--- RBK0003I: Backupgröße wird von 15.06 GiB auf 7.36 GiB reduziert.
--- RBK0085I: Backuperstellung vom Typ dd gestartet. Bitte Geduld.
--- RBK0078I: Backupzeit: 00:00:00.
--- RBK0007I: Services werden gestartet: 'service samba start;service mysql start;service cron start'.
--- RBK0159I: 5 Backups werden für den Backuptyp dd aufbewahrt.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0049I: Meldungen wurden in /root/raspiBackup-cede1d7b5792ac326bfd90ca443d27b58ea249dd/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /root/raspiBackup-cede1d7b5792ac326bfd90ca443d27b58ea249dd/raspiBackup.log gesichert.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.5.2-dev (350a607) Fre Jun  5 15:27:22 CEST 2020 beendet mit Returncode 0.
--- RBK0017I: Backup erfolgreich beendet.
--- RBK0167I: Eine eMail wird versendet.

framps, 2020-06-05:
Das sieht ja schon vielversprehend aus. Kannst Du auch noch mal testen ob bei Dir auch das eMailColoring geht?

git-olicat, 2020-06-05:
Hi!

Funktioniert auf Konsole wie auch in der Mail.
Evolution zeigt die Farben bei mir an.
Roundcube als Webmailer, Outlook 2013 auf dem Windows-Rechner und Aquamail auf dem Smartphone nicht - aber das mag eine Einstellungssache sein. Die html-Tags sind jedenfalls drin.
Insofern: Ich bin zufrieden! Vielen Dank!

Schoenes Wochenende, Oliver

framps, 2020-06-05:
:+1: Perfekt. Vielen Dank fuer Deine hilfreiche Unterstuetzung. War sehr hilfreich dass Du mitgedacht hast, eine Loesung vorgeschlagen hast und entsprechende Logs geliefert hast.

Ich werde den Fix in die 0.6.5.1 einfliessen lassen und bis morgen published haben. D.h. Du bekommst diesen neuen Stand dann mit `raspiBackup.sh -U -S` installiert (Selbige Release, neuere Revision). Dann noch mal kurzen test mit `-F` und dann bitte den Issue closen.

git-olicat, 2020-06-08:
Danke!

mcmellow, 2021-02-15:
> Hi!
>
> Funktioniert auf Konsole wie auch in der Mail.
> Evolution zeigt die Farben bei mir an.
> Roundcube als Webmailer, Outlook 2013 auf dem Windows-Rechner und Aquamail auf dem Smartphone nicht - aber [...]

Hi Oliver,

I try to get nullmailer working with raspiBackup, but there are no mails send from raspiBackup in the fake mode.  Can you give me some tips to set the correct options in raspBackup and nullmailer

kinde regards
McMelloW

git-olicat, 2021-02-16:
Hi!

This is definitely not the right place for support regarding nullmailer.
In any case, not much needs to be set with the nullmailer.
See these links:
https://itwelt.org/anleitungen-howto/linux-740817696/624-nullmailer-einrichten-mails-an-smtp-relay-weiterleiten
http://www.troubleshooters.com/linux/nullmailer/

The only important thing is that /etc/nullmailer/remotes and /etc/nullmailer/defaulthost are filled correctly.
As far as I remember there's no need for special configuration regarding raspiBackup and nullmailer.

Regards, Oliver

mcmellow, 2021-02-16:
Thanks Oliver,  for your reply. I will dig in it further.
Framps point me on this issue after a question on raspiBackup & nullmailer.  Therefore I dare to ask you about it

aki42, 2024-08-15:
> Das sieht ja schon vielversprehend aus. Kannst Du auch noch mal testen ob bei Dir auch das eMailColoring geht?

Zur Info:
Bei mir hat sich aus der Kombination nullmailer + eMailColoring ebenfalls ein Problem ergeben - ich hatte aber sehr lange nicht bemerkt, dass mir raspiBackup.sh keine E-Mails mehr schickt, weiß also nicht, mit welcher Version das Problem begann.

DEFAULT_COLORING hatte in meiner Konfiguration den Wert "C". Als Fehlermeldung ergab sich laut Protokoll immer:
`nullmailer-inject: Invalid header line:`
`  To: , k@@@@e(21)`
`  Unable to parse the addresses.`
`Can't send mail: sendmail process failed with error code 1`

Die seltsame Kombination der Ziel-Adressen in "To:" ließ sich auf der Kommandozeile mit "mail '' -s […]" reproduzieren, also auch hier war die wohl durch das deaktivierte Coloring leere Zeichenkette Auslöser.
Der Fehler verschwand erst, als ich DEFAULT_COLORING auf "CM" (das alleine reichte noch nicht) _und_ DEFAULT_EMAIL_COLORING von "SUBJECT" auf "OPTION" umgestellt hatte.

framps, 2024-08-16:
@aki42 Vielen Dank fuer den Hinweis.

Leider ist das mit den eMailern kein Zuckerschlecken da immer welche gibt die ihr eigenes Sueppchen bzgl Coloring koechen.

Bist Du sicher dass ein Upgrade von raspiBackup den Fehler hervorgerufen hat? In dieser Ecke habe ich schon lange nichts mehr geaendert, Ich vermute eher dass ein Upgrade des Nullmailers dazu gefuehrt hat.

Leider kann ich ohne Debuglog schlecht die Ursache finden. Um den Fehler zu lokalisieren musst Du mir schon ein Debuglog von dem Zustand wo es nicht funktioniert und eines wo es funktioniert liefern :wink:

aki42, 2024-08-17:
> Leider ist das mit den eMailern kein Zuckerschlecken da immer welche gibt die ihr eigenes Sueppchen bzgl Coloring koechen.

Jau, kann ich mir gut vorstellen.

> Bist Du sicher dass ein Upgrade von raspiBackup den Fehler hervorgerufen hat? In dieser Ecke habe ich schon lange nichts mehr geaendert, Ich vermute eher dass ein Upgrade des Nullmailers dazu gefuehrt hat.

Ich kann und will da überhaupt keine Verantwortlichkeiten zuordnen - ich habe halt nur bemerkt, dass es mal funktioniert hat, sich das aber ohne mein wissentliches Zutun irgendwann geändert hat. :shrug: 

> Leider kann ich ohne Debuglog schlecht die Ursache finden. Um den Fehler zu lokalisieren musst Du mir schon ein Debuglog von dem Zustand wo es nicht funktioniert und eines wo es funktioniert liefern 😉

Hab ich beides in den Sicherungen. Brauchst Du die kompletten raspiBackup.logs oder nur bestimmte Abschnitte? Ich muss ja, wenn ich die Dateien hier hochlade, dadurch nicht z. B. jedem auf die Nase binden, wohin der Raspi meine Online-Backups schaufelt … :)

framps, 2024-08-17:
> Hab ich beides in den Sicherungen. Brauchst Du die kompletten raspiBackup.logs oder nur bestimmte Abschnitte? Ich muss ja, wenn ich die Dateien hier hochlade, dadurch nicht z. B. jedem auf die Nase binden, wohin der Raspi meine Online-Backups schaufelt … :)

Das klingt mir so als gibt es noch etwas was ich im Debuglog maskieren sollte. Wuerde ich gerne tun - nur muss ich wissen was es ist :thinking: . Kannst Du mit die Teile aus dem Debuglog - dann von Dir manuell maskiert - zeigen wo noch Maskierung fehlt?

```
20240811-030600 DBG 4386:      --> sendEMail 
20240811-030600 DBG 4429:          --- Coloring option: CM
20240811-030600 DBG 4429:          --- eMailColoring: OPTION
20240811-030600 DBG 4429:          --- subject: @HOSTNAME@: Backup finished successfully.
20240811-030600 DBG 4429:          --- coloring: -a Content-Type: text/html;
20240811-030600 MSG 4435:          --- RBK0167I: Sending email. 
20240811-030600 DBG 4452:          --- eMail: a@@@@e(16)
20240811-030600 DBG 4453:          --- eMail Program: mail
20240811-030600 DBG 4454:          --- Subject: @HOSTNAME@: Backup finished successfully.
20240811-030600 DBG 4455:          --- ColoringOption: -a Content-Type: text/html;
20240811-030600 DBG 4456:          --- ContentType: 
20240811-030600 DBG 4457:          --- Parms: 
20240811-030600 DBG 4462:          --- mail -a Content-Type: text/html; -s "@HOSTNAME@: Backup finished successfully." a@@@@e(16)
20240811-030600 DBG 4465:          --- mail: RC: 0
20240811-030600 DBG 4508:      <-- sendEMail 
```
Den LogTeil von sendEMail brauche ich aus beiden Logs.

aki42, 2024-08-17:
> > Hab ich beides in den Sicherungen. Brauchst Du die kompletten raspiBackup.logs oder nur bestimmte Abschnitte? Ich muss ja, wenn ich die Dateien hier hochlade, dadurch nicht z. B. jedem auf die Nase binden, wohin der Raspi meine Online-Backups schaufelt … :)
> 
> Das klingt mir so als gibt es noch etwas was ich im Debuglog maskieren sollte. Wuerde ich gerne tun - nur muss ich wissen was es ist 🤔 . Kannst Du mit die Teile aus dem Debuglog - dann von Dir manuell maskiert - zeigen wo noch Maskierung fehlt?

Ich denke nicht, dass sich derlei mit vernünftigem Aufwand automatisiert maskieren lässt - es geht um Einträge in der fstab bzw. Zeilen in der Auflistung von mounts. Da stehen zwar auch bei mir keine credentials drin, aber des einen Paranoia resultierte hier wohl in des anderen Overengineering. ;-)

> Den LogTeil von sendEMail brauche ich aus beiden Logs.

Ok. E-Mail-Versand funktioniert nicht:
```
20240812-022616 DBG 4386:      --> sendEMail 
20240812-022616 MSG 4435:          --- RBK0167I: Eine eMail wird versendet. 
20240812-022616 DBG 4452:          --- eMail: k@@@@e(21)
20240812-022616 DBG 4453:          --- eMail Program: mail
20240812-022616 DBG 4454:          --- Subject: @HOSTNAME@: Backup erfolgreich beendet.
20240812-022616 DBG 4455:          --- ColoringOption: 
20240812-022616 DBG 4456:          --- ContentType: 
20240812-022616 DBG 4457:          --- Parms: 
20240812-022617 DBG 4462:          --- mail  -s "@HOSTNAME@: Backup erfolgreich beendet." k@@@@e(21)
nullmailer-inject: Invalid header line:
  To: , k@@@@e(21)
  Unable to parse the addresses.
Can't send mail: sendmail process failed with error code 1
20240812-022617 DBG 4465:          --- mail: RC: 1
20240812-022617 MSG 4504:          !!! RBK0197W: eMail mit mail versenden endet fehlerhaft mit RC 1. 
20240812-022617 DBG 4508:      <-- sendEMail 
```

E-Mail-Versand funktioniert:
```
20240817-021220 DBG 4386:      --> sendEMail 
20240817-021220 DBG 4429:          --- Coloring option: CM
20240817-021220 DBG 4429:          --- eMailColoring: OPTION
20240817-021220 DBG 4429:          --- subject: @HOSTNAME@: Backup erfolgreich beendet.
20240817-021220 DBG 4429:          --- coloring: -a Content-Type: text/html;
20240817-021220 MSG 4435:          --- RBK0167I: Eine eMail wird versendet. 
20240817-021220 DBG 4452:          --- eMail: k@@@@e(21)
20240817-021220 DBG 4453:          --- eMail Program: mail
20240817-021220 DBG 4454:          --- Subject: @HOSTNAME@: Backup erfolgreich beendet.
20240817-021220 DBG 4455:          --- ColoringOption: -a Content-Type: text/html;
20240817-021220 DBG 4456:          --- ContentType: 
20240817-021220 DBG 4457:          --- Parms: 
20240817-021220 DBG 4462:          --- mail -a Content-Type: text/html; -s "@HOSTNAME@: Backup erfolgreich beendet." k@@@@e(21)
20240817-021220 DBG 4465:          --- mail: RC: 0
20240817-021220 DBG 4508:      <-- sendEMail 

```

framps, 2024-08-17:
Vielen Dank fuer den Debuglogauszug. Leider fehlen mir da weiter Infos :cry: Wir koennen jetzt hier ein Frage- und Antwortspiel beginnen was ineffizient ist und ich - und wohl auch Du - nicht haben will.

Mein Vorschlag - sofern von Dir akzeptierbar: Schicke mir das vollstaendige Debuglog vom Failurefall an `framp at <meineHomepage>` und maskiere alle fuer Dich sensiblen Infos - oder loesche die Teile einfach - entsprechend gekennzeichnet.

> Ich denke nicht, dass sich derlei mit vernünftigem Aufwand automatisiert maskieren lässt - es geht um Einträge in der fstab bzw. Zeilen in der Auflistung von mounts. Da stehen zwar auch bei mir keine credentials drin, aber des einen Paranoia resultierte hier wohl in des anderen Overengineering. ;-)

Es gibt viele Nutzer von raspiBackup die nutzen eine Standardkonfigurtion des OS (fstab etc). Fuer die versuche ich moeglichst alles zu maskieren denn das Debuglog ist die einzige Moeglichkeit fuer mich bei Problemen zu helfen und sollten moeglichst keine sensitiven Infos beinhalten und diese providen das Debuglog ohne gross reinzusehen.

Bei Dir sieht es anders aus und eine gesunde Paranoia ist bzgl sensitiver Infos immer gut :+1:. 

Meine Bestrebungen moeglichst alles soweit Moegliche zu maskieren zu versuchen als Overengineering zu bezeichnen finde ich allerdings etwas unpassend :cry:

framps, 2024-08-20:
Vielen Dank fuer die Debug Logs. Leider haben sie mir nicht geholfen. Deshalb habe ich ein System mit Nullmailer aufgesetzt und weder mit C und CM und SUBJECT bekomme ich einen Fehler. Es liegt offensichtlich am Nullmailer.

```
pi@raspberrypi-bookworm-64-lite-cm4-nvme:~ $ apt-cache policy nullmailer
nullmailer:
  Installed: 1:2.2-4
  Candidate: 1:2.2-4
  Version table:
 *** 1:2.2-4 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
```
Mir ist aufgefallen dass Du Bullseye nutzt. Ich gehe davon aus dass da eine aeltere Nullmailer Version aktiv ist in der irgendwann mal durch einen Update dieser Fehler reingekommen ist und in der aktuellen Version wieder gefixed wurde.

aki42, 2024-09-04:
Hinsichtlich der nullmailer-Version:
```
root@nas:~ # apt-cache policy nullmailer
nullmailer:
  Installiert:           1:2.2-3
  Installationskandidat: 1:2.2-3
  Versionstabelle:
 *** 1:2.2-3 500
        500 http://deb.debian.org/debian bullseye/main arm64 Packages
        100 /var/lib/dpkg/status
```

Eventuell liegt auch es an einem Detail meiner spezifischen Nullmailer-Konfiguration. Es ehrt Dich, dass Du ins Debugging gegangen bist, aber es gibt ja zumindest im Kontext von raspiBackup problemfreie(re), offiziell unterstützte Alternativen zum Nullmailer. Daher wollte ich sowieso eigentlich nur für die, die über mein Problem stolpern und im Netz nach Infos dazu suchen, dokumentieren, welcher Workaround zumindest auf meinem System funktioniert. :)

framps, 2024-09-05:
> Daher wollte ich sowieso eigentlich nur für die, die über mein Problem stolpern und im Netz nach Infos dazu suchen, dokumentieren, welcher Workaround zumindest auf meinem System funktioniert. :)

Vielen Dank.

Ich habe auch noch [fuer mich dokumentiert](https://www.linux-tips-and-tricks.de/de/raspibackupcategoried/675-nullmailer-konfigurieren#google_vignette) wie ich den Nullmailer fuer meinen Provider aufgesetzt habe. Das hilft vielleicht auch Leuten die den Nullmailer mit raspiBackup nutzen wollen. Allerdings ist der Nullmailer weiterhin nicht offiziell supported. Aber er funktioniert durchaus :wink:

closed 2024-10-08 by github-actions[bot]