framps/raspiBackup #243: v0.6.5.1: Restore fails " RBK0253E: Target device /dev/sdb too small. Available bytes: .

platgeo, 2020-06-08:
With raspiBackup v0.6.5.1 it is not possible anymore to restore images with the -d option. Following error occurs:

`[13:47:16] openhabian@openhab:~$ sudo raspiBackup.sh -d /dev/sdb /misc/amanda/pi4b/openhab/openhab-ddz-backup-20200607-230246
[sudo] password for openhabian:
--- RBK0009I: openhab: raspiBackup.sh V0.6.5.1 (d66749e) started at Mon 08 Jun 2020 01:48:17 PM CEST.
--- RBK0229I: Telegram notified.
--- RBK0128I: Using logfile /home/openhabian/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0138I: Using bootbackup /misc/amanda/pi4b/openhab/openhab-ddz-backup-20200607-230246/openhab-ddz-backup-20200607-230246.img.gz.
--- RBK0068I: Using bootpartition backup files starting with /misc/amanda/pi4b/openhab/openhab-ddz-backup-20200607-230246 from directory openhab-backup.
gzip: /misc/amanda/pi4b/openhab/openhab-ddz-backup-20200607-230246 is a directory -- ignored
??? RBK0253E: Target device /dev/sdb too small. Available bytes: . Required bytes: 7231715361.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0077E: Restore failed with RC 116. Check previous error messages.
--- RBK0010I: openhab: raspiBackup.sh V0.6.5.1 (d66749e) stopped at Mon 08 Jun 2020 01:48:20 PM CEST with rc 116.
??? RBK0005E: Backup failed. Check previous error messages for details.
--- RBK0229I: Telegram notified.
`

Switching version back to v0.6.5.0 everything works fine:

`[14:01:27] openhabian@openhab:~$ sudo raspiBackup.sh -d /dev/sdb /misc/amanda/pi4b/openhab/openhab-ddz-backup-20200607-230246
--- RBK0009I: openhab: raspiBackup.sh V0.6.5 (f822b75) started at Mon 08 Jun 2020 02:01:44 PM CEST.
--- RBK0229I: Telegram notified.
--- RBK0128I: Using logfile /home/openhabian/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0138I: Using bootbackup /misc/amanda/pi4b/openhab/openhab-ddz-backup-20200607-230246/openhab-ddz-backup-20200607-230246.img.gz.
--- RBK0068I: Using bootpartition backup files starting with /misc/amanda/pi4b/openhab/openhab-ddz-backup-20200607-230246 from directory openhab-backup.
!!! RBK0065W: Device /dev/sdb will be repartitioned and all data will be lost.
--- RBK0067I: Current partitions on /dev/sdb:
Number  Start   End      Size     Type     File system  Flags
 1      4.19MB  273MB    268MB    primary  fat32        lba
 2      277MB   14957MB  14680MB  primary  ext4
--- RBK0066I: Device /dev/sdb will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/sdb1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/sdb2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N
y
--- RBK0050I: Restoring backup from /misc/amanda/pi4b/openhab/openhab-ddz-backup-20200607-230246.
0+446630 records in
0+446630 records out
14956888064 bytes (15 GB, 14 GiB) copied, 332.117 s, 45.0 MB/s
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0076I: Restore finished successfully.
--- RBK0010I: openhab: raspiBackup.sh V0.6.5 (f822b75) stopped at Mon 08 Jun 2020 02:07:25 PM CEST with rc 0.
--- RBK0017I: Backup finished successfully.
--- RBK0229I: Telegram notified.
`
The problem is reproducable on a Pi3b running Raspbian GNU/Linux 9 (stretch) as well as on a Raspberry 4b running Raspbian GNU/Linux 10 (buster).

Also, the problem is occuring on different storage devices. I tried to restore on a 500GB HDD as well as on a 32GB SD in a card reader. On both devices, restore worked with version V0.6.5.0.

* [x] I attached raspiBackup debug log
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4746116/raspiBackup.log)

framps, 2020-06-08:
Thank you very much for the log. 

I thought it's a good idea to check whether the size of the dd(z) image will fit on the -d device - in particular for people who use a sd card for -d. Unfortunately the check is buggy for ddz images :cry: .

I just removed the buggy code I added to test this and will add this test in an upcomming release - and test it much more carefully :wink: . Just upgrade again to 0.6.5.1 and you will be able to restore again. Please close the issue then.

Note: People who already upgraded to 0.6.5.1 with this bug can just use options `-U -S` to grab the latest 0.6.5.1 code which doesn't include the test code any more.

platgeo, 2020-06-08:
Thanks for your prompt reply, and indeed you are right, I installed 0.6.5.1 again (first switch back with -V and then -U also worked) and with the "fresh" code I can restore without error.

closed 2020-06-08 by platgeo