framps/raspiBackup #244: New user questions and suggestions

cbhushan, 2020-06-13:
If you have a question about raspiBackup please 

* [x] Read the raspiBackup FAQ's which answer a lot of common questions [English](https://www.linux-tips-and-tricks.de/en/faq) - [German](https://www.linux-tips-and-tricks.de/de/faq)

* [x] Provide first line of output of `sudo raspiBackup.sh --version` or `sudo raspiBackupInstallUI.sh --version`
```
# raspiBackup.sh --version
Version: 0.6.5.1 CommitSHA: a056338 CommitDate: 2020-06-08 CommitTime: 15:28:45
```

## Questions:

- How to use smart-recycle? I am using rsync backup mode and I have `DEFAULT_SMART_RECYCLE=1` with `DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 0"` in conf file. After this **nothing** seems to be getting backed up!! All new folders have only two raspiBackup.log and raspiBackup.msg. The [smart-recyle](https://www.linux-tips-and-tricks.de/en/smart-recycle/) page does explain the concept but does not really explain how to actually use it.

- How to email user@localhost? The interface keep telling me its invalid email, which is not true. Other local applications can send email without any issue.

- What does the Standardplugins "temp mem disk execute" do? The [help page](https://www.linux-tips-and-tricks.de/en/13-raspberry/443-raspibackup-extensions/) only lists them and do not explain what they do. Having even one-line option will help a lot.

- Can raspiBackup still be used without `raspiBackupInstallUI.sh` interface? I am happy to modify the config files and cron.d myself. No offence, but it feels like UI is not aware of all possible config options.

- Conceptually raspiBackup is similar to [rsnapshot](https://github.com/rsnapshot/rsnapshot) and [timeshift](https://github.com/teejee2008/timeshift). It might be useful to explain how raspiBackup is different (& similar) to them and how raspiBackup is a better raspberry option.

- Also, not everyone may be willing or allowed to install raspiBackup as root on another system (like laptop) for the restore option. Can `raspiBackup.sh` work (for backup/restore) by putting it in a user directory? If so, where to download the working `raspiBackup.sh`?

## Suggestions:
Feel free to ignore, it devs disagree with these: 

- FAQ page mentions that "... it should be possible to restore the backup with Linux means only..." but the (pseudo) steps are not listed anywhere. It might be helpful and more encouraging if it is clear how to do it on your own. 

- It is bit unclear how "restore" and "backup" syntax are different? [Restore](https://www.linux-tips-and-tricks.de/en/restore/) and [backup](https://www.linux-tips-and-tricks.de/en/backup/) page literally list same syntax. It might be better to have an explicit flag, say `--restore`, to indicate raspiBackup is running in restore mode.
  - The restore page might need some clear instructions on how to restore from different possible options of backup-mode. There are no instructions for Linuxusers listed on the [restore page](https://www.linux-tips-and-tricks.de/en/restore/).

- The documentation and download options are very intertwined to website www.linux-tips-and-tricks.de. There is nothing wrong with it. But somehow having everything, specially downloads, on Github feels more clear, transparent and reliable (in case the website goes down in future). May be `raspiBackup.sh` can be released/linked as downloadable file as part of Github release page.

framps, 2020-06-13:
>     * The [smart-recyle](https://www.linux-tips-and-tricks.de/en/smart-recycle/) page does explain the concept but does not really explain how to actually use it.

Die you read `Smart recylce backup strategy - Options` on this page? There option `--smartRecycleDryrun` is explained :wink: 

>     * How to email user@localhost? The interface keep telling me its invalid email, which is not true. Other local applications can send email without any issue.

I need more details about the issue. Please provide the debug log.

>     * What does the Standardplugins "temp mem disk execute" do? The [help page](https://www.linux-tips-and-tricks.de/en/13-raspberry/443-raspibackup-extensions/) only lists them and do not explain what they do. Having even one-line option will help a lot.

Agree. I will add a sentence which will describe their function.

>     * Can raspiBackup still be used without `raspiBackupInstallUI.sh` interface? I am happy to modify the config files and cron.d myself. No offence, but it feels like UI is not aware of all possible config options.

Sure. Just [read this page](https://www.linux-tips-and-tricks.de/en/all-pages-about-raspibackup/539-raspibackup-manual-installation/)

>     * Conceptually raspiBackup is similar to [rsnapshot](https://github.com/rsnapshot/rsnapshot) and [timeshift](https://github.com/teejee2008/timeshift). It might be useful to explain how raspiBackup is different (& similar) to them and how raspiBackup is a better raspberry option.

Pleas read [FAQ0](https://www.linux-tips-and-tricks.de/en/faq/#a0) about the history of raspiBackup and then read the features of raspiBackup listed [on this page](https://www.linux-tips-and-tricks.de/en/backup/) . 

>     * Also, not everyone may be willing or allowed to install raspiBackup as root on another system (like laptop) for the restore option. Can `raspiBackup.sh` work (for backup/restore) by putting it in a user directory? If so, where to download the working `raspiBackup.sh`?

Sure. Because raspiBackup requires root access it's installed in /usr/local/bin. But you can place it anywhere. Just follow the manual installation steps and use a different target directory.

>     * FAQ page mentions that "... it should be possible to restore the backup with Linux means only..." but the (pseudo) steps are not listed anywhere. It might be helpful and more encouraging if it is clear how to do it on your own.

There are so many Linux distros and Releases out there and I cannot test all of them. Just follow the instructions how to use raspiBackup on the Raspberry and try it out on your Linux box. In general it should work but may fail if some older Linux tool versions are on your Linux box. raspiBackup works on Debian Wheezy, Jessie and Buster.

> * It is bit unclear how "restore" and "backup" syntax are different? [Restore](https://www.linux-tips-and-tricks.de/en/restore/) and [backup](https://www.linux-tips-and-tricks.de/en/backup/) page [...]

Flag for restore is `-d` which defines the restore device. 

>     * The documentation and download options are very intertwined to website [www.linux-tips-and-tricks.de](http://www.linux-tips-and-tricks.de). There is nothing wrong with it. But somehow having everything, specially downloads, on Github feels more clear, transparent and reliable (in case the website goes down in future). May be `raspiBackup.sh` can be released/linked as downloadable file as part of Github release page. Also, this may allow github users to contribute to documentations.

As I wrote in [FAQ0](https://www.linux-tips-and-tricks.de/en/faq/#a0) raspiBackup started as a tiny backup program outside of github and evolved over the years and got all the current features step by step. That's why I added all doc on my website step by step. It's a bunch of doc (there are raspiBackup users out there which complain about the size of doc already :cry: ). I see your point it would help to have the doc in github and users can help in documenting raspiBackup. But this will require

1. Transfer of all the current doc into github which is too much effort for me
2. Frankly somebody experienced with information development should consolidate all the docs.

cbhushan, 2020-06-16:
Thank you for you response. I managed to get things working. 

> I need more details about the issue. Please provide the debug log.

The email issue was in the UI. I don't think I have logs - I ended up removing a bunch for raspiBackup files before manual install. Nonetheless emails are working after I modified the conf file manually.

> There are so many Linux distros and Releases out there and I cannot test all of them. Just follow the instructions how to use raspiBackup on the Raspberry and try it out on your Linux box. In [...]

May be my question was not clear. I meant to ask what are the steps to restore the rsync and tgz backup **without** using raspiBackup. May the answer is to go through the source code :smile:). But still some rough conceptual explantation of a restore process works would be useful in [Backupspace directory structure section](https://www.linux-tips-and-tricks.de/en/backup/#directory).

cbhushan, 2020-06-16:
Some additional questions:

- How does DEFAULT_KEEPBACKUPS and DEFAULT_SMART_RECYCLE interact with each other. Say DEFAULT_KEEPBACKUPS is set to 2, would that override DEFAULT_SMART_RECYCLE specifications? 

- Is it advisable to keep more than one type of backup? Like rsync + tgz?

framps, 2020-06-16:
> The email issue was in the UI. I don't think I have logs

:cry: Would be great if you remember what the issue was so I can fix it. 

> I meant to ask what are the steps to restore the rsync and tgz backup **without** using raspiBackup.

You don't have to read the source code :laughing: . A raspiBackup user had the same question some time ago and I helped him to get it done. He finally created a step by step instructions I published [here](https://www.linux-tips-and-tricks.de/en/all-pages-about-raspibackup/498-raspibackup-manual-restore/). 

>     * How does DEFAULT_KEEPBACKUPS and DEFAULT_SMART_RECYCLE interact with each other.

The KEEP options are ignored as soon as you use SMART_RECYCLE. I will add a note for these options in the users guide.

>     * s it advisable to keep more than one type of backup? Like rsync + tgz?

Well, I don't think so. But there are raspiBackup users which have a very sophisticated backup strategy and use different backup types.

cbhushan, 2020-06-27:
Thanks for your help. For me, it seems things are working okay. :smiley: 

I have one small suggestion/feature request: For tar/tgz (& may be even dd) backup, it may be useful to have some sort of hash saved along with backup files. Something simple like md5 or sha-256 should be enough. I can help to quickly verify that backup files are not corrupted.

framps, 2020-06-27:
> Thanks for your help. For me, it seems things are working okay

Glad raspiBackup works now for you :+1: 

> it may be useful to have some sort of hash saved along with backup files.

I frankly don't understand why you need this hash. A backup created by raspiBackup is consistent. Can you give me a use case when you need this hash to verify the back?

framps, 2020-06-29:
Please close this issue if you don't have any updates.

closed 2020-07-11 by framps