framps/raspiBackup #245: Partitionsdaten können nicht mit sfdisk gesammelt werden

ChrizZz90, 2020-06-22:
Nachdem ich letztens iobroker installiert habe, ist mein automatisches Backup letzte Nacht fehlgeschlagen. Das log sah so aus:

`20200622-010113 MSG 3833:              --- RBK0092I: Partition mmcblk0p7 (61,4GB) wird gesichert ...
20200622-010113 DBG 3231:                 -> rsyncBackup
20200622-010113 DBG 3237:                 -- ls raspberrypi-rsync-backup-20200608-010003
20200622-010113 DBG 3237:                 -- raspberrypi-rsync-backup-20200615-010009
20200622-010113 DBG 3237:                 -- raspberrypi-rsync-backup-20200622-010004
20200622-010113 DBG 3258:                 -- LastBackupDir: /backup/raspberrypi/raspberrypi-rsync-backup-20200615-010009
20200622-010113 DBG 3265:                 -- LinkDest: --link-dest="/backup/raspberrypi/raspberrypi-rsync-backup-20200615-010009"
20200622-010113 MSG 3268:                 --- RBK0133I: Verzeichnis /backup/raspberrypi/raspberrypi-rsync-backup-20200615-010009 wird für Hardlinks benutzt.
20200622-010113 MSG 3271:                 --- RBK0158I: rsync Backup "/backup/raspberrypi/raspberrypi-rsync-backup-20200622-010004" wird erstellt.
20200622-010113 MSG 3305:                 --- RBK0085I: Backuperstellung vom Typ rsync gestartet. Bitte Geduld.
20200622-010113 DBG 1188:                    -> executeCmd Command: rsync --exclude="/backup"                   --exclude="/mmcblk0p7/backup/raspberrypi/raspberrypi-rsync-backup-20200622-010004/raspiBackup.log"                      --exc$
20200622-010113 DBG 1189:                    -- Redirect: & - Skips:
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/install, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/install/windows, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/backup-objects, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/admin.admin, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/discovery.admin, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin/lib, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin/lib/css, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin/lib/data, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin/lib/fonts, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin/lib/img, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin/lib/img/clock, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin/lib/img/icons, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin/lib/img/logos, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/info.admin/lib/js, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/mihome.admin, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/iobroker-data/files/mihome.admin/icons, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/lib, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/log, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/log/iobroker..2020-06-Sa.6, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/log/iobroker..2020-06-Sa.6/20, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/node_modules, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/.npmrc.Zu8VsZ, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/.CHANGELOG_FIXER_LINUX.md.yLLUJs, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/.CHANGELOG_INSTALLER_LINUX.md.pps9vX, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/.INSTALLER_INFO.txt.cXolUu, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/.LICENSE.ZWDbs5, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/.README.md.mx4glF, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/.fix_installation.sh.P45WSh, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/.iobroker.eAxmJZ, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/node_modules/.bin, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/node_modules/@iobroker, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/node_modules/@iobroker/adapter-core, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/node_modules/@iobroker/adapter-core/build, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/node_modules/@iobroker/plugin-base, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/node_modules/@iobroker/plugin-base/lib, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/node_modules/@iobroker/plugin-sentry, ACL_TYPE_ACCESS): Operation not supported (95)
rsync: set_acl: sys_acl_set_file(mmcblk0p7/opt/iobroker/node_modules/@root, ACL_TYPE_ACCESS): Operation not supported (95)
...`

Daraufhin habe ich in der Konfiguration den iobroker ergänzt und vor Beginn des Backups gestoppt, um das Tool anschließend wieder zu starten:
`# Durch && getrennte Befehle, die vor dem Starten des Backups auszuführen sind
DEFAULT_STOPSERVICES="systemctl stop pm2-pi.service && systemctl stop cron && systemctl stop iobroker"

# Durch && getrennte Befehle, die nach dem Starten des Backups auszuführen sind
DEFAULT_STARTSERVICES="systemctl start cron && systemctl start pm2-pi.service && systemctl start iobroker"`

Doch anschließend im Backup Testlauf erhalte ich nun folgenden Fehler: **??? RBK0110E: Partitionsdaten können nicht mit sfdisk gesammelt werden. RC 1.**

So sieht der Auszug aus dem Log aus:
`--- RBK0044I: Backup des Partitionlayouts wird in /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20200622-141143/@HOSTNAME@-backup.sfdisk erstellt.
/usr/local/bin/raspiBackup.sh: Zeile 3791: /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20200622-141143/@HOSTNAME@-backup.sfdisk: Datei oder Verzeichnis nicht gefunden
20200622-141159 DBG 2862:                          --> colorAnnotation 1
20200622-141159 DBG 2886:                          <-- colorAnnotation 
??? RBK0110E: Partitionsdaten können nicht mit sfdisk gesammelt werden. RC 1.
20200622-141159 DBG 1406:                          --> exitError 111
20200622-141159 DBG 1413:                          <-- exitError 111
20200622-141159 DBG 3308:                          --> cleanup`

Ich check leider nicht woran es hängt :/

framps, 2020-06-22:
Ich bitte jeden bevor ein Issue erstellt wird die FAQs zu lesen und das auch zu bestaetigen :disappointed: 

Lies bitte [FAQ24](https://www.linux-tips-and-tricks.de/de/faq/#a24) zum Thema `Operation not supported`. Zum Thema RBK0110E: Dazu brauche ich das Debuglog.

ChrizZz90, 2020-06-22:
> 
> 
> Ich bitte jeden bevor ein Issue erstellt wird die FAQs zu lesen und das auch zu bestaetigen 😞

Sorry, shame on me. Habe nach dem Ursprungsfehler gar nicht geschaut, dachte das hätte ich mit dem Stop des ioBroker gelöst.

> Zum Thema RBK0110E: Dazu brauche ich das Debuglog.

Den lade ich hiermit hoch.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4815383/raspiBackup.log)

framps, 2020-06-22:
```
20200622-215917 DBG 1598:      --- Options:  -P -F
```
Du rufst raspiBackup im Fake Mode auf. So wie es aussieht gibt es mit dem partitionsorientierten Modus mit dem Fake Mode ein Problem dass das Backupverzeichnis nicht erstellt wurde in welches eine Datei geschrieben werden soll.

Lass mal das `-F` weg :wink:

ChrizZz90, 2020-06-23:
oh man -__- na das war ja "einfach". 

Ohne -F erhalte ich aber nun erneut einen Fehler: _??? RBK0213E: Mount von /dev/mmcblk0p2 an /tmp/mmcblk0p2 ist fehlerhaft.RBK0213E: Mount von 32 an ist fehlerhaft._

Vielleicht muss ich dazu sagen, dass ich durch den initialen Fehler mich mit dem Backupthema beschäftigt habe und gleich auch das Update auf die neue Version gemacht. Da es ja scheinbar ein mount Problem gibt noch der Hinweis, dass ich mein Synology NAS einbinde, das Backup laufen lasse und anschließend per umount die Verbindung zum NAS wieder abbreche. Daran habe ich bisher nichts geändert, da letzte Woche noch alles in Ordnung war.

Angehängt erneut das Log.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4817587/raspiBackup.log)

Die FAQs habe ich diesmal auch überflogen und nur den Hinweis entdeckt, dass ein Linux Befehl nicht durchgeführt werden konnte. Kurzes googlen hat eher mehr als weniger Fragen verursacht...

framps, 2020-06-23:
Im Log steht
```
mount: /tmp/mmcblk0p2: can't read superblock on /dev/mmcblk0p2.
```.
D.h. Deine zweite Partition ist irgendwie nicht OK.

> Vielleicht muss ich dazu sagen, dass ich durch den initialen Fehler mich mit dem Backupthema beschäftigt habe und gleich auch das Update auf die neue Version gemacht.

Dann wuerde ich mit der Option -V wieder auf die vorherige Version zurueckgehen um zu testen ob es an der neuen Version liegt :wink: .

ChrizZz90, 2020-06-23:
jetzt wird es strange (zumindest für mich als Anfänger).

Mit der Version 0.6.4.3 kommt der Fehler nicht. Dafür wieder der Fehler mit "ACL_TYPE_ACCESS, Operation not supported", obwohl ich in der /etc/mke2fs.conf acl beim Attribut default_mntopts gelöscht habe.

Ich habe in meiner Config Datei das Attribut DEFAULT_RSYNC_BACKUP_OPTIONS nicht. Ist das optional und kann einfach hinzugefügt werden?

framps, 2020-06-24:
Du kannst 
```
DEFAULT_RSYNC_BACKUP_OPTIONS="-aH"
```
ohne `xA` in die Config aufnehmen. Dann werden keine xattrs und ACLs gesichert. Du solltest dann aber das Backup gut testen ob es auch ohne ACLs unx xattrs funktioniert.

ChrizZz90, 2020-06-24:
also nun funktioniert es wieder mit der Kombination Version 0.6.4.3, default_mntopts in der mke2fs.conf ohne acl sowie DEFAULT_RSYNC_BACKUP_OPTIONS="-aHx --delete"

Wo ist denn der Unterschied zwischen "-aH" und "-aHx --delete"?
Ich werde es morgen nochmal mit der aktuellen Version probieren, überraschend für mich war aber, dass die Anpassung in der mke2fs.conf alleine nichts gebracht hat.

framps, 2020-06-25:
> Wo ist denn der Unterschied zwischen "-aH" und "-aHx --delete"?

`--delete` bewirkt dass geloeschte Dateien im Backup nicht mehr per Hardlinks verlinked werden. Du darfst diese Option aber nur mit der Version 0.6.5 benutzen. Bei 0.6.4.3 werden auch wichtige Dateien im Backup geloescht - d.h. das Backup ist unbrauchbar :warning: 

D.h. wenn Du jetzt wieder die 0.6.4.3 benutzt loesche die `--delete` Option.

Wg des Fehlers RBK0213E: In der Version 0.6.5 wird nach jedem Mount geprueft ob der Mount OK war um raspiBackup robuster zu machen. Offensichtlich tritt der Fehler auch bei der 0.6.4.3 auf - fuehrt aber bei Deinem NOOBs irgendwie nicht zu einem Backupfehler.

ChrizZz90, 2020-06-26:
alles klar, gestern habe ich dann noch ein Backup mit 0.6.4.3 und ohne die Option `--delete` durchgeführt. Soweit funktioniert alles.

Hast du noch eine Idee oder einen Tipp wie ich das Backup vllt. doch unter 0.6.5 zum Laufen bekomme? Ich kann auch mit der Version 0.6.4.3 leben...

framps, 2020-06-26:
Das Problem ist dass die Partition 2 gemaess [hier](https://github.com/raspberrypi/noobs/wiki/NOOBS-partitioning-explained)  eine extended Partition ist die man nicht mounten kann. Du kannst die Partitionen die gesichert werden sollen mit der Option `-T` bzw der entsprechenden Config Option definieren. Ich denke wenn Du diese beim Backup ausnimmst sollte auch die 0.6.5.1 bei Dir funktionieren :wink:

ChrizZz90, 2020-06-28:
Super, vielen vielen Dank für die Hilfe! Wenn ich die Partion 2 raus lasse kann ich auch die neuste Version nutzen.

closed 2020-06-28 by ChrizZz90