framps/raspiBackup #246: Is raspiBackup likely to work under CentOS 7 on a RPi3B+?

smoore-33, 2020-06-30:
I use both Raspbian and CentOS and it would be helpful to use a backup solution that would be the same on both. Thanks.

framps, 2020-06-30:
I see your point but as of now I have a test regression suite for Raspbian to have confidence the backup and restore works in any new release and I can publish the new release without worries. raspiBackup has to be reliable. This regression test suite took me multiple months to implement ...

Any other OS supported by raspiBackup will require another testsuite for every OS and this also will require additional test HW. I don't have any sponsor :cry: 

raspiBackup was created by me in my spare time the last couple of years and I consider raspiBackup to be mature now. I don't plan to invest any effort to support other OS. It's too time consuming. Keep in mind: raspiBackup is open source and available for free and everything I invested in raspiBackup was for free.

smoore-33, 2020-06-30:
Thanks for your reply. It would not require additional hardware, other than another SD card, but I understand that testing would require more time. I wouldn't expect you to do that. On an untested basis, would you speculate that it **might** work.

If the whole thing is still in bash, I might try to learn bash and see if I can use parts of it. It would not be feasible for me to try to learn go or any other language.

Thanks so much for all your work.

framps, 2020-06-30:
> On an untested basis, would you speculate that it **might** work.

I have no idea. Just give it a try. You just need another SD card and some backupspace :wink:

framps, 2020-07-11:
Any update? Otherwise I will close the issue in 1 week because of inactivity.

closed 2020-07-11 by framps