framps/raspiBackup #248: Updating raspiBackupInstallUI.sh repaces the script with an HTML file

realrosid, 2020-07-06:
I updated raspiBackupInstallUI.sh 0.4.1 -> 0.4.3.2, and the contents of raspiBackupInstallUI.sh were replaced with HTML code (see attachment).

* [x] I checked whether my message was listed on the messages page and the suggested action fixes your issue? [English](https://linux-tips-and-tricks.de/en/rmessages) - [German](https://linux-tips-and-tricks.de/de/fehlermeldungen)

* [x ] I provided first line from output of `sudo raspiBackup.sh --version` or `sudo raspiBackupInstallUI.sh -h`
raspiBackupInstallUI.sh 0.4.1, 2019-04-21/19:41:04 - afd65f7

* [x ] I provided the output of `cat /etc/os-release`
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

[raspiBackupInstallUI.sh.txt](https://github.com/framps/raspiBackup/files/4879836/raspiBackupInstallUI.sh.txt)

framps, 2020-07-06:
That's ugly. Somehow the landing page of linux-tips-and-tricks.de made it into raspiBackupInstallUI.sh :thinking: 

Which command did you execute on 0.4.1 to upgrade to 0.4.3.2? Are you sure you upgraded to 0.4.3.2? raspiBackup only upgrades to the latest available version which is 0.6.5.1 as of now.

realrosid, 2020-07-06:
I used the command available via GUI, see the attached Screenshot.
[image]

framps, 2020-07-06:
This screenshot is very helpful :wink: . I have to check the code now.

framps, 2020-07-06:
In version 0.4.1 there was an issue with the update functionality. You have to update raspiBackupInstallUI.sh manually and execute following steps:

```
curl -sLO https://www.linux-tips-and-tricks.de/raspiBackupInstallUI.sh
sudo mv raspiBackupInstallUI.sh /usr/local/bin
sudo chmod +x /usr/local/bin/raspiBackupInstallUI.sh
sudo chown root.root /usr/local/bin/raspiBackupInstallUI.sh
```
and now verify

```
sudo /usr/local/bin/raspiBackupInstallUI.sh -h
```

Sorry for the inconvenience.

realrosid, 2020-07-06:
It worked:
raspiBackupInstallUI.sh 0.4.3.2, 2020-06-01/22:01:13 - d66749e

Thanks a lot for the quick help!

framps, 2020-07-06:
:+1: Please close the issue if you are fine now.

closed 2020-07-07 by realrosid