framps/raspiBackup #249: RBK0110E: Unable to collect partition data with sfdisk. RC 1.

Help-i-am-lost, 2020-07-07:
Hi,

trying to figure out a backup, but failing with the error code in the title. I have (older version) of the script running on another raspberry and it's working okay with very similar setup to this. Would you be able to give me a clue of where to look for?

I'm trying to take tgz -backup and store that to USB -stick at /media/markku/USB Storage/HAL-9001/SD. From there a nextcloud -client would then upload it to the NAS I have. This setup is running on another system for me already and works okay for my purposes. But I cannot find any clue about that sfdisk -problem now?

Thanks in advance,
-Markku

```
The messages (note, there's a notion of raspiBackup.sh: line 3732 ambiguous redirect -thing...?

--- RBK0009I: HAL-9001: raspiBackup.sh V0.6.5.1 (a056338) started at Tue 07 Jul 2020 01:12:40 PM EEST.
--- RBK0128I: Using logfile /media/markku/USB Storage/HAL-9001/SD/HAL-9001/HAL-9001-tgz-backup-20200707-131239/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0151I: Using backuppath /media/markku/USB Storage/HAL-9001/SD.
!!! RBK0157W: No services to stop.
--- RBK0081I: Creating backup of type tgz in /media/markku/USB Storage/HAL-9001/SD/HAL-9001/HAL-9001-tgz-backup-20200707-131239.
--- RBK0036I: Saving partition layout.
--- RBK0044I: Creating backup of boot partition in /media/markku/USB Storage/HAL-9001/SD/HAL-9001/HAL-9001-tgz-backup-20200707-131239/HAL-9001-backup.img.
256+0 records in
256+0 records out
268435456 bytes (268 MB, 256 MiB) copied, 12.5374 s, 21.4 MB/s
--- RBK0045I: Creating backup of partition layout in /media/markku/USB Storage/HAL-9001/SD/HAL-9001/HAL-9001-tgz-backup-20200707-131239/HAL-9001-backup.sfdisk.
/usr/local/bin/raspiBackup.sh: line 3732: $LOG_FILE: ambiguous redirect
??? RBK0110E: Unable to collect partition data with sfdisk. RC 1.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0043I: Removing incomplete backup in /media/markku/USB Storage/HAL-9001/SD/HAL-9001/HAL-9001-tgz-backup-20200707-131239. This may take some time. Please be patient.
--- RBK0049I: Messages saved in /home/markku/raspiBackup.msg.
--- RBK0026I: Debug logfile saved in /home/markku/raspiBackup.log.
--- RBK0010I: HAL-9001: raspiBackup.sh V0.6.5.1 (a056338) stopped at Tue 07 Jul 2020 01:12:59 PM EEST with rc 111.
??? RBK0005E: Backup failed. Check previous error messages for details.
```

```
sudo raspiBackup.sh --version

#outputs
Version: 0.6.5.1 CommitSHA: a056338 CommitDate: 2020-06-08 CommitTime: 15:28:45
```

```
cat /etc/os-release

#outputs
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

[raspiBackup_log.txt](https://github.com/framps/raspiBackup/files/4884152/raspiBackup_log.txt)
[raspiBackup_msg.txt](https://github.com/framps/raspiBackup/files/4884153/raspiBackup_msg.txt)

Don't forget to select all items below when you finished your problem report to check and confirm you followed all steps.

* [x] I checked whether my message was listed on the messages page and the suggested action fixes your issue? [English](https://linux-tips-and-tricks.de/en/rmessages) - [German](https://linux-tips-and-tricks.de/de/fehlermeldungen)
* [x] I provided first line from output of `sudo raspiBackup.sh --version` or `sudo raspiBackupInstallUI.sh -h`
* [x] I provided the output of `cat /etc/os-release`
* [x] I attached raspiBackup debug log
* [x] Scripts from the helper directory are sample code and not part ot the raspiBackup package and thus have limited support. If scripts from the helper subdirectory are used mention their names and if there are major changes I posted the contents of the files.

framps, 2020-07-07:
It looks like he root cause for your issue is

1) you use a backup path which contains spaces `--- RBK0128I: Using logfile /media/markku/USB Storage/@HOSTNAME@/SD/@HOSTNAME@/@HOSTNAME@-tgz-backup-20200707-131239/raspiBackup.log.`
2) raspiBackup does not support spaces in pathnames
3) raspiBackup check for spaces in path names but this test fails for some reasons.

Please use a backup path with no spaces (use `-` or `_` for spaces) and try again.

Help-i-am-lost, 2020-07-07:
Hello,

and thank you. Yes, the space seemed to be the issue. I changed the "USB Storage" to "USB-Storage" and now it works. Thank you for your help :)

-Markku

closed 2020-07-07 by Help-i-am-lost