framps/raspiBackup #251: Can't exclude 3rd partition on SSD during backup.

k-alle, 2020-07-10:
Don't forget to select all items below when you finished your problem report to check and confirm you followed all steps.

* [x ] I checked whether my message was listed on the messages page and the suggested action fixes your issue? [English](https://linux-tips-and-tricks.de/en/rmessages) - [German](https://linux-tips-and-tricks.de/de/fehlermeldungen)
* [x ] I provided first line from output of `sudo raspiBackup.sh --version` or `sudo raspiBackupInstallUI.sh -h`
Version: 0.6.5.1 CommitSHA: a056338 CommitDate: 2020-06-08 CommitTime: 15:28:45
* [ x] I provided the output of `cat /etc/os-release`
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
* [x ] I attached raspiBackup debug log
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4903828/raspiBackup.log)

* [ x] Scripts from the helper directory are sample code and not part ot the raspiBackup package and thus have limited support. If scripts from the helper subdirectory are used mention their names and if there are major changes I posted the contents of the files.
None

I'm having some issues using rsync.

I've succesfully migrated from SD card to SSD which has 3 partitions.
Everything was working real good before migrating to SSD.
**sda1 -> /boot
sda2 -> /
sda3 -> /home**

*This manual command:*
 **sudo raspiBackup.sh -a : -o : -m detailed -P -T "1 2" -l debug**

*Exits with fault below:*
1 2
--- RBK0009I: raspberrypi: raspiBackup.sh V0.6.5.1 (a056338) started at Fri 10 Jul 16:39:11 CEST 2020.
--- RBK0128I: Using logfile /mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-163910/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
!!! RBK0236W: Partition oriented backup will not be maintained any more and disable somewhere in the future.
**??? RBK0211E: External partition PARTUUID=2fed7fee-01 mounted on /dev/sda1 will not be saved with option -P.
??? RBK0211E: External partition PARTUUID=2fed7fee-02 mounted on /dev/sda2 will not be saved with option -P.**
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0043I: Removing incomplete backup in /mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-163910. This may take some time. Please be patient.
--- RBK0049I: Messages saved in /home/@user/raspiBackup.msg.
--- RBK0026I: Debug logfile saved in /home/@user/raspiBackup.log.
--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.5.1 (a056338) stopped at Fri 10 Jul 16:39:14 CEST 2020 with rc 107.
**??? RBK0005E: Backup failed. Check previous error messages for details.**

framps, 2020-07-10:
@k-alle Would you please correct the link to the debug log? I cannot open the file.

k-alle, 2020-07-10:
I can open the file in the first post. 
How do I give you a new file?

framps, 2020-07-10:
Works now for me. It was a link to aws when I tried it a couple of minutes ago. Now it points ot github.

framps, 2020-07-10:
Until now nobody tried to create a -P backup when usb boot mode is used. There is a check in raspiBackup which refuses -P backup when an external (no sd card) partition is detected. 

I modified the code and now external partitions can be saved if all partitions are on the boot device, i.e. an external usb disk or ssd. I created a [branch](https://github.com/framps/raspiBackup/tree/issue_251_skip_partitions) with the fix. Please grab raspiBackup from there, test the fix and give feedback.

k-alle, 2020-07-10:
Cool, there's a first for everything. 

Sorry for being dumb, but how do I try this branch? 
Like how do I grab it from there?

framps, 2020-07-10:
NP. 

Click on my branch link above. Click on the link on file raspiBackup.sh. Then hoover over the raw button (top right), use left button `save link as` and download changed raspiBackup.sh. Then copy this file to your Raspberry (e.g. /home/pi) and execute `sudo bash ./raspiBackup.sh` to start raspiBackup.

That's the github way.

Another more convenient way is to execute 
`wget https://raw.githubusercontent.com/framps/raspiBackup/issue_251_skip_partitions/raspiBackup.sh`
on your Raspberry as user pi in his home directory and then `sudo bash ./raspiBackup.sh` to start raspiBackup.

k-alle, 2020-07-10:
OK. I hope I did it correct. 
I downloaded that branch as a zip and extracted raspiBackup.sh to /usr/local/bin and made a backup on the old one. 

Executed this command again: 
sudo raspiBackup.sh -a : -o : -m detailed -P -T "1 2" -l debug

And for a while it did its magic even though it threw som errors on chmod and chgrp which I believe is because I did not stop services. 
But now it looks like it is stuck.
Using htop in another window I can see that it is still on but no activity. 
See attached picture. 
[image]

This is the partially (?) populated folder:
mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-201056 $ ls -la
total 212
drwxr-xr-x  4 root root    4096 Jul 10 20:11 .
drwxrwxr-x  5 pi   users   4096 Jul 10 20:10 ..
-rw-r--r--  1 root root     290 Jul 10 20:11 raspberrypi-backup.blkid
-rw-r--r--  1 root root     546 Jul 10 20:11 raspberrypi-backup.fdisk
-rw-r--r--  1 root root     512 Jul 10 20:11 raspberrypi-backup.mbr
-rw-r--r--  1 root root     163 Jul 10 20:11 raspberrypi-backup.parted
-rw-r--r--  1 root root     266 Jul 10 20:11 raspberrypi-backup.sfdisk
-rw-r--r--  1 root root  173578 Jul 10 20:13 raspiBackup.log
-rw-r--r--  1 root root    2319 Jul 10 20:11 raspiBackup.msg
drwxr-xr-x  3 root root    4096 Jan  1  1970 sda1
drwxr-xr-x 25 root root    4096 Jul 10 20:10 sda2

framps, 2020-07-10:
You didn't get the error message any more  :wink: 

Just be patient. As far as I can see a backup is created (you have subdir sda1 and sda2). This will take some time.

k-alle, 2020-07-10:
True, that is working good now :)

It is created but I woul also say that it has hung now. 
The Pi has no CPU nor network usage since the last output on the screen. 
The server for the mounpoint is not seeing any activity either. 
A backup has never taken more than 8 min before and that was with a much slower SD card.

Attached is the debug log where it stopped.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4904860/raspiBackup.log)

framps, 2020-07-10:
```
sync: chown "/mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-201056/sda2/etc/pihole" failed: Invalid argument (22)
rsync: chgrp "/mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-201056/sda2/etc/.dhcpcd.conf.mgyTvj" failed: Invalid argument (22)
rsync: chgrp "/mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-201056/sda2/etc/.gshadow.4tAtUf" failed: Invalid argument (22)
rsync: chgrp "/mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-201056/sda2/etc/.gshadow-.c7AsE4" failed: Invalid argument (22)
```

There is an authorization issue. Not sure whether it's because you invoke raspiBackup not the usual way.

Please execute as user pi in his home directory
`chmod +x raspiBackup.sh`
`sudo raspiBackup.sh`

If you still get the chown error messages there is some authorization issue on your /dev/sda2 partition.

k-alle, 2020-07-10:
pi@raspberrypi:~ $ chmod +x ./raspiBackup.sh
chmod: cannot access './raspiBackup.sh': No such file or directory

framps, 2020-07-10:
I suggested to download raspiBackup to /home/pi. Looks like you downloaded it elsewhere. Just go into this directory and execute the commands.

k-alle, 2020-07-10:
Oh my bad. 
I did the wget to home directory now and performed your commands but it still freezes up at the same spot.

framps, 2020-07-10:
I think we're past the error message and now struggle with another issue you have now with rsync an authorizations ...

The log contains the rsync command executed by raspiBackup which creates the rsync error messages:
```
20200710-201105 DBG 1431:                          --> executeCmd Command: rsync --exclude="/mnt/backup" 			--exclude="/sda2/mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-201056/raspiBackup.log" 			--exclude="/sda2/mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-201056/raspiBackup.msg" 			--exclude='.gvfs' 			--exclude=/sda2/proc/* 			--exclude=/sda2/lost+found/* 			--exclude=/sda2/sys/* 			--exclude=/sda2/dev/* 			--exclude=/sda2/boot/* 			--exclude=/sda2/tmp/* 			--exclude=/sda2/run/* 			 			 			--link-dest="/mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200630-163053" 			--numeric-ids 			-aHAx --delete 			 			 			/tmp/sda2 			"/mnt/backup/raspberrypi/raspberrypi-rsync-backup-20200710-201056" 	
```

I suggest to just execute this command without raspiBackup and check what's going on and where the error messages come from.

k-alle, 2020-07-10:
Thank you for the suggestions framps but I found the issue. 
I had activated NFSv4 on my Synology which totaly cripples the permissions since it is half assed implemented by Synology. 
Once deactivated the backup finished without issues and I **think** the permissions was preserved.

--- RBK0010I: raspberrypi: raspiBackup.sh V0.6.5.1-hotfix (95a5800) stopped at Fri 10 Jul 22:43:09 CEST 2020 with rc 0.
--- RBK0017I: Backup finished successfully.

framps, 2020-07-10:
> Thank you for the suggestions framps but I found the issue.

:+1: 

> Once deactivated the backup finished without issues and I _think_ the permissions was preserved.

Please double check, restore the backup and test the restored image. But I'm very confident it will succeed if you don't get these chown error messages from rsync any more.

Please close this issue if you're fine now.

closed 2020-07-10 by k-alle