framps/raspiBackup #253: In case of backup failure write message where the logs will be kept

framps, 2020-07-12:
If a backup fails the backup dir will be deleted. Default log location is the backup dir. Then the logs will be deleted. But raspiBackup saves the logs in either /root or /home/$USER directory before deleting the backup directory.

Write a message where the logs will be saved.

Siggi0904, 2020-07-12:
Message will be written in log file:
07-12-2020 15:53:19 --- RBK0026I: Debug Logdatei wurde in /home/@USER@/raspiBackup.log gesichert.

If the job is done by cron, should be in root?
Or at all /home/user.

framps, 2020-10-03:
> If the job is done by cron, should be in root?
> Or at all /home/user.

Cron jobs run as root so the log will be saved in /root .

framps, 2020-10-11:
Already implemented.

closed 2020-10-11 by framps