framps/raspiBackup #254: Error at rsync backup at perl/5.28.1 files

Siggi0904, 2020-07-12:
Hi,

I've started today a rsync backup and got some errors at perl/5.28.1 corrupt files.
Backup should ignore corrupted files and get ahead with messages.

See log file:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4909094/raspiBackup.log)

framps, 2020-07-13:
The error messages are created by rsync and finally rsync finishes with a return code 23 which means some files were deleted during rsync backup. There exists a rsync option `--ignore-errors` which can be used but this option will ignore any errors but it's  strongly recommended **not** to use this option because then also any write errors will be ignored an an inconsistent backup will be created without any warning/error.

There exists an undocumented option which allows to ignore rsync RC 23. See [FAQ32](https://www.linux-tips-and-tricks.de/en/faq/#a32) for details. I frankly don't suggest to use this option and instead add an ignore statement to ignore the directory with the vanishing files or even better - to stop the service which deletes the files during backup.

Siggi0904, 2020-07-26:
thanks for explaining that point.
What can I do to fix that, or what information's you need for an updated version?

framps, 2020-07-26:
There is nothing raspiBackup can do for you. You have to decide which option described above you want to use:

1. Add `--ignore-errors` to rsync options used by raspiBackup
2. See FAQ32 how to ignore RC32 from rsync
3. Add an ignore statement for the directory which holds the vanishing files (use option `-u`)
4. Stop the service which cause the files to vanish (use option `-o`)

Siggi0904, 2020-08-02:
thanks for the list. This is very good for the work flow.

I'd like to add the -u option in the conf file.
How I can add this to DEFAULT_EXCLUDE_LIST in the conf file?
Is DEFAULT_EXCLUDE_LIST = "/usr/share/perl/5.28.1/Pod/*" correct?

How  can add more then one folder? Is it comma separated?

framps, 2020-08-02:
The exclude statement has to follow the exclude syntax of the used backup tool rsync or tar. See the man pages of the tools for details.

Your exclude statement should become `DEFAULT_EXCLUDE_LIST = "--exclude=/usr/share/perl/5.28.1/Pod/*" `. If you have multiple directories you want to exclude just add another `--exclude=<dir>` statement separated with a space from the previous exclude statement. See also the [users guide](https://www.linux-tips-and-tricks.de/en/backup/#parameters) for addtl description of option `-u`.

Siggi0904, 2020-08-02:
great thanks, looks like it works,
Yes, I can read the manual but it's easier to as you. :-D

framps, 2020-08-02:
> Yes, I can read the manual but it's easier to as you. :-D

Then please consider to [give a donation](https://www.linux-tips-and-tricks.de/en/donations/) ...

If you're fine now please close the issue.

Siggi0904, 2020-08-03:
I'll do and I'll do after final testing.

Siggi0904, 2020-08-09:
Donated per PayPal and topic closed.

closed 2020-08-09 by Siggi0904

framps, 2020-08-09:
Great raspiBackup works foir you and thank you very much for your donation :+1: