framps/raspiBackup #255: RBK0061E: Unable to find bootpartition files

longtom99, 2020-07-25:
When i want to restore a tar Backup i get following errors (On Raspberry Pi 4B with Ubuntu or also on a Ubuntu Laptop, all on 20.04).

??? RBK0061E: Unable to find bootpartition files /media/harald/d1c3db74-1ef4-4285-be64-aee23af23ea8/ubuntu/ubuntu-tar-backup-20200724-133655 starting with ubuntu-backup.
??? RBK0061E: Unable to find bootpartition files /media/harald/d1c3db74-1ef4-4285-be64-aee23af23ea8/ubuntu starting with ubuntu-backup-20200724-133655.
??? RBK0061E: Unable to find bootpartition files /media/harald/d1c3db74-1ef4-4285-be64-aee23af23ea8/ubuntu starting with ubuntu-backup.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0077E: Restore failed with RC 102. Check previous error messages.
--- RBK0010I: ubuntu: raspiBackup.sh V0.6.5.1 (a056338) stopped at Sa 25 Jul 2020 10:44:52 CEST with rc 102.
??? RBK0005E: Backup failed. Check previous error messages for details.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/4975133/raspiBackup.log)

framps, 2020-07-25:
There are important files missing which should have been created during backup, e.g. *.sfdisk, *.mbr, Unfortunately there is no `ls` debug statement which would proof this but given the error messages I'm sure they are missing.

Question now is why they are missing. Would you please also provide the debug log of the debug run? The log should be available in the backup directory `/media/harald/d1c3db74-1ef4-4285-be64-aee23af23ea8/@HOSTNAME@/@HOSTNAME@-tar-backup-20200724-133655/`

longtom99, 2020-07-25:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4975442/raspiBackup.log)

framps, 2020-07-25:
I found the root cause for the issue. It's a bug :disappointed: .

I created a [branch](https://github.com/framps/raspiBackup/tree/255_missing_sfdisk) with a hotfix for you. Would you please grab raspiBackup.sh from there and create another backup. Then please verify there is a .sfdisk and .mbr file in the backup directory first. Then test the restore now with this backup.

longtom99, 2020-07-25:
Sorry, the same with new file.
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4976338/raspiBackup.log)

framps, 2020-07-25:
`--- RBK0009I: @HOSTNAME@: raspiBackup.sh V0.6.5.1 (a056338) started at Sat Jul 25 21:44:39 CEST 2020.`

You used the old version of raspiBackup :cry: .

I'm sorry, there is no user friendly way to grab a hotfix from github. Maybe I should think about to add some code in raspiBackup which will make this more convenient...

Please execute following commands as user pi:

```
cd ~
wget https://raw.githubusercontent.com/framps/raspiBackup/255_missing_sfdisk/raspiBackup.sh
sudo bash ./raspiBackup.sh
```

longtom99, 2020-07-26:
Hmm, same again, am i too dumb for linux? ;-)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/4977232/raspiBackup.log)

framps, 2020-07-26:
Not sure what you mean with `same again`. I find in the log

```
--- RBK0009I: @HOSTNAME@: raspiBackup.sh V0.6.5.1-hotfix (a54de5d) started at Sun Jul 26 07:39:17 CEST 2020.
...
--- RBK0045I: Creating backup of partition layout in /backup/@HOSTNAME@/@HOSTNAME@-tar-backup-20200726-073916/@HOSTNAME@-backup.sfdisk.
--- RBK0046I: Creating backup of master boot record in /backup/@HOSTNAME@/@HOSTNAME@-tar-backup-20200726-073916/@HOSTNAME@-backup.mbr.
```

which means you use the hotfix I created and the missing files were now created. Please try a restore now.

longtom99, 2020-07-26:
This is why i wrote Restore is NOK.

ubuntu@ubuntu:~$ sudo bash ./raspiBackup.sh -d /dev/sdb /backup/ubuntu/ubuntu-tar-backup-20200726-073916/
--- RBK0009I: ubuntu: raspiBackup.sh V0.6.5.1-hotfix (a54de5d) started at Sun Jul 26 11:24:34 CEST 2020.
!!! RBK0173W: =========> NOTE  <=========
!!! RBK0173W: This is a temporary hotfix and has to be upgraded to next available version as soon as one is available.
!!! RBK0173W: =========> NOTE <=========
--- RBK0128I: Using logfile /home/ubuntu/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0138I: Using bootbackup /backup/ubuntu/ubuntu-tar-backup-20200726-073916/ubuntu-tar-backup-20200726-073916.tar.
??? RBK0061E: Unable to find bootpartition files /backup/ubuntu/ubuntu-tar-backup-20200726-073916 starting with ubuntu-backup.
??? RBK0061E: Unable to find bootpartition files /backup/ubuntu starting with ubuntu-backup-20200726-073916.
??? RBK0061E: Unable to find bootpartition files /backup/ubuntu starting with ubuntu-backup.
--- RBK0033I: Please wait until cleanup has finished.
??? RBK0077E: Restore failed with RC 102. Check previous error messages.
--- RBK0010I: ubuntu: raspiBackup.sh V0.6.5.1-hotfix (a54de5d) stopped at Sun Jul 26 11:24:36 CEST 2020 with rc 102.
??? RBK0005E: Backup failed. Check previous error messages for details.

[raspiBackup.log](https://github.com/framps/raspiBackup/files/4977378/raspiBackup.log)

framps, 2020-07-26:
I found your Ubuntu has following fstab and partitioning:
```
20200726-073918 DBG 1553:          *** cat /etc/fstab
20200726-073918 DBG 1553:              LABEL=writable	/	 ext4	defaults	0 0
20200726-073918 DBG 1553:              LABEL=system-boot       /boot/firmware  vfat    defaults        0       1
```
and
```
20200726-073922 DBG 4231:                          /dev/mmcblk0p1: LABEL_FATBOOT="system-boot" LABEL="system-boot" UUID="0468-A52F" TYPE="vfat" PARTUUID="87c6153d-01"
20200726-073922 DBG 4231:                          /dev/mmcblk0p2: LABEL="writable" UUID="5fcce78f-2de3-4805-8ffa-d0f11247d5bb" TYPE="ext4" PARTUUID="87c6153d-02"
```
raspiBackup requires either a /boot and a / partition or /boot and / on one partition. Your Ubuntu has /boot/firmware on /dev/mmcblk0p1 and / on /dev/mmcblk0p2 which doesn't meet the requirements of raspiBackup.

As far as i remember raspiBackup could be used to backup Ubuntu in the past even it's not officially supported. But it looks like they changed the partitioning in latest `Ubuntu 20.04.1 LTS` in such a way raspiBackup cannot handle.

If you want to use raspiBackup to backup your system I see only one option for you: You have to switch to officially supported Raspbian. Sorry I cannot offer you a better alternative.