framps/raspiBackup #261: use of second mmcblk (/dev/mmcblk1 instead of /dev/mmcblk0)

repohub, 2020-08-25:
Hello,

I'm using a laptop with an installed mmc and an internal cardreader. Therefore my sd card in the cardreader has the id /dev/mmcblk1 instead of /dev/mmcblk0. In the script line number 5709 and 5718 you check only for /dev/mmcblk0, which results in the error device not found:  
--- RBK0053I: Erste Partition (Bootpartition) wird auf /dev/mmcblk11 zurückgespielt.
--- RBK0054I: Zweite Partition (Rootpartition) /dev/mmcblk12 wird formatiert.
??? RBK0185E: Rootpartitionserstellung endet fehlerhaft mit RC 1.

After manually fixing the lines to /dev/mmcblk1 the restore is running as expected. Please change it in the script.

Kind regards.

framps, 2020-08-25:
Thank you very much you reported the issue. Did you test also whether rhe restored image works as expected?

I have to check whether a new option or some discovery is required to implement.

repohub, 2020-08-25:
Hello framps, thanks for the fast reply. First the script was running as expected without any error, but the SD Card was not working in the Pi. After a second run including the -C option, the restore is now fine and running. So it should be definitely a feature.

framps, 2020-08-25:
When option -C fixed your boot issue your SD card has some bad blocks. So I think it's time to buy another new SD card :wink: and next time the restore doesn't need time consuming option -C

framps, 2020-08-25:
Please verify whether [this updated version](https://github.com/framps/raspiBackup/blob/issue261_cardreader_restore/raspiBackup.sh) fixes your issue.

repohub, 2020-08-26:
It works as expected. Thanks for the fast response!

closed 2020-08-26 by framps