framps/raspiBackup #262: Add option to run post plugins after a failed backup

bcutter, 2020-09-05:
When a backup fails, post scripts (plugins/extensions) are not run.

**Implement an option to run post scripts also for failed backups.**

Additionally
a) being able to transfer the state (backup failed/successful) to a post script OR
b) being able to select one post script for successul and another for failed backups
would be perfect (that way a post script making notifications of backup status can adjust the text accordingly).

framps, 2020-09-06:
This is a valid request :+1: 

I will come back with an implementation proposal.

framps, 2020-09-06:
I checked the code and found out it's actually a bug. The return code is already passed in the plugin call as the first parameter and can be inspected but the plugins are not called when an error occurred.  In addition the services listed in option -a are not started in case of an error which is also not intended.

framps, 2020-09-06:
@bcutter I created a fix in the development branch. Please grab the code and check whether it's working as expected.

closed 2020-10-11 by framps

bcutter, 2020-12-18:
Sorry for turning back that late. I saw this has been shipped with 0.6.6 meanwhile. Without adjusting my post scripts now the post scripts for notyfing successful backups are always run, even when the backup failed.

Therefore: what return codes can I use to check wheter the raspiBackup was successful? I would also be fine with option b) noted in my original post.

framps, 2020-12-19:
> Without adjusting my post scripts now the post scripts for notifying successful backups are always run, even when the backup failed.

Great it works now.

>  what return codes can I use to check wheter the raspiBackup was successful

raspiBackup uses standard Linux convention. Returns 0 if everything was OK and a RC != 0 otherwise. The RC is also passed into the plugins as first parameter and can be checked there. 

> I would also be fine with option b) noted in my original post.

That's not required. Just create a small wrapper post plugin which checks the RC passed in as first parameter and handle the success case and failure case by calling a success script and a failure script.

bcutter, 2020-12-19:
> 
> That's not required. Just create a small wrapper post plugin which checks the RC passed in as first parameter and handle the success case and failure case by calling a success script and a failure script. 

That sounds like a good idea. I'll figure out how to do that (Debian bash syntax for handling RCs).

bcutter, 2020-12-20:
Can you tell me how to create a "raspiBackup failed" state (provokate a non-"0" RC) so I can check my post script?