framps/raspiBackup #263: Perform a callback when backup finished in addition/instead of e-mail

m42e, 2020-09-06:
Allow configuration of options:

```
DEFAULT_CALLBACK_SUCCESS=https://........
DEFAULT_CALLBACK_FAIL=https://........
DEFAULT_CALLBACK_DATA='{"somefield": "some value", "log": {{log}}}'
```

In usage with https://healthchecks.io

framps, 2020-09-06:
You can do this already :wink: Just create a plugin which is called when the backup finishes and use curl calls to invoke the service. For details about raspiBackup plugins see [here](https://www.linux-tips-and-tricks.de/en/all-pages-about-raspibackup/443-raspibackup-extensions/).

m42e, 2020-09-06:
@framps Ok, got it. could you add the log or the path to the log file to the post call?

framps, 2020-09-06:
Well, you can access the log file name already :wink: . The plugins are called from raspiBackup and thus you have access to all variables used by raspiBackup. Just use `$LOG_FILE` to get the log file name. If you need the backup directory use `$BACKUPTARGET_DIR`.

That's not a clean interface between raspiBackup and the plugins to use variables directly. But I think it's OK as long as you use variables which are static from a coding point of view and both variable names will not change in the future.

Note there is an open issue with plugins and their call in case of a backup error (See #262). But as far as I can see you're not affected by this issue.

m42e, 2020-09-06:
Ok Thanks a lot. Got it working as I liked to.

closed 2020-09-06 by m42e

bcutter, 2020-09-06:
Nice to explore another service like PushingBox; healthcheck.io looks more advanced. Great other users also care bout their raspiBackups :-)