framps/raspiBackup #264: MIME-Version and Content-Type in subject field.

m42e, 2020-09-06:
Whenever raspiBackup is sending me mails, It places the MIME-Version and the Content-Type information in the Subject line, which causes my spam filter to discard the mail.

E.g. I get the following subject line:
```
raspberrypi: Backup finished successfully. MIME-Version: 1.0 Content-Type: text/html; charset=utf-8
```
[raspiBackup.log](https://github.com/framps/raspiBackup/files/5179978/raspiBackup.log)

framps, 2020-09-06:
Root cause is email coloring is enabled with `COLORING=CM`. If you turn eMail coloring off the issue will be gone.

Unfortunately the way to enable coloring in the eMails is handled differently by the MUAs/MTAs. Which eMail MUA and MTA do you use?

m42e, 2020-09-06:
I thought I had setup the two raspberries the same way, but on one bsd-mailx was installed, the other used GNU Mailutils. bsd-mailx did not work!

so
```
$ mail -V
```
should not print an error :)

closed 2020-09-06 by m42e