framps/raspiBackup #265: Ordner für das Backup ignorieren

derDennis99, 2020-09-10:
Hallo,

wenn ich ein Backup machen möchte bekomme ich folgenden Fehler:
`rsync: readlink_stat("/home/pi/thinclient_drives") failed: Permission denied (13)`

damit das nicht mehr passiert wollte ich diesen Pfad excluden mit:
`sudo raspiBackup.sh -u "--exclude=/home/pi/thinclient_drives"`

als Ergebnis kommt:
`??? RBK0090E: Option -u erwartet einen Parameter. Falls der Parameter mit '-' beginnt beginne stattdessen mit '\-'.`

dann habe ich folgendes probiert:
`sudo raspiBackup.sh -u "\-exclude=/home/pi/thinclient_drives"`

dann fängt das Backup an zu laufen... bekomme allerdings den selben Fehler wie oben. Der Pfad wird anscheinend nicht Ignoriert.

Wie kann ich das Problem lösen?

Gruß

framps, 2020-09-10:
Versuche Mal

```
sudo raspiBackup.sh -u "\--exclude=/home/pi/thinclient_drives
```

Falls das nicht funktioniert gibt bitte Bescheid. Dann muss ich reinsehen woran es liegt. Als Workaround kannst Du auch die Datei im Config File excluden. Das sollte auf alle Fälle funktionieren.

derDennis99, 2020-09-10:
Danke dir für deine Antwort! :)
`sudo raspiBackup.sh -u "\--exclude=/home/pi/thinclient_drives`
hat funktioniert!

Kurze Frage dazu: Bei den Einstellungen von dem automatischen Regelmäßigen update: Wird da auch diese Einstellung gespeichert? oder kann man das dort auch einstellen? wenn nicht wie ist das mit der Config-Datei wird das dann auch für ALLE Updates übernommen sprich auch für die Automatischen?

Gruß

framps, 2020-09-10:
> Bei den Einstellungen von dem automatischen Regelmäßigen update: Wird da auch diese Einstellung gespeichert?

Nein. Der Installer erlaubt nur die wichtigsten Optionen zu konfigurieren. 

> wenn nicht wie ist das mit der Config-Datei wird das dann auch für ALLE Updates übernommen sprich auch für die Automatischen?

Ja. Es ist kein Unterschied ob raspiBackup manuell oder automatisch aufgerufen wird. Immer werden die Konfigurationsdaten beruecksichtigt.

derDennis99, 2020-09-10:
Ok letzte Frage ;) : wo in der Config-Datei muss ich einfügen, dass der Pfad nicht beachtet werden soll?

derDennis99, 2020-09-10:
Habs gefunden :)

```
# Excludeliste für das benutzte Backuprogramm
DEFAULT_EXCLUDE_LIST=""
```

Danke für die Hilfe! :)

closed 2020-09-10 by derDennis99

framps, 2020-09-10:
> Danke für die Hilfe! :)

NP. Bitte den Issue closen :wink: