framps/raspiBackup #267: Restore: Rootpartitionserstellung endet fehlerhaft mit RC 1.

derDennis99, 2020-09-11:
Hallo lieber Entwickler/Leser,

ich wollte heute mal das Restore testen. Dabei bin ich wie folgt vorgegangenen:
`sudo fdisk -l | egrep "^Disk /|^/dev"`
Ausgabe:
```
Disk /dev/ram0: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram1: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram2: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram3: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram4: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram5: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram6: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram7: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram8: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram9: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram10: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram11: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram12: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram13: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram14: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/ram15: 4 MiB, 4194304 bytes, 8192 sectors
Disk /dev/mmcblk0: 29,7 GiB, 31914983424 bytes, 62333952 sectors
/dev/mmcblk0p1        8192   532479   524288  256M  c W95 FAT32 (LBA)
/dev/mmcblk0p2      532480 62333951 61801472 29,5G 83 Linux
Disk /dev/sda: 931,5 GiB, 1000204883968 bytes, 1953525164 sectors
/dev/sda1        2048 1953521663 1953519616 931,5G  7 HPFS/NTFS/exFAT
```
Dem konnte ich entnehmen, das /dev/sda mit 931,5 GB meine USB Platte ist auf der die Backups abgelegt werden sollen und /dev/mmcblk0 die SD-Karte im Raspi ist.

Darauf hin habe ich folgenden Befehl ausgeführt:
`sudo raspiBackup.sh -d /dev/mmcblk0 /media/pi/USB-Platte/Raspi_Backups/raspberrypi/raspberrypi-rsync-backup-20200911-100001/`
Das gab mir den Fehler: 
`??? RBK0154E: Ein Restore ist nicht möglich wenn eine Partition von /dev/mmcblk0 gemounted ist.`

Also habe ich geprüft, welche Partitionen gemountet sind:
`sudo mount | grep /dev/mmcblk0`
Ausgabe:
```
/dev/mmcblk0p2 on / type ext4 (rw,noatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
```

der umount für /dev/mmcblk0p1 lief der für /dev/mmcblk0p2 allerdings nicht da bekomme ich die Ausgabe:
`umount: /: target is busy.`

ich habe trotzdem nochmal versucht ein Restore zu machen und bekomme dann den Fehler:
`??? RBK0185E: Rootpartitionserstellung endet fehlerhaft mit RC 1.`

Was habe ich falsch gemacht?

Danke schon mal fürs durchlesen :)

Gruß

framps, 2020-09-11:
Ich habe an Deinen Informationen gesehen dass Du einen rsync Backup auf ein NTFS formatiertes Filesystem erstellst. Das gibt leider ein unbrauchbares Backup. Siehe dazu auch issue #259 . D.h. Du solltest auf ein tar Backup umstellen oder die Backupartition mit ext3/4 formatieren.

framps, 2020-09-21:
@derDennis99 Gibt es neue Informationen? Ansonsten bitte den Issue closen.

derDennis99, 2020-09-26:
@framps sorry, dass ich jetzt erst antworte! Ich habe jetzt das Backup auf DD gestellt habe auch den Restore getestet läuft alles :) (tar hat leider auch nicht funktioniert)

closed 2020-09-26 by derDennis99