framps/raspiBackup #268: Mount /dev/mmcblk0p2 to /tmp/mmcblk0p2 failed. RC 32

vsc55, 2020-09-11:
When updating to the latest version (V0.6.5.1 (a056338)) one of my raspberries has started to give errors.
This rasberry has Noobs installed and this may be the cause of the problem.
If I fly to version V0.6.4.3 (2d927a2) it works again without problems.

```bash
* Attached raspiBackup debug log
--- RBK0009I: pdc01: raspiBackup.sh V0.6.5.1 (a056338) started at vie sep 11 19:45:21 CEST 2020.
--- RBK0128I: Using logfile /backup/pdc01/pdc01-rsync-backup-20200911-194520/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
!!! RBK0236W: Partition oriented backup will not be maintained any more and disable somewhere in the future.
--- RBK0151I: Using backuppath /backup.
!!! RBK0157W: No services to stop.
--- RBK0081I: Creating backup of type rsync in /backup/pdc01/pdc01-rsync-backup-20200911-194520.
--- RBK0085I: Backup of type rsync started. Please be patient.
--- RBK0036I: Saving partition layout.
--- RBK0045I: Creating backup of partition layout in /backup/pdc01/pdc01-rsync-backup-20200911-194520/pdc01-backup.sfdisk.
--- RBK0045I: Creating backup of partition layout in /backup/pdc01/pdc01-rsync-backup-20200911-194520/pdc01-backup.blkid.
--- RBK0045I: Creating backup of partition layout in /backup/pdc01/pdc01-rsync-backup-20200911-194520/pdc01-backup.parted.
--- RBK0045I: Creating backup of partition layout in /backup/pdc01/pdc01-rsync-backup-20200911-194520/pdc01-backup.fdisk.
--- RBK0046I: Creating backup of master boot record in /backup/pdc01/pdc01-rsync-backup-20200911-194520/pdc01-backup.mbr.
??? RBK0213E: Mount /dev/mmcblk0p2 to /tmp/mmcblk0p2 failed. RC 32.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0043I: Removing incomplete backup in /backup/pdc01/pdc01-rsync-backup-20200911-194520. This may take some time. Please be patient.
--- RBK0049I: Messages saved in /root/raspiBackup.msg.
--- RBK0026I: Debug logfile saved in /root/raspiBackup.log.
--- RBK0010I: pdc01: raspiBackup.sh V0.6.5.1 (a056338) stopped at vie sep 11 19:45:39 CEST 2020 with rc 102.
??? RBK0005E: Backup failed. Check previous error messages for details.
```

framps, 2020-09-11:
Would you please provide the debug log of the failing run? You provided the messages but not the debug log :disappointed:

vsc55, 2020-09-11:
Sorry, the log file was not attached.

https://github.com/framps/raspiBackup/files/5210900/raspiBackup.log

framps, 2020-09-11:
NP. Thank you very much for the log. I found
```
mount: /tmp/mmcblk0p2: can't read superblock on /dev/mmcblk0p2.
```
which doesn't help much :-(

Would you please provide the debug log of the successful run with the previous version?

vsc55, 2020-09-11:
Right now I am making a copy with the previous version. As soon as I finish I upload it.

vsc55, 2020-09-11:
> 
> 
> NP. Thank you very much for the log. I found
> 
> ```
> mount: /tmp/mmcblk0p2: can't read superblock on /dev/mmcblk0p2.
> ```
> 
> which doesn't help much :-(
> 
> Would you please provide the debug log of the successful run with the previous version? But please note: The previous version does not masquerade sensitive information in the log !

It seems to try to mount an extended partition as if it were a logical partition.
`/dev/mmcblk0p2      3300782 31223807 27923026 13,3G  5 Extended`

framps, 2020-09-12:
Good catch. Yes, I changed the logic because I want to catch and handle all mount errors. They caused issues. This change made visible a coding bug: The actual root cause for the issue is https://github.com/framps/raspiBackup/blob/2cced034d8c53bd1abf213e846358170befe588f/raspiBackup.sh#L4963. There a check for extended partition is made to skip extended partitions. 85 is used but the partition you have on your system has type 5. 
```
20200911-194535 DBG 3797:                      --- sfdisk
20200911-194535 DBG 3798:                      --- label: dos
20200911-194535 DBG 3798:                      --- label-id: 0x000a3a0e
20200911-194535 DBG 3798:                      --- device: /dev/mmcblk0
20200911-194535 DBG 3798:                      --- unit: sectors
20200911-194535 DBG 3798:                      --- 
20200911-194535 DBG 3798:                      --- /dev/mmcblk0p1 : start=        8192, size=     3292590, type=e
20200911-194535 DBG 3798:                      --- /dev/mmcblk0p2 : start=     3300782, size=    27923026, type=5
20200911-194535 DBG 3798:                      --- /dev/mmcblk0p5 : start=     3301376, size=       65534, type=83
20200911-194535 DBG 3798:                      --- /dev/mmcblk0p6 : start=     3366912, size=      141312, type=c
20200911-194535 DBG 3798:                      --- /dev/mmcblk0p7 : start=     3514368, size=    27709440, type=83
```

Would you please modify this line to
```
			if [[ $type != 85 &&  $type != 5 && $size > 0 ]]; then # skip empty and extended partitions
```

and try again?

vsc55, 2020-09-12:
Yes, that solves the problem and now starts the copy without problem.
It seems that "Noobs" when I install it I create the type 5 extended partition.

framps, 2020-09-12:
Thank you very much you verified my patch. A fix will be available in next release. You either keep the patched version or you also can use the exclude partition option to exclude partition 2. Up to you :wink:

vsc55, 2020-09-12:
Here is a list of all the partition types:
https://en.wikipedia.org/wiki/Partition_type

framps, 2020-09-12:
I had NOOBs installed when I wrote the code for the partition oriented backup mode and there 85 was used (Linux extended). Looks like they decided to use 5 (extended) for some reasons.

You hopefully know the partition oriented backup mode is deprecated :wink:

framps, 2020-09-12:
Given the fact the patch is uncritical I patched and published the current version of raspiBackup. if you use options `-U -S` you will update your local version with the latest one :wink:

Would you please verify the update and close the issue then?

closed 2020-09-12 by vsc55