framps/raspiBackup #269: While starting "sudo raspiBackup.sh -a : -o : -m detailed" i receive error-message

Thomas-G007, 2020-09-13:
Please execute all following actions first and provide all requested information:

* Check whether my error messages were listed on the messages page and all suggested actions don't fixes my issue. [English](https://linux-tips-and-tricks.de/en/rmessages) - [German](https://linux-tips-and-tricks.de/de/fehlermeldungen)
* Attached raspiBackup debug log

If any action was not executed or any requested information was not provided the issue will be rejected.

framps, 2020-09-13:
>     * Attached raspiBackup debug log

:weary:

Thomas-G007, 2020-09-14:
Sorry, but i thought i would have included the backup.log - but I don't see it here as an attachment, so i copy and paste it here.

Best regards Thomas

20200912-220638 DBG 1673:      --> initializeDefaultConfig 
20200912-220638 DBG 1805:      <-- initializeDefaultConfig 
20200912-220638 DBG 2548:      --> readConfigParameters 
20200912-220639 DBG 2604:      <-- readConfigParameters 
20200912-220639 DBG 7128:      --> checkOptionParameter -a :
20200912-220639 DBG 7149:      <-- checkOptionParameter :
20200912-220639 DBG 7128:      --> checkOptionParameter -o :
20200912-220639 DBG 7149:      <-- checkOptionParameter :
20200912-220639 DBG 7128:      --> checkOptionParameter -m detailed
20200912-220639 DBG 7149:      <-- checkOptionParameter detailed
20200912-220639 DBG 1810:      --> initializeConfig 
20200912-220639 DBG 1878:      <-- initializeConfig 
20200912-220639 DBG 1907:      --> substituteNumberArguments 
20200912-220639 DBG 1934:      <-- substituteNumberArguments 
20200912-220639 DBG 6968:      --> check4RequiredCommands 
20200912-220639 DBG 6989:      <-- check4RequiredCommands 
20200912-220639 DBG 7717:  --- RESTORE: 0 - fileParameter: 
20200912-220639 DBG 7736:  --- Enabling trap handler
20200912-220639 DBG 1321:  --- TRAP cleanup SIGINT SIGTERM EXIT
20200912-220639 DBG 2609:      --> setupEnvironment 
20200912-220639 DBG 2613:      --- Previous logfile: //raspiBackup.log
20200912-220639 DBG 2718:      --- +================================================================================================================================================+
20200912-220639 DBG 2719:      --- | ===> A lot of sensitive information is masqueraded in this log file. Nevertheless please check the log carefully before you distribute it <=== |
20200912-220639 DBG 2720:      --- +================================================================================================================================================+
20200912-220639 DBG 2722:      --- LOG_OUTPUT: 2
20200912-220639 DBG 2723:      --- Using logfile /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20200912-220638/raspiBackup.log
20200912-220639 DBG 2724:      --- Using msgfile /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20200912-220638/raspiBackup.msg
20200912-220640 DBG 2732:      --- BACKUPTARGET_DIR: /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20200912-220638
20200912-220640 DBG 2733:      --- BACKUPTARGET_FILE: /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20200912-220638/@HOSTNAME@-rsync-backup-20200912-220638
20200912-220640 DBG 2735:      <-- setupEnvironment 
20200912-220640 DBG 2862:      --> colorAnnotation 1
20200912-220640 DBG 2886:      <-- colorAnnotation 
--- RBK0009I: @HOSTNAME@: raspiBackup.sh V0.6.5.1 (a056338) started at Sat Sep 12 22:06:40 CEST 2020.
20200912-220640 DBG 2862:      --> colorAnnotation 1
20200912-220640 DBG 2886:      <-- colorAnnotation 
--- RBK0128I: Using logfile /backup/@HOSTNAME@/@HOSTNAME@-rsync-backup-20200912-220638/raspiBackup.log.
20200912-220640 DBG 2862:      --> colorAnnotation 1
20200912-220640 DBG 2886:      <-- colorAnnotation 
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
20200912-220640 DBG 7761:  --- Read config /usr/local/etc/raspiBackup.conf : 0.1.4
20200912-220640 DBG 7761:  --- VERSION_CONFIG="0.1.4"
20200912-220640 DBG 7761:  --- DEFAULT_BACKUPPATH="/backup"
20200912-220640 DBG 7761:  --- DEFAULT_BACKUPTYPE="rsync"
20200912-220640 DBG 7761:  --- DEFAULT_ZIP_BACKUP="0"
20200912-220640 DBG 7761:  --- DEFAULT_DD_BACKUP_SAVE_USED_PARTITIONS_ONLY=0
20200912-220640 DBG 7761:  --- DEFAULT_KEEPBACKUPS="5"
20200912-220640 DBG 7761:  --- DEFAULT_KEEPBACKUPS_DD=0
20200912-220640 DBG 7761:  --- DEFAULT_KEEPBACKUPS_DDZ=0
20200912-220640 DBG 7761:  --- DEFAULT_KEEPBACKUPS_TAR=0
20200912-220640 DBG 7761:  --- DEFAULT_KEEPBACKUPS_TGZ=0
20200912-220640 DBG 7761:  --- DEFAULT_KEEPBACKUPS_RSYNC=0
20200912-220640 DBG 7761:  --- DEFAULT_SMART_RECYCLE=0
20200912-220640 DBG 7761:  --- DEFAULT_SMART_RECYCLE_DRYRUN=1
20200912-220640 DBG 7761:  --- DEFAULT_SMART_RECYCLE_OPTIONS="7 4 12 1"
20200912-220640 DBG 7761:  --- DEFAULT_STOPSERVICES="systemctl stop cron && systemctl stop openhab2 && systemctl stop xrdp-sesman && systemctl stop xrdp"
20200912-220640 DBG 7761:  --- DEFAULT_STARTSERVICES="systemctl start xrdp && systemctl start xrdp-sesman && systemctl start openhab2 && systemctl start cron"
20200912-220640 DBG 7761:  --- DEFAULT_BEFORE_STOPSERVICES=""
20200912-220640 DBG 7761:  --- DEFAULT_AFTER_STARTSERVICES=""
20200912-220640 DBG 7761:  --- DEFAULT_EMAIL="T@@@@e(23)"
20200912-220640 DBG 7761:  --- DEFAULT_SENDER_EMAIL=""
20200912-220640 DBG 7761:  --- DEFAULT_EMAIL_PARMS=""
20200912-220640 DBG 7761:  --- DEFAULT_MAIL_PROGRAM="mail"
20200912-220640 DBG 7761:  --- DEFAULT_TELEGRAM_TOKEN=""
20200912-220640 DBG 7761:  --- DEFAULT_TELEGRAM_CHATID=""
20200912-220640 DBG 7761:  --- DEFAULT_TELEGRAM_NOTIFICATIONS="F"
20200912-220640 DBG 7761:  --- DEFAULT_NOTIFY_START=0
[... 256 lines cut ...]
20200912-220651 DBG 2862:                          --> colorAnnotation 1
20200912-220651 DBG 2886:                          <-- colorAnnotation 
--- RBK0167I: Sending email.
20200912-220651 DBG 2862:                          --> colorAnnotation 0
20200912-220651 DBG 2886:                          <-- colorAnnotation 
20200912-220651 DBG 3044:                      --- eMail: T@@@@e(23)
20200912-220651 DBG 3045:                      --- eMail Program: mail
[... cut ...]
20200912-220651 DBG 3057:                      --- mail: RC: 0
20200912-220651 DBG 3100:                      <-- sendEMail 
20200912-220651 DBG 3209:                  --- Masquerading eMail
20200912-220651 DBG 3232:                  --- Masquerading some mount options
20200912-220651 DBG 3250:                  --- Masquerading home directory name
20200912-220651 DBG 3255:                  --- Masquerading hostname
20200912-220652 DBG 3260:                  --- Masquerading sensitive non local IPs

Thomas-G007, 2020-09-14:
.... and the strange thing is, that I could save the SD-Card manuel with
dd if=/dev/mmcblk0  .....
without problems. But this is not the solution for me because I like raspiBackup and I would like to get it running succuessful.

Best regards - Thomas

framps, 2020-09-14:
```
20200912-220641 DBG 1553:          *** cat /etc/fstab
20200912-220642 DBG 1553:              /dev/mmcblk0p2 / ext4 defaults,noatime,nodiratime,commit=600,errors=remount-ro 0 1
20200912-220642 DBG 1553:              tmpfs /tmp tmpfs defaults,nosuid 0 0
20200912-220642 DBG 1553:              /dev/sda1 /backup ext4	noatime,data=ordered 0 0
```
I miss a definition for /dev/mmcblk0p1

```
20200912-220642 DBG 4856:              /dev/mmcblk0p1 on /media/pi/BPI-BOOT1 type vfat (rw,nosuid,nodev,relatime,uid=1000,gid=1000,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,showexec,utf8,flush,errors=remount-ro,uhelper=udisks2)
```

/dev/mmcblk0p1 is mounted on /media/pi which is non standard Raspbian.

```
20200912-220642 DBG 5116:                  *** cat /proc/cmdline
20200912-220642 DBG 5116:                      BOOT_IMAGE=/boot/vmlinuz-5.1.1-bpi root=UUID=0f6bf366-e545-49bb-87e8-07a2c88c5563 ro quiet splash
```

Looks like you use a read only overlay file system. Do you? raspiBackup was never tested with a read only overlay filesystem.

framps, 2020-09-21:
@Thomas-G007 Any update on this issue?

Thomas-G007, 2020-09-22:
Dear Framps,

I did the raspiBackup-Script this evening again --- and now it ended successful. Wonderful!! 
I don't know exact, what is the difference between my call one week ago and now but i think some of my software-installatons and removes of software in this time - perhaps - did a good influence to have another base. One reason more to save this succuessful backup as a restart-point for problems with other software in the future.

Of course, I will test the restore procedure in the next days.

Thanks for all - Thomas

framps, 2020-09-22:
Great it works now. 

Frankly I would be happier to know why you got the issue just in case to be prepared for the next time somebody faces the same issue.

Please close the issue if you're fine now.

closed 2020-09-22 by Thomas-G007