framps/raspiBackup #271: which files are created - uninstall?

ka223, 2020-09-22:
Hi, 

which files are created during install and how could it be completely removed ?
Just asking, if I need to remove raspiBackup.

Thanks

framps, 2020-09-22:
That's a valid question :+1: . I also don't like to have any orphan files on my system when I deleted a tool.

The raspiBackup installer can also be used to uninstall raspiBackup and most files created for raspiBackup are deleted :wink: 

Following files will be removed by the installer:
1) /usr/local/bin/raspiBackup.sh
2) /usr/local/bin/raspiBackupInstallUI.sh
3) /usr/local/etc/raspiBackup.conf
4) /etc/cron.d/raspiBackup

But there are still some files which will not be deleted by the installer:
1) Files in the /tmp directory. They will be cleaned up when /tmp will be cleaned up.  All of them are prefixed with raspiBackup.
2) All backup directories created by raspiBackup
3) If you have upgraded raspiBackup there are old versions of raspiBackup saved in /usr/local/bin in order to be able to revert to a previous version. All of them are prefixed with raspiBackup.
4) Every raspiBackup run will create log files called raspiBackup.log and raspiBackup.msg in the home directory or current directory

framps, 2020-10-03:
@ka223 I updated the installer (see #272) and will publish the updated installer the next days. Please close the issue if you're fine now :wink:

closed 2020-10-04 by ka223