framps/raspiBackup #276: Backup problems: seems not to support softlinks

jesusdeveloper, 2020-10-12:
Hello!

I've done all configurations but when I start my backup, the process is stopped and the log shows this message:
`??? RBK0020E: Filesystem of rsync backup directory /media/rasperryMYcloud/backups_raspberry seems not to support softlinks.`
I've a network hard disk Western digital and in the Rasperry OS this directory is always functional. It seems that my directory of network hard disk not support softlinks, but... what does this want to say?

I'm very interested because the backup destination I want to store in this network hard disk...

Thank you!

framps, 2020-10-13:
raspiBackup is available now for more than 7 years and in this timeframe I got a lot of error reports. A lot of them were because the environment doesn't support features required to successfully backup and restore a system. Every time such a missing feature was detected I added a check to make sure raspiBackup reports the issue and terminates.

One of them which created nasty errors on the restored system was the missing capability to save softlinks in the backup directory. `ln -s <target> <linkname>` has to create a softlink. rsync requires this feature. As an alternative you ca use tar and  you will not get this error. But then you don't profit from hardlinks. 

You wrote you don't have any issue with data on this partition. As long as you don't use softlinks everything is OK. But a Linux OS uses softlinks a lot of time and a rsync backup depends on softlinks to be complete.

framps, 2020-11-06:
@jesusdeveloper Any update on this issue? Otherwise I'll close the issue in a week.

closed 2020-11-14 by framps