framps/raspiBackup #278: DDZ Backup written into logs

eloo, 2020-10-28:
Hi, 
i'm currently checking my backups which are made with DDz and i see something very weird.
The backup file is empty. But the log file is around 13gb oO

Then i've checked the log file and it looks like this:

```
10-01-2020 05:00:07 --- RBK0081I: Backup vom Typ ddz wird in /media/red/***/Sicherung/NAS/NAS-ddz-backup-20201001-050001 erstellt.
20201001-050007: MSG           10-01-2020 05:00:07 --- RBK0081I: Backup vom Typ ddz wird in /media/red/***/Sicherung/NAS/NAS-ddz-backup-20201001-050001 erstellt.
10-01-2020 05:00:09 --- RBK0085I: Backuperstellung vom Typ ddz gestartet. Bitte Geduld.
20201001-050009: MSG              10-01-2020 05:00:09 --- RBK0085I: Backuperstellung vom Typ ddz gestartet. Bitte Geduld.
9Fu_��	X�����Dq0QQGs����wTD-�WqC����D3+ײE3�r�w����JŭA��9����f����y�=�<���a6fyg�3��}FIʥ�s�ؤ;�)�e��>�!��zZE	
�9ն�q���#�4$��v>7��zB�T��W}T�K��9�w�n��^E9ACU�
                                              u�UU�GFGGN����7��Eύ6FLk�檎U}.�+�j�:T�a�{���cAL�-mbN���c�����u�${��d{,�ஆgY���;|�y�^�ߘ��M���'�'#xX�s�ѱW�Us����~��s����ש�Ψm8�{S���E��VSΩ�/�դ#֭6��u�M�x[7��g=k��Y�`�Tc䌩�����Y��c�12jz�},�]T�o5�������DM�Ć��*��
``` 

This locks like the backup was written into my log file. 
Do you have ideas why this could happen? 

The version i use is:  0.6.4.2

Do you maybe have any ideas how i can "extract" the backup out of the log file? :D Maybe strip out some bytes? 

Thanks

framps, 2020-10-28:
This is an issue which was reported and fixed in #112. When I got the issue I broadcasted this severe issue at various places. It looks like you didn't notice this broadcast and still use the buggy version :cry: Please upgrade to the latest release with option `-U` to get the fix.

> Do you maybe have any ideas how i can "extract" the backup out of the log file? :D Maybe strip out some bytes?

I think you will be able to strip the leading and trailing log messages with `head` and `tail`.

eloo, 2020-10-28:
Awww.. Thats a pitty that i missed this :D 

Thanks for the information.

FYI: I've tried to figure out the used bytes of the log with head and tail and now i'm trying to strip out the img.gz file of it.. 
I will report my result here

framps, 2020-10-28:
This incidence pushed me to add some notification feature inside of raspiBackup to mark a release as deprecated and have raspiBackup to write a message to upgrade the current version. 

I don't hope there will be others out there who will face the same issue - but just in case - would be great if you could document in this issue how you recovered (crossing fingers you will get it done).

eloo, 2020-10-29:
Okay. so for the record. 
I was NOT able to extract the backup out of the "log".
I guess the problem would was thats its pretty hard to identify the exact byte range you need. 
Also when you are using a blocksize of 1 in `dd` you need a good amount of time :D 

 Maybe my approach would work if the backup is no compressed as gz.

But at least i had some older backup so i'm still safe :D

Thanks

framps, 2020-10-29:
Thank you very much for the update. Fortunately you have an older Backup 🙂. 

I think the problems are
1) to find the starting offset which I think should'nt be difficult. Just Start after the last LF of the last Message in the Header
2) find the ending offset Witz e hex editor. Just go back from the end to the beginning of the first message in the trailing Messages
3) In think there will be other messages indienthe binary block and they ard difficult to find and to strip 😟

🤞Crossing fingers you are the only one facing this nasty Issue 🤞

Please close the issue if you are fine now. (Don't forget to update raspiBackup 😉 )

eloo, 2020-10-29:
Thanks for your help.
And yes i will update my raspiBackup :D

closed 2020-10-29 by eloo