framps/raspiBackup #279: Permission issue on restore

Kobbe1, 2020-10-29:
Hi,

Great tool first of all, has served me well on various occasions.

Here is what I want to share:
I had to restore a standard raspberry OS (raspian) 32 Gb SD card recently.
While the restore worked and the system booted it seems the file permissions where not restored - no folders/files are owned by root. I found a similar issue here but it seems to have a different root cause.
On further inspection I believe that this is related to the fact that the backup media is an NTFS formatted SSD.
Google reveals that is has to be mounted with the -permissions option - however I did not do any manual fstab editing etc - it did show up as drive and was writable when I just plugged it in so I used it like this. The test restore did not show it to me as I only booted the system successfully into the desktop which worked, then I went back to the original SD calling the restore good.

Without the -permissions option it seems that setting permissions seems to fail silently - I can set owners with chown without error, but checking the file afterwards shows owner did not change.

Going forward I will use a different file system to avoid this as I do on other machines, but I wonder:
Is the above correct and an known issue? Maybe mentioned in the documentation to avoid NTFS?
Obviously this is not caused by the backup tool - but maybe the script could do something to prevent this - throw a warning for NTFS file systems or actually do a test file and set permissions and check they fit properly set ...

I think I will need to start with a new image and manually restore the data, which should be straightforward compared to the task of trying to manually restore file permissions for the whole OS ...

Best,
-T

framps, 2020-10-29:
Thank you very much you report the issue. :+1:

It's a very nasty issue and was recently also reported by another raspiBackup user in a comment on my website. He fortunately detected the issue when he tested the restored backup and told me about the issue. That's why I created #247 which checks for valid filesystems when rsync is used (ext2/3/4 or nfs) and thus will reject a ntfs filesystem.

This issue is already implemented and will be published in upcoming next release of raspiBackup.  There are already a lot of consistency- and environment checks done by raspiBackup but because I'm not a Windows user I wasn't aware about this :sad: .

I frankly didn't add anything about this issue in the documentation until now. But I see your point and will document this at some prominent place right now.

> actually do a test file and set permissions and check they fit properly set

I don't have a Windows system and cannot execute any tests on NTFS. Thus I decided just to reject any ntfs filesystem when rsync is used.

>  think I will need to start with a new image and manually restore the data, which should be straightforward compared to the task of trying to manually restore file permissions for the whole OS ...

I think that's the right way to go. Sorry for the inconvenience. Crossing fingers you will get your OS up and running again.

framps, 2020-11-01:
There is an additional check added in raspiBackup (See #247) to make sure NTFS cannot be used as backup filesystem when rsync is used.

framps, 2020-11-01:
@Kobbe1 Please close the issue if you are fine now :wink:

Kobbe1, 2020-11-01:
Hey,

thanks for the quick turnaround. Machine is up and running again, I found that I had an older backup that was done with dd, so I restored from there.
Really great to see this is fixed to save others the hassle.
Keep up the great work!

Best,

-T

closed 2020-11-01 by Kobbe1

framps, 2020-11-02:
Glad you managed to restore your data. It's a very nasty issue and therefore I decided to react immediately. Thanky you very much you brought this Issue to my attention. :+1: