framps/raspiBackup #280: Tar backup ends with error

eloo, 2020-10-31:
Hi,

i've just seen that some one of my backup jobs is not working as expected.
In the logs i get a 109 error.

System:
Debian GNU/Linux 10 (buster) (5.6.0-0.bpo.2-amd64)
Intel NUC

Partitions:
NAME   MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda      8:0    0 119,2G  0 disk 
├─sda1   8:1    0 115,4G  0 part /
├─sda2   8:2    0     1K  0 part 
└─sda5   8:5    0   3,9G  0 part [SWAP]

Backup target:
NFS share

More information can be found in the log.

Maybe its a problem with the changing files?
But the weird thing is.. With the beta version before the backup on the same system worked :D 
AFAIK i didn't have changed the settings since the last beta.

Thanks for your help
eloo
[raspiBackup-NUC.log](https://github.com/framps/raspiBackup/files/5469424/raspiBackup-NUC.log)

framps, 2020-10-31:
> Maybe its a problem with the changing files?

Yes.

```
tar: /var/lib/mysql/ibdata1: Datei hat sich beim Lesen geändert.
/138701b751c6ea56c4b12657e31e3e73a2b9a8dcf39d3d7a69d5508d61771a03-json.log: Datei hat sich beim Lesen geändert.
```

Looks like somebody now uses mysql and modified the json log file. 

> But the weird thing is.. With the beta version before the backup on the same system worked :D

Sounds it's some kind of race condition you have. Maybe the next backup run will succeed again. I strongly recommend to stop mysql first with options -a and -o and also make sure the service which modifies the json log is also stopped.

eloo, 2020-11-01:
okay. sounds good that this maybe could be the problem. i will give it a try to stop and resume some service.

framps, 2020-11-10:
@eloo Any update?

eloo, 2020-11-10:
Last backup was also not successful. 
But I need to check first the log to ensure that I have excluded all necessary directories.

framps, 2020-11-20:
@eloo Any update :wink:

eloo, 2020-11-27:
last backup was still failing.. but i figured out that my exclude was not working as expected. 
just started a new try with a working exclude list

eloo, 2020-12-02:
okay.. could make it work as it seems that my files are changing to fast on this system.
nevertheless i've changed to btrfs anyway. 
so for me we could close this

closed 2020-12-02 by framps