framps/raspiBackup #281: Add check for compressed files?

eloo, 2020-10-31:
Hi, 
as you may know i had an issue before with my backup. 
And i was thinking how i could improve the integrity of my backups and i came across the `gzip -t ...` command.

Maybe it would be a good idea to add an option to add a verify step using `gzip -t` for compressed backups.

What do you think?

Thanks

framps, 2020-10-31:
I see your point. But this will increase the backup time and most raspiBackup users want to have a short backup time. I may add this enhancement with a new option but I frankly don't think many people will use this option. 

You can easily use a [post plugin](https://www.linux-tips-and-tricks.de/en/all-pages-about-raspibackup/443-raspibackup-extensions/) and invoke `gzip -t` against the created zip file and abort the backup if this command fails.

eloo, 2020-11-01:
okay.. i see your point with the backup time. 
but maybe its possible to have the integrity check after the restart services step.
So technically the backup is already done and additionally the integrity cyn be checked with the result in the log/email.

But i will have a look at the post plugin

framps, 2020-11-24:
@eloo Any update so far?

framps, 2020-12-19:
Closing because of inactivity

closed 2020-12-19 by framps