framps/raspiBackup #284: restored SD-card issues Network Error when connecting

mcmellow, 2020-11-02:
Can not find anything the same on your website or in this github.

I have created a backup of the 32GB Sd-Card on a RPi 2B to an USB-Stick with 
`sudo raspiBackup.sh -a : -o : -m detailed`  
see also this log
`--- RBK0009I: raspi-spare-2B: raspiBackup.sh V0.6.5.1 (9cc17fc) started at Mon  2 Nov 11:59:48 CET 2020.
--- RBK0128I: Using logfile /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0031I: Checking whether a new version of raspiBackup.sh is available.
--- RBK0151I: Using backuppath /backup.
!!! RBK0157W: No services to stop.
--- RBK0081I: Creating backup of type rsync in /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946.
--- RBK0036I: Saving partition layout.
--- RBK0044I: Creating backup of boot partition in /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946/raspi-spare-2B-backup.img.
--- RBK0045I: Creating backup of partition layout in /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946/raspi-spare-2B-backup.sfdisk.
--- RBK0046I: Creating backup of master boot record in /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946/raspi-spare-2B-backup.mbr.
--- RBK0158I: Creating native rsync backup "/backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946".
--- RBK0085I: Backup of type rsync started. Please be patient.
--- RBK0078I: Backup time: 00:28:41.
--- RBK1001I: Memory usage - Pre backup - Used: 194 MB Free: 489 MB - Post backup - Used: 203 MB Free: 312 MB
--- RBK1000I: CPU temperature pre and post backup: 37.9'C - 42.8'C
--- RBK1002I: Disk usage pre backup: Used: 3.53 GiB Free: 11.89 GiB
--- RBK1003I: Disk usage post backup: Used: 7.10 GiB Free: 8.32 GiB
--- RBK1004I: Free change: -3.56 GiB (-30.00 %)
!!! RBK0156W: No services to start.
--- RBK0159I: 3 backups kept for rsync backup type.
--- RBK0205I: Deleting oldest backup in /backup/raspi-spare-2B. This may take some time. Please be patient.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0049I: Messages saved in /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946/raspiBackup.msg.
--- RBK0026I: Debug logfile saved in /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946/raspiBackup.log.
--- RBK0010I: raspi-spare-2B: raspiBackup.sh V0.6.5.1 (9cc17fc) stopped at Mon  2 Nov 12:29:47 CET 2020 with rc 0.
--- RBK0017I: Backup finished successfully.
`

The restored this backup to a new SD-Card with the same RPi 2B  with
`sudo rsapiBackup.sh -d /dev/sdc  /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946`
see also this log
`--- RBK0009I: raspi-spare-2B: raspiBackup.sh V0.6.5.1 (9cc17fc) started at Mon  2 Nov 12:46:13 CET 2020.
--- RBK0128I: Using logfile /home/pi/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
--- RBK0138I: Using bootbackup /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946.
--- RBK0068I: Using bootpartition backup files starting with /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946 from directory raspi-spare-2B-backup.
!!! RBK0018W: Target /dev/sdc with 59.47 GiB is larger than backup source with 29.49 GiB. root partition will be expanded accordingly to use the whole space.
!!! RBK0065W: Device /dev/sdc will be repartitioned and all data will be lost.
--- RBK0067I: Current partitions on /dev/sdc:
Number  Start   End      Size     Type     File system  Flags
 1      1.05MB  63865MB  63864MB  primary  fat32        lba
--- RBK0066I: Device /dev/sdc will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/sdc1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/sdc2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N 
--- RBK0050I: Restoring backup from /backup/raspi-spare-2B/raspi-spare-2B-rsync-backup-20201102-115946.
--- RBK0052I: Creating partition(s) on /dev/sdc.
--- RBK0004I: Adjusting second partition from 29.23 GiB to 59.22 GiB.
--- RBK0053I: Restoring first partition (boot partition) to /dev/sdc1.
--- RBK0054I: Formating second partition (root partition) /dev/sdc2.
--- RBK0055I: Restoring second partition (root partition) to /dev/sdc2.
--- RBK0184I: Rootpartition check started.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0076I: Restore finished successfully.
--- RBK0010I: raspi-spare-2B: raspiBackup.sh V0.6.5.1 (9cc17fc) stopped at Mon  2 Nov 13:04:06 CET 2020 with rc 0.
--- RBK0017I: Backup finished successfully.
`

Changed the SD-Cards of the RPi 2B and try to boot from this SD-Card.   (headless)
When I try to connect with Kitty or Putty _ receive the following error

[image]
Did I forget something or did I missed something.  I expected to boot nicely from this new SD-Card.  Please help.

framps, 2020-11-02:
This should work. But it's impossible to find the root cause without any additional information. Could you mount the SD card which got the backup and check in /var/log for any unexpected error messages?

There are some common root causes for this kind of error:
1) You use a NTFS filesystem as backup storage. **NTFS cannot be used with rsync**. I just add code which checks this in the next release.
2) Your SD card you restore the backup is broken

mcmellow, 2020-11-02:
My USB-Stick has NTFS.  Did not look to that.  I will change it and test the same procedure again.

mcmellow, 2020-11-03:
> 1. You use a NTFS filesystem as backup storage. **NTFS cannot be used with rsync**. I just add code which checks this in the next release.

Changed the USB-stick from NTFS to EXT4  and made backup again.  Restore this on a new SD-Card  and Booted up with this card. It worked like a charm.  Thanks very much for your support.

closed 2020-11-03 by mcmellow