framps/raspiBackup #285: Allow to use rsync Server for rsync backuptype

framps, 2020-11-03:
rsync can use ssh to transfer files to a remote server running ssh. Just add option 
`-e ssh / root@<server>:/<backupPath>`
in the rsync call.

Prerequ: Configure root access from to be saved system to backup server via keys

This also requires to wrap all commands executed from the to be save system on the backup space in an ssh call. 

This is a feature for Linux folks only because the backup server has to be a Linux system. So question is whether there will be a significant number of raspiBackup users which use this feature and justify the development effort.

framps, 2020-11-30:
I did some more investigations on this request. It's no big issue to use rsync protocol to create the backup. But unfortunately raspiBackup was not designed for remote command execution which is required in order to execute various commands on a remote system. For example all the backup version handling requires remote command execution.

To add this feature a big redesign and huge rewrite of raspiBackup is required :cry:

framps, 2021-07-24:
Reopen this feature request. Will create a prototype to check whether it's possible to implement without a whole rewrite.

vguna, 2021-07-24:
What a coincidence... I was just now looking for that option :).

My scenario: I don't want to have another physical medium connected directly to my pi for backup (it has already enough devices attached via USB - which require quite some amount of power) and since my cloud server is 24/7 up anyway, I could store my backup there. Also it has the charme of a decentral backup - if the house burns down :).

First I thought about NFS, but it's not so secure (at least no easy to do) and requires additional ports to be open. SSH I already have and also already used Duplicati with remote rsync invocation via SSH for my Windows backups.

So just wanted to ping, that someone would also be interested in it :).
Thanks for all the effort you put into rasiBackup!

vguna, 2021-07-24:
Hm, one workaround for now could also be to let raspiBackup rsync backup to a local folder, and then setup an additional "remote ssh rsync" for that backup folder to the remote server 🤔

framps, 2021-07-24:
I like you idea. Actually I think that's not a workaround but the final way to go. Given the fact a local attached disk odr ssd is much faster than remote storage the downtime of services on the Raspberry will be much shorter. 

So the idea would be to create and keep all the time one local rsync backup and rsync this backup to a remote storage which may and can take much longer. As of now this can be implement with the wrapper script from the helper subdirectory to invoke raspiBackup and add the remote rsync command at the end.

The raspiBackup implementation will just keep one local backup and housekeep the remote backup versions just by executing the housekeeping commands (intelligent backupstrategy or just keep a given number of backups) via ssh. 

There is one issue I don't have a good idea right now how to solve: If there are some networking- or remote server issues during remote rsync the remote backup will be stale and has to be erased  at some point in time. A similar issue exists if a connectivity issue happens during backup housekeeping.

vguna, 2021-07-24:
Good point @ reduce downtime of services 👍. With "housekeeping commands" you mean managing the multiple backups on the remote server? So in your scenario, those won't live locally, but only on the remote server? Hm that would make somehow sense, since you don't want to fill up your local diskspace of the actual system with the different backups. But yeah, that would be somehow intrusive to raspiBackup and how it manages those backups I guess (has to happen on remove server now).

Meanwhile I tried to backup to /backup (root fs) but it won't let me. I did a default exclude in the config for it, but it still won't let me. But maybe I did something wrong.

I think this approach is also not an option if disk space is filling up and there's no way anymore to store the backup on the actual system. So in the end we're back to execute rsync remotely through ssh for everything - so no local storage (external or root fs) is needed. But yeah, service downtime bites again :).

Another option (for me at least) could also be, to not use the cloud server, but WOL my Ubuntu NAS before doing the actual backup and live with a normal NFS mount to the backup directory. Of course, losing the decentral backup idea - which might be acceptable.

framps, 2021-07-24:
> Meanwhile I tried to backup to /backup (root fs) but it won't let me.

Try undocumented option '-c' 😉

vguna, 2021-07-25:
FYI: for now I just went like described. rsync via NFS to local NAS via WOL.

I also got those ACL errors described in the [FAQ](https://www.linux-tips-and-tricks.de/de/faq/#a24) for some of my openhab files (I'm using openHabianPi). The FAQ says to use NFS4 which I do, but that did not the trick for me. Checked client and server. So I went with 3) and 🤞 that it will do no harm - the upcoming test-restore will tell.

One thing to maybe mention in the FAQ wrt NFS and performance. When I started, the backup took ages - although dd test was 50 mb/s on the NFS share. I actually didn't wait to finish - after 1 hour. Only after I activated `async` in the NFS export options, the 8GB systems was backedup in 9 minutes. That made a HUGE difference.
A complete rsync increment still takes 4 minutes though. I think that is the drawback of a remote share - it will have to read each file through the network to create the hashes to decide whether/what to sync :(.

So for me personally, that remote rsync through ssh is still something I would be interested in :). In this case the hashing of the files will be done locally on each system and only IF something has to be transferred it will costs bandwith/time. I think also the downtime of the services is not really an issue then. And even if, it's still just an option to choose from. The housekeeping could be even done while services are already up again - since the actual backup is finished.

framps, 2021-07-25:
> Only after I activated `async` in the NFS export options, the 8GB systems was backedup in 9 minutes. That made a HUGE difference.

Thank you very much for sharing this finding :+1:

> So for me personally, that remote rsync through ssh is still something I would be interested in :). In this case the hashing of the files will be done locally on each system and only IF something [...]

That's what I'll will start with because it sounds to be the most useful use of remote rsync.with the benefits you mentioned: 

I will keep you posted when I have a prototype handy :wink:

vguna, 2021-07-25:
Amazing and thanks again for that great tool 😃👍

framps, 2022-03-01:
Spent a lot of time in this issue and now face a seg fault in remote ssh execution :-(. I suspend the issue again. Would be much easier to implement in a higher language than bash. But this will require to redesign and rewrite raspiBackup in a higher language. Not sure whether I want to do this ...

closed 2022-03-01 by framps

Thomas-Sparber, 2024-11-29:
I also have the same requirement.

My possible "solution" I came up with is to mount the target backup directory using sshfs to /backup and then setting raspiBackup up to store the backup there.
I also created an `/etc/fstab` entry to mount it permanently.

BUT to be honest: I think a perfect solution would be to mount the target backup directory to a temporory location on the pi using sshfs, then doing the backup and finally unmounting the directory. I think this could be done quite easily.

framps, 2024-11-29:
Thank you very much for your comment. 

I agree, sshfs is another way to mount a remote backup directory instead to use nfs or samba or even davfs. But then there is still no rsync server used which will give you additional benefits. 

In addition it's a bit tricky to configure rsync to use a remote rsync server. I just read a thread where somebody stopped maintenance of a nice backup tool for a Synology because he got so many issues because users had trouble to configure the remote rsync ssh access.