framps/raspiBackup #286: Use systemd instead of cron

framps, 2020-11-06:


framps, 2021-09-08:
@Hofei90 Will use your provided code. Hope that's OK with you.

Hofei90, 2021-09-09:
Sure

Für mögliche Risiken und Nebenwirkungen fragen sie ihren ... oder ...

p1r473, 2021-09-10:
I prefer cron, maybe offer a choice to the users

framps, 2021-09-10:
The installer will be changed to create and update systemd configs. For backward compatibility existing cron definitions will still be supported by the installer. I don't plan a migration from cron to systemd so everybody who uses raspiBackup already will still continue to use cron. But everybody who installs raspiBackup from scratch will get systemd configs. 

If somebody like you still wants to use cron with a fresh installed raspiBackup that's still possible. Just create a cron config file, delete the systemd configs and you're done :wink:

framps, 2023-04-03:
Not sure whether I will keep support of cron. Unfortunately the installer which uses whiptail is not very developmentfriendly and requires a lot of manual tests. If I keep crontab this will double my tests :-1:

p1r473, 2023-04-03:
Up to you of course, but here I am giving support for Cron 👍🏻 Cron way easier then systemd

framps, 2023-05-11:
> Cron way easier then systemd

Agree. Therefore I move this issue again in the backlog :-)

framps, 2023-11-21:
> I prefer cron, maybe offer a choice to the users

raspiBackup will check whether systemd is available and use systemd and cron otherwise. Do you want raspiBackup to use cron even the system runs with systemd?

framps, 2023-11-26:
> @p1r473 Do you want raspiBackup to use cron even the system runs with systemd?

Default is systemd if systemd is available and if no cron config was found. crond otherwise. The default can be change with option -t and that way an old system still uses crond and new system can be setup with crond even systemd is available.