framps/raspiBackup #287: Excluding files [msg.sockets] is not working proper

PatrickGlatz, 2020-11-08:
Hello Framps,

love your raspiBackup script so far and i am using it on 2 raspberries to backup on a Synology NAS DS216+ii. 

One issues which shows up a lot is, that if i try to afterwards backup my Synology, it is having problems with the msg.sock files. Thus, i tried to exclude the folder from being backuped using the DEFAULT_EXCLUDE_LIST in /usr/local/etc/raspiBackup.conf
`DEFAULT_EXCLUDE_LIST="/var/lib/samba/private/msg.sock/*"`

But what happens, is that these files are now backuped to the "base" directory `/`and not correctly excluded.

The attached picture shows the directory which should normally be excluded. Tomorrow (when the next backup has run) i will additionally attach the backup directory on my synology (have deleted now all the msg.sock files manually for all the last backups and thus cannot post the picture now ;-) ) to illustrate the backuped state.

Did not find any information, whether this is an issue or a bug or intention. Would be happy to get some infos from you, with pleasure also in german (which makes it sometimes a little easier but thought of internationality here at github ;-) ) and how i can fix this issue.

Thanks in advance
Patrick

[image]

framps, 2020-11-08:
Sieh Dir bitte die [Beschreibung von DEFAULT_EXCLUDE_LIST](https://www.linux-tips-and-tricks.de/de/raspibackup/#parm_u) an :wink: . Es fehlt ein fuehrendes `--exclude` wodurch Du dem Backuptool sagst es soll alle Sockets in / kopieren und nicht excluden.

PatrickGlatz, 2020-11-08:
Ich gebe Dir morgen Bescheid, das könnte es gut sein 🙄. Sorry für den Aufwand und vielen Dank schon einmal für den Vermerk.

PatrickGlatz, 2020-11-09:
> Sieh Dir bitte die [Beschreibung von DEFAULT_EXCLUDE_LIST](https://www.linux-tips-and-tricks.de/de/raspibackup/#parm_u) an 😉 . Es fehlt ein fuehrendes `--exclude` wodurch Du dem Backuptool sagst es soll alle Sockets in / kopieren und nicht excluden.

As framps already mentioned, obviously i made a mistake with the "DEFAULT_EXCLUDE_LIST" Option and missed the `--exclude=` string. Issue is fixed, thanks a lot for the quick support! Awesome!

closed 2020-11-09 by PatrickGlatz