framps/raspiBackup #290: Script keeps running after it is finished

Kamairo, 2020-11-16:
* Provide the output of `sudo raspiBackup.sh --version` or `sudo raspiBackupInstallUI.sh --version`
```
Version: 0.6.5.1 CommitSHA: 9cc17fc CommitDate: 2020-10-31 CommitTime: 21:52:43
```

* Provide the output of `cat /etc/os-release`
```
PRETTY_NAME="Raspbian GNU/Linux 9 (stretch)"
NAME="Raspbian GNU/Linux"
VERSION_ID="9"
VERSION="9 (stretch)"
VERSION_CODENAME=stretch
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

I just upgraded to the version above and did not have this issue before. Did not change my crontab, which executes rapsiBackup.sh with several -a and -o commands twice a week:
```
00 03 * * Mon,Thu /usr/local/bin/raspiBackup.sh -a "service apache2 start && service supervisor start && service grafana-server start && service mariadb start && service mysql start && service mysqld start && su - pi -c 'node-red-start &' && /etc/init.d/mosquitto start" -o "service apache2 stop && service supervisor stop && service grafana-server stop && service mariadb stop && service mysql stop && service mysqld stop && node-red-stop && /etc/init.d/mosquitto stop" -m detailed
```

Now, these processes are still running after the cronjob is executed (I get the email that raspiBackup was finished successfully):
`ps -aux | grep raspi`:
```
root      9671  0.0  0.4   7828  3860 ?        S    03:00   0:00 /bin/bash /usr/local/bin/raspiBackup.sh -a service apache2 start && service supervisor start && service grafana-server start && service mariadb start && service mysql start && service mysqld start && su - pi -c 'node-red-start &' && /etc/init.d/mosquitto start -o service apache2 stop && service supervisor stop && service grafana-server stop && service mariadb stop && service mysql stop && service mysqld stop && node-red-stop && /etc/init.d/mosquitto stop -m detailed
root      9672  0.0  0.4   7828  3924 ?        S    03:00   0:00 /bin/bash /usr/local/bin/raspiBackup.sh -a service apache2 start && service supervisor start && service grafana-server start && service mariadb start && service mysql start && service mysqld start && su - pi -c 'node-red-start &' && /etc/init.d/mosquitto start -o service apache2 stop && service supervisor stop && service grafana-server stop && service mariadb stop && service mysql stop && service mysqld stop && node-red-stop && /etc/init.d/mosquitto stop -m detailed
root      9673  0.0  0.0   3536   356 ?        S    03:00   0:00 tee -ia /media/usb/backups/BigPi/BigPi-tgz-backup-20201116-030001/raspiBackup.log
root      9674  0.0  0.0   3536   316 ?        S    03:00   0:00 tee -ia /media/usb/backups/BigPi/BigPi-tgz-backup-20201116-030001/raspiBackup.log
```

framps, 2020-11-16:
Actually raspiBackup has code to make sure you cannot start a second time. Would you please go back to your previous raspiBackup Release with option -V and provide the release details (--version output) ? 

Then please kill the active raspiBackup processes and upgrade again. When a new backup was created with the new release please provide the debug log.

Kamairo, 2020-11-16:
Did not have any old versions installed with the -V option.
But I still have the mails from older versions, it was V0.6.3.2 commit 5659ee1.

I did a uninstall via the installer now and reinstalled + copied my config from the previous installation (also copied the newly generated UUID filed from the new config, not sure if that is necessary. So, let's wait for a new cronjob, will report back.

BTW those four processes I posted were running after **one** execution of the cronjob, not multiple. After I noticed that some processes were still running I killed all of them and waited for the next cronjob and the above was the result.

framps, 2020-11-18:
@Kamairo Any update ? :wink:

Kamairo, 2020-11-19:
Same result, it ran from 3:00 to 3:51 tonight.
```

> $ ps -aux | grep raspi
root      2519  0.0  0.4   7828  3860 ?        S    03:00   0:00 /bin/bash /usr/local/bin/raspiBackup.sh -a service apache2 start && service supervisor start && service grafana-server start && service mariadb start && service mysql start && service mysqld start && su - pi -c 'node-red-start &' && /etc/init.d/mosquitto start -o service apache2 stop && service supervisor stop && service grafana-server stop && service mariadb stop && service mysql stop && service mysqld stop && node-red-stop && /etc/init.d/mosquitto stop -m detailed
root      2520  0.0  0.4   7828  3924 ?        S    03:00   0:00 /bin/bash /usr/local/bin/raspiBackup.sh -a service apache2 start && service supervisor start && service grafana-server start && service mariadb start && service mysql start && service mysqld start && su - pi -c 'node-red-start &' && /etc/init.d/mosquitto start -o service apache2 stop && service supervisor stop && service grafana-server stop && service mariadb stop && service mysql stop && service mysqld stop && node-red-stop && /etc/init.d/mosquitto stop -m detailed
root      2521  0.0  0.0   3536   348 ?        S    03:00   0:00 tee -ia /media/usb/backups/BigPi/BigPi-tgz-backup-20201119-030001/raspiBackup.log
root      2522  0.0  0.0   3536   344 ?        S    03:00   0:00 tee -ia /media/usb/backups/BigPi/BigPi-tgz-backup-20201119-030001/raspiBackup.log
```

Here is the log file:
[https://pastebin.com/ZHgcAkRf](https://pastebin.com/ZHgcAkRf)

framps, 2020-11-19:
:disappointed:  I checked on one of my Raspberries which still runs Stretch like your system and everything is fine. So it looks like it's some kind of configuration or environmental issue you face. According the log raspiBackup finishes successful. 

I following questions now:
1) Does a file /var/lock/raspiBackup still exist?
2) What happens when raspibackup ran the next day? Do you have 4 instances, does raspiBackup not start or does raspiBackup do it's job and you have again 2 instanes?
3) Please check whether raspiBackup is configured in /etc/crontab and/or /etc/cron.d/raspiBackup and please provide the contents of the files if it's configured there.
4) It's just a guess. But you have very long -a and -o parameters. Please define them in /usr/local/etc/raspiBackup.conf. In any case I suggest to keep the commandline options as small as possible in crontab.
5) Did you use the UI installer or the previous command line installer to install raspiBackup?

Kamairo, 2020-11-24:
1. No
2. It starts and runs successfully, but I have indeed additional processes, basically the same as posted above twice, so yes, four in total.
3. It's not configured in either of those files. There is an entry in `/etc/cron.d/raspiBackup` but it's commented out:
```
#
# Crontab entry for raspiBackup.sh
#
# (C) 2017-2019 framp at linux-tips-and-tricks dot de
#
# Create a backup once a week on Sunday morning at 5 am (default)
#
#0 5 * * 0      root    /usr/local/bin/raspiBackup.sh
```
I only ever configured a cronjob with `sudo crontab -e`, which did not change with the version change.
4. Will try it.
5. I think initially I used the command line installer. When I re-installed raspiBackup for this issue, I used the UI installer.

Kamairo, 2020-11-25:
4. Nope, that doesn't change anything.

framps, 2020-11-27:
:cry: It's difficult for me now to help you because I need on site access to your system. But there are some ideas I have how to nail down the issue:
1) Please use `-a : -o :` and check whether the issue is still there.
2) Try option `-l 0` (lower case L) which turns off logging and check whether the issue is still there.
3) Use a different backup target. Right now you use
```/dev/sda1 on /media/usb type fuseblk (rw,nosuid,nodev,noexec,noatime,user_id=0,group_id=0,default_permissions,allow_other,blksize=4096)```
Maybe there is some kind of issue with fuseblk. 
4) Don't use option `-z`. I already had some strange effects with `-z` and tar in the past. Maybe this is another one.

Sorry for the inconvenience.

framps, 2020-12-07:
@Kamairo Any update?

framps, 2020-12-19:
Closing issue because of inactivity

closed 2020-12-19 by framps

Kamairo, 2020-12-19:
Sorry, for the late reply, did not have that much time to try a lot of things on the RPi.
Found the issue today: It's node red's startup script. It's weird, the command "node-red-start" doesn't really return, it just keeps posting to the console.

framps, 2020-12-19:
> Sorry, for the late reply,

No problem :smiley: 

Thank you very much for the update.

Kamairo, 2020-12-23:
And just to add the solution for anyone who has the same issue:
Don't call node-red's startup commands `node-red-start` and `node-red-stop` as they don't seem to return. Instead, use systemctl to start and stop the service via: `systemctl stop nodered.service` and `systemctl start nodered.service`.