framps/raspiBackup #291: Backup with complete system on external hdd

kugelkopf123, 2020-11-28:
If you have a question about raspiBackup please execute all following actions first and provide all requested information:

* Read raspiBackup FAQ's which answers a lot of common questions [English](https://www.linux-tips-and-tricks.de/en/faq) - [German](https://www.linux-tips-and-tricks.de/de/faq)

* Provide the output of `sudo raspiBackup.sh --version` or `sudo raspiBackupInstallUI.sh --version`
Version: 0.6.5.1 CommitSHA: 9cc17fc CommitDate: 2020-10-31 CommitTime: 21:52:43
* Provide the output of `cat /etc/os-release`
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

If any action was not executed or any requested information was not provided the question will be rejected.

I have already looked in the user manual, but have not found a satisfactory answer.
```
I have installed the above mentioned RaspberryOS on my Raspberrypi4b 8GB
On this device I do not use an SD card but only an external USB -HDD
The HDD is partitioned as follows:
/dev/sda1: LABEL_FATBOOT="boot" LABEL="boot" UUID="4AD7-B4D5" TYPE="vfat" PARTUUID="83d6ee50-01"
/dev/sda2: LABEL="rootfs" UUID="2887d26c-6ae7-449d-9701-c5a4018755b0" TYPE="ext4" PARTUUID="83d6ee50-02"
/dev/sda3: LABEL="raspiBak" UUID="591554f9-13a4-4de1-8454-3a77692cb93e" TYPE="ext4" PARTUUID="83d6ee50-03"

With the "normal" backup via RSYNC I get the following error:
--- RBK0009I: raspi4b: raspiBackup.sh V0.6.5.1 (9cc17fc) Sa 28. Nov 16:18:45 CET 2020 gestartet.
--- RBK0128I: Logdatei ist /mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-161842/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackupRSYNC.conf wird benutzt.
??? RBK0013E: Es existieren mehr als zwei Partitionen, die nur mit dem Backuptype DD oder DDZ oder der Option -P gesichert werden können.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0043I: Unvollständiges Backup /mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-161842 in wird gelöscht. Das kann etwas dauern. Bitte Geduld.
--- RBK0049I: Meldungen wurden in /home/pi/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert.
--- RBK0010I: raspi4b: raspiBackup.sh V0.6.5.1 (9cc17fc) Sa 28. Nov 16:19:00 CET 2020 beendet mit Returncode 115.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.

```
I change to the no longer supported partitioned form:
```
--- RBK0009I: raspi4b: raspiBackup.sh V0.6.5.1 (9cc17fc) Sa 28. Nov 16:19:47 CET 2020 gestartet.
--- RBK0128I: Logdatei ist /mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-161944/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackupRSYNC.conf wird benutzt.
!!! RBK0236W: Partitionsorientierter Modus wird nicht mehr weiter gewartet und irgendwann in Zukunft nicht mehr verfügbar sein.
??? RBK0211E: Externe Partition PARTUUID=83d6ee50-01 die an /dev/sda1 gemounted ist wird mit Option -P nicht gesichert.
??? RBK0211E: Externe Partition PARTUUID=83d6ee50-02 die an /dev/sda2 gemounted ist wird mit Option -P nicht gesichert.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0043I: Unvollständiges Backup /mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-161944 in wird gelöscht. Das kann etwas dauern. Bitte Geduld.
--- RBK0049I: Meldungen wurden in /home/pi/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /home/pi/raspiBackup.log gesichert.
--- RBK0010I: raspi4b: raspiBackup.sh V0.6.5.1 (9cc17fc) Sa 28. Nov 16:20:18 CET 2020 beendet mit Returncode 107.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
```

How can I still create a RSYNC backup on my 3rd partition.
Details about the backup plan:
7 Days Rsync backup to 3rd partition of the ext.USB-HDD
1x weekly Tar Backup on QNAP NAS via SMB.

framps, 2020-11-28:
You ask for a feature which will be available in next release of raspiBackup. There is already a beta for next release available. Just give it a try. :wink: 

See [here](https://www.linux-tips-and-tricks.de/de/raspibackupcategorie/587-raspibackup-beta-0-6-6/) and [here](https://www.linux-tips-and-tricks.de/en/all-pages-about-raspibackup/595-raspibackup-how-can-i-install-and-test-the-beta-version/) for details about raspiBackup beta.

kugelkopf123, 2020-11-28:
Okay! I updated and started a normal rsync backup with --ignoreAdditionalPartitions again.
Its working quite long... The 2 Partitions are perhaps only about 8gb in total. 
I looked at iotop and pgrep:

```
28690 be/4 root       63.06 K/s    0.00 B/s  0.00 % 10.15 % rsync --exclud~20201128-170450
28692 be/4 root        0.00 B/s   55.64 K/s  0.00 %  1.10 % rsync --exclud~20201128-170450
```
The data transfer is really slow. Max 1MB/s.
Pgrep:
```
pi@raspi4b:/mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450 $ pgrep rsync -al
28690 rsync --exclude=/mnt/raspiBak/raspiBackupRSYNC --exclude=/mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450/raspiBackup.log --exclude=/mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450/raspiBackup.msg --exclude=.gvfs --exclude=/proc/* --exclude=/lost+found/* --exclude=/sys/* --exclude=/dev/* --exclude=/boot/* --exclude=/tmp/* --exclude=/run/* --exclude=/var/cache/* --exclude=/raspi4b-backup.img --exclude=/raspi4b-backup.tmg --exclude=/raspi4b-backup.sfdisk --exclude=/raspi4b-backup.mbr --exclude=/raspiBackup.log --exclude=/raspiBackup.msg /var/swap /mnt /media/pi/ --numeric-ids -aHAx --delete / /mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450
28691 rsync --exclude=/mnt/raspiBak/raspiBackupRSYNC --exclude=/mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450/raspiBackup.log --exclude=/mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450/raspiBackup.msg --exclude=.gvfs --exclude=/proc/* --exclude=/lost+found/* --exclude=/sys/* --exclude=/dev/* --exclude=/boot/* --exclude=/tmp/* --exclude=/run/* --exclude=/var/cache/* --exclude=/raspi4b-backup.img --exclude=/raspi4b-backup.tmg --exclude=/raspi4b-backup.sfdisk --exclude=/raspi4b-backup.mbr --exclude=/raspiBackup.log --exclude=/raspiBackup.msg /var/swap /mnt /media/pi/ --numeric-ids -aHAx --delete / /mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450
28692 rsync --exclude=/mnt/raspiBak/raspiBackupRSYNC --exclude=/mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450/raspiBackup.log --exclude=/mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450/raspiBackup.msg --exclude=.gvfs --exclude=/proc/* --exclude=/lost+found/* --exclude=/sys/* --exclude=/dev/* --exclude=/boot/* --exclude=/tmp/* --exclude=/run/* --exclude=/var/cache/* --exclude=/raspi4b-backup.img --exclude=/raspi4b-backup.tmg --exclude=/raspi4b-backup.sfdisk --exclude=/raspi4b-backup.mbr --exclude=/raspiBackup.log --exclude=/raspiBackup.msg /var/swap /mnt /media/pi/ --numeric-ids -aHAx --delete / /mnt/raspiBak/raspiBackupRSYNC/raspi4b/raspi4b-rsync-backup-20201128-170450
```
What is strange, i added the folders /mnt /media/pi/ and /var/swap in my raspiBackup.conf file. But it seams, that its not the right syntax to exclude it?

framps, 2020-11-28:
> --ignoreAdditionalPartitions

That's an option I added on user request because he had multiple partitions but only wanted to backup the first two partitions. As far as I understand you want to save the third partition - which is **NOT** saved with this option. That's possible with the next release I mentioned before.

> The data transfer is really slow. Max 1MB/s

That's out of scope of raspiBackup. Looks like you use a network connection which is quite slow. Also keep in mind the first rsync backup will take much longer because everything is saved. Later backups will only save the modified and new files.

> What is strange, i added the folders /mnt /media/pi/ and /var/swap in my raspiBackup.conf file. But it seams, that its not the right syntax to exclude it?

Please check [this page](https://www.linux-tips-and-tricks.de/en/backup/#parm_u).

kugelkopf123, 2020-11-29:
Thank you very much for your quick answer.
I will try to explain it more clearly once again. 

My system is on an external hard disk with the partitions:
Boot,Rootfs,RaspiBak.

My Raspberrypi4b runs completely without an SD card.
It was installed directly on an external USB3-HDD.
Afterwards I added another partition via gparted. This is the partition RaspiBak.

I want to create daily rSync backups to exactly this RaspiBak partition. (Note that this partition is on the same USB-HDD).
These RSync backups should rotate over 7 days.
Once a week I want to make a backup to an external NAS. But this will be a TAR backup.

So what are the right options.
By the way, I am not completely new to Raspibackup. I've been using it in a different environment for a long time to backup a Raspi to another NAS. 

Thanks!

framps, 2020-11-30:
raspiBackup refuses to store a backup on the same device the system runs on. Given this you will not be able to save the system on RaspiBak. But there is an undocumented option `-c` available which turns off this check :wink: 

Said this you should use following options with the current 0.6.6 beta to test it works and as soon as olfficial released 0.6.6 is available use this release in production.

`-P` -t rsync -T "1 2 3"

kugelkopf123, 2020-12-04:
@framps 
Nice, that works!
It seems that a backup created on the same HDD, whether rSync or Tar, is incredibly slow. As a test I have made a backup on another USB2 Stick instead of on the same HDD. The same settings were used as before when backing up to the USB3 HDD but instead of saving to the 3rd partition it was saved to the "slow" USB stick. The backup to the USB stick took just over 30 minutes. So... much faster.

When benchmarking the USB3HDD itself, I could not find any errors. There I reach usual speed values. 

Also with the incremented backup (on 3rd partition) via rSync the backup takes unbelievably long. Via ioTop I was able to determine that the jbd2 service causes a 99% io load during the backup.

So far so my findings.

framps, 2020-12-04:
Glad it works now for you. Keep in mind to make sure to copy the backup from the HDD to some external storage from time to time. If your disk dies your backups are gone :cry:  

>  whether rSync or Tar, is incredibly slow.

That's not a raspiBackup issue. raspiBackup just uses the standard backup tools rsync and tar. Keep in mind  the head of the disk has to move from read position to write position all the time which takes time ... I think it will be much faster if you use a SSD.

kugelkopf123, 2020-12-04:
Thanks a lot for your help! 

I will try again with an SSD. Just for fun. =)

Right. That's why the strategy is to have daily rotating rsync backups on the 3rd partition to be able to recover certain data quickly and weekly TAR backups on a NAS.

closed 2020-12-04 by kugelkopf123