framps/raspiBackup #292: COMMAND_EXIT_CODE

christianTF, 2020-12-04:
Hi framp,

I've can reproduce the problem with the raspiBackup commit shown in the log.
As I thought, the fstab extension fails, I've added the debugging line `Returncode:` to the extension. But the extension correctly returns 0, but the `[COMMAND_EXIT_CODE="1"]`` is still set.

I've searched your repo at Github, and raspiBackup.sh from master and beta, and cannot find the string `COMMAND_EXIT_CODE`anywhere.

Could you give me any advice, that this COMMAND_EXIT_CODE stands for?

Thanks,
Christian

```
script started on 2020-12-04 09:17:19+01:00 []
--- RBK0009I: fepi: raspiBackup.sh V0.6.6-beta (53096c4) started at Fri Dec 4 09:17:26 CET 2020.
!!! RBK0165W: =========> NOTE <=========
!!! RBK0165W: This is a betaversion and should not be used in production.
!!! RBK0165W: =========> NOTE <=========
--- RBK0128I: Using logfile /opt/loxberry/log/plugins/lbbackup/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
!!! RBK0124W: Fake mode on.
--- RBK0151I: Using backuppath /media/smb/homeserver/Raspberry-Backups with partition type cifs.
!!! RBK0157W: No services to stop.
--- RBK0081I: Creating backup of type ddz in /media/smb/homeserver/Raspberry-Backups/fepi/fepi-ddz-backup-20201204-091720.
--- RBK1005I: Copy /etc/fstab to /media/smb/homeserver/Raspberry-Backups/fepi/fepi-ddz-backup-20201204-091720/raspiBackup/extensionSaveArea
total 1
drwxrwxr-x 2 loxberry loxberry 0 Dec 4 09:17 .
drwxrwxr-x 2 loxberry loxberry 0 Dec 4 09:17 ..
-rw-rw-r-- 1 loxberry loxberry 628 Dec 4 2020 fstab
Returncode: 0
--- RBK0085I: Backup of type ddz started. Please be patient.
--- RBK0078I: Backup time: 00:00:00.
--- RBK0159I: 3 backups kept for ddz backup type.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0049I: Messages saved in /opt/loxberry/log/plugins/lbbackup/raspiBackup.msg.
--- RBK0026I: Debug logfile saved in /opt/loxberry/log/plugins/lbbackup/raspiBackup.log.
--- RBK0010I: fepi: raspiBackup.sh V0.6.6-beta (53096c4) stopped at Fri Dec 4 09:17:54 CET 2020 with rc 0.
--- RBK0017I: Backup finished successfully.

Script done on 2020-12-04 09:17:59+01:00 [COMMAND_EXIT_CODE="1"]
```

LoxBerry Backup then says,
`ERROR: Backup failed with error code 1`

christianTF, 2020-12-04:
PS: Currently I'm a bit confused, as the lines
`script started on 2020-12-04 09:17:19+01:00 []`
and 
`Script done on 2020-12-04 09:17:59+01:00 [COMMAND_EXIT_CODE="1"]`
neither seem to come from my nor from your code.....

christianTF, 2020-12-04:
I've found:
My call is
`system("script --quiet --flush --return --append " . $log->filename() . " --command '" . $paramstr . "'");`
The `script` tool was my workaround to capture your output to tty (instead of stderr/stdout).

I'll further investigate...

framps, 2020-12-04:
> The `script` tool was my workaround to capture your output to tty (instead of stderr/stdout).

In former times I didn't use sterr and stdout in raspiBackup. But because you told me about your issue I changed the code at some point in time and all stuff is written to stdout and stderr. So i think you don't need any workaround any more. Just redirect or pipe the output of raspiBackup :wink:

christianTF, 2020-12-04:
Yes, I will do some changes on installing (-> Release) and running (piping to a file) the backup.

In a second step, I will add a Release/Beta switching option.

closed 2020-12-04 by christianTF

framps, 2020-12-04:
Sounds good :+1: 

Just let me know if you face an issue with the stdout and/or stderr redirection.