framps/raspiBackup #293: Beta: RC 1 on successful backup

christianTF, 2020-12-04:
Hi framp,

i still get an return code 1, also with directly starting the backup from shell:
```
root@loxberrypi3a:/opt/loxberry/bin/plugins/lbbackup# /usr/local/bin/raspiBackup.sh -F -o ":" -a ":" -k 3 -t ddz -L current -l debug -m detailed -v /opt/loxberry/system/storage/smb/homeserver/Raspberry-Backups
--- RBK0009I: loxberrypi3a: raspiBackup.sh V0.6.6-beta (53096c4) started at Fri  4 Dec 11:28:04 CET 2020.
!!! RBK0165W: =========> NOTE  <=========
!!! RBK0165W: This is a betaversion and should not be used in production.
!!! RBK0165W: =========> NOTE <=========
--- RBK0128I: Using logfile /opt/loxberry/bin/plugins/lbbackup/raspiBackup.log.
--- RBK0116I: Using config file /usr/local/etc/raspiBackup.conf.
!!! RBK0124W: Fake mode on.
--- RBK0151I: Using backuppath /media/smb/homeserver/Raspberry-Backups with partition type cifs.
!!! RBK0157W: No services to stop.
--- RBK0081I: Creating backup of type ddz in /media/smb/homeserver/Raspberry-Backups/loxberrypi3a/loxberrypi3a-ddz-backup-20201204-112803.
--- RBK0085I: Backup of type ddz started. Please be patient.
--- RBK0078I: Backup time: 00:00:00.
--- RBK0159I: 3 backups kept for ddz backup type.
--- RBK0033I: Please wait until cleanup has finished.
--- RBK0049I: Messages saved in /opt/loxberry/bin/plugins/lbbackup/raspiBackup.msg.
--- RBK0026I: Debug logfile saved in /opt/loxberry/bin/plugins/lbbackup/raspiBackup.log.
--- RBK0010I: loxberrypi3a: raspiBackup.sh V0.6.6-beta (53096c4) stopped at Fri  4 Dec 11:28:11 CET 2020 with rc 0.
--- RBK0017I: Backup finished successfully.
root@loxberrypi3a:/opt/loxberry/bin/plugins/lbbackup# echo $?
1
root@loxberrypi3a:/opt/loxberry/bin/plugins/lbbackup#

```
I've attached the full log. 
Is this a known beta issue?

lg, Christian

[raspiBackup.log](https://github.com/framps/raspiBackup/files/5642077/raspiBackup.log)

framps, 2020-12-04:
> Is this a known beta issue?

No. Unfortunately the final rc was not passed to the OS :disappointed: . It's fixed now :wink: . Please use `raspiBackup.sh -U -S` to grab the fixed version.

christianTF, 2020-12-04:
Fixed! Thanks!
I think, for now this will also fix the issue of the LoxBerry users.
But I already have changed things in my code, but this needs somre more testing.

closed 2020-12-04 by christianTF