framps/raspiBackup #295: Backup Question

swvalenti194, 2020-12-20:
Trying to write directly to the SD card but get the below error.  Do I have to write to USB or can I write to existing card?  Trying to write to a Buffalo NAS drive on my network but can't seem to get that to work so just wanted a quick backup instead.  

`pi@raspberrypi:~ $ curl -sSLO https://www.linux-tips-and-tricks.de/raspiBackup.sh
pi@raspberrypi:~ $ sudo bash ./raspiBackup.sh /media/pi
??? RBK0149E: File /media/pi does not exist.
pi@raspberrypi:~ $ 

`

framps, 2020-12-20:
When you execute the curl command the file `raspiBackup.sh` is created in the current directory. Then you have to invoke it with 
```
sudo ./raspiBackup.sh
```

Why don't you user the installer? Manual installation is described [here](https://www.linux-tips-and-tricks.de/en/all-pages-about-raspibackup/539-raspibackup-manual-installation/) in detail.

swvalenti194, 2020-12-20:
Ok so I used the installer but how do I execute a backup now?  Sorry I am new to RPi and trying to get the hang of this.

framps, 2020-12-20:
When you used the installer just use
```
sudo raspiBackup.sh <any parameters>
```
and the parameters are [described on this page](https://www.linux-tips-and-tricks.de/en/backup/)

swvalenti194, 2020-12-20:
I added below to `sudo nano /etc/fstab`

`//10.30.1.233/passport/mnt/share /backup  cifs user=XXXX,password=XXXX,sec=ntlm,iocharset=utf8,vers=1.0 defaults 0     0

I reboot and run `findmnt` and it doesn't appear
`

framps, 2020-12-20:
> Sorry I am new to RPi

Everybody once started with RPi and Linux :smile: 

Please use some the excellent Linux and/or Raspbian Foren and ask for help on your Linux questions. I give support for raspiBackup but when I start to give support for every Linux related question I'll busy all the day :wink:

framps, 2020-12-20:
Please see also [FAQ47](https://www.linux-tips-and-tricks.de/en/faq/#a47)

closed 2020-12-23 by framps