framps/raspiBackup #296: RBK0178E: Erzeugung von /backup/... Datei endet fehlerhaft mit RC 1.

PatrickGlatz, 2020-12-29:
Hallo @framps 

ich nutze schon seit Jahren dein raspiBackup und bin super happy damit. Nun musste ich die IP und den Hostname meines Pi wechseln, alles andere ist gleich geblieben. Allerdings läuft seit 07.12.2020 mein Backup nicht mehr

Ich rufe das Skript über ein root crontab auf:
  00 03  *   *   1     mount -t nfs -o soft 192.169.179.165:/volume1/Sicherungen/rsync /backup && /usr/local/bin/raspiBackup.sh --systemstatus

Seit dem 07.12.2020 (letzter erfolgreicher Backup) erhalte ich nun leider folgenden log:

```
[19:29:46] gfadmin@openhabpiold:~$ sudo mount -t nfs -o soft 192.169.179.165:/volume1/Sicherungen/rsync /backup && sudo /usr/local/bin/raspiBackup.sh --systemstatus
--- RBK0009I: openhabpiold: raspiBackup.sh V0.6.6 (7989828) Di 29. Dez 19:35:27 CET 2020 gestartet.
--- RBK0128I: Logdatei ist /backup/openhabpiold/openhabpiold-rsync-backup-20201229-193526/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
--- RBK0031I: Prüfe ob eine neue Version von raspiBackup.sh verfügbar ist.
--- RBK0151I: Backuppfad /backup mit Partitionstyp nfs4 wird benutzt.
--- RBK0008I: Services werden gestoppt: 'systemctl stop telegraf.service && systemctl stop influxdb.service'.
--- RBK0081I: Backup vom Typ rsync wird in /backup/openhabpiold/openhabpiold-rsync-backup-20201229-193526 erstellt.
--- RBK0036I: Partitionslayout wird gesichert.
--- RBK0044I: Backup der Bootpartition wird in /backup/openhabpiold/openhabpiold-rsync-backup-20201229-193526/openhabpiold-backup.img erstellt.
dd: Fehler beim Lesen von '/dev/mmcblk0p1': Eingabe-/Ausgabefehler
122+1 Datensätze ein
122+1 Datensätze aus
128028672 bytes (128 MB, 122 MiB) copied, 151,269 s, 846 kB/s
??? RBK0178E: Erzeugung von /backup/openhabpiold/openhabpiold-rsync-backup-20201229-193526/openhabpiold-backup.img Datei endet fehlerhaft mit RC 1.
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0007I: Services werden gestartet: 'systemctl start influxdb.service && systemctl start telegraf.service'.
--- RBK0043I: Unvollständiges Backup /backup/openhabpiold/openhabpiold-rsync-backup-20201229-193526 in wird gelöscht. Das kann etwas dauern. Bitte Geduld.
--- RBK0049I: Meldungen wurden in /home/gfadmin/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /home/gfadmin/raspiBackup.log gesichert.
--- RBK0010I: openhabpiold: raspiBackup.sh V0.6.6 (7989828) Di 29. Dez 19:38:11 CET 2020 beendet mit Returncode 114.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
--- RBK0167I: Eine eMail wird versendet.
--- RBK0229I: Telegram benachrichtigt.
```

Ein df und mount und  zeigt folgendes Bild:
```
[19:49:04] gfadmin@openhabpiold:~$ df
Dateisystem                               1K-Blöcke   Benutzt  Verfügbar Verw% Eingehängt auf
/dev/root                                  30400236  16960896   12171968   59% /
devtmpfs                                    1853084         0    1853084    0% /dev
tmpfs                                       1985180         0    1985180    0% /dev/shm
tmpfs                                       1985180     50740    1934440    3% /run
tmpfs                                          5120         4       5116    1% /run/lock
tmpfs                                       1985180         0    1985180    0% /sys/fs/cgroup
/dev/mmcblk0p1                               258096     56122     201974   22% /boot
tmpfs                                        397036         0     397036    0% /run/user/1000
192.169.179.165:/volume1/Sicherungen/rsync 3746098176 747281280 2998816896   20% /backup

[19:50:03] gfadmin@openhabpiold:~$ mount
/dev/mmcblk0p2 on / type ext4 (rw,noatime)
devtmpfs on /dev type devtmpfs (rw,relatime,size=1853084k,nr_inodes=97615,mode=755)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
proc on /proc type proc (rw,relatime)
securityfs on /sys/kernel/security type securityfs (rw,nosuid,nodev,noexec,relatime)
tmpfs on /dev/shm type tmpfs (rw,nosuid,nodev)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,gid=5,mode=620,ptmxmode=000)
tmpfs on /run type tmpfs (rw,nosuid,nodev,mode=755)
tmpfs on /run/lock type tmpfs (rw,nosuid,nodev,noexec,relatime,size=5120k)
tmpfs on /sys/fs/cgroup type tmpfs (ro,nosuid,nodev,noexec,mode=755)
cgroup2 on /sys/fs/cgroup/unified type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
cgroup on /sys/fs/cgroup/systemd type cgroup (rw,nosuid,nodev,noexec,relatime,xattr,name=systemd)
none on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,relatime,mode=700)
cgroup on /sys/fs/cgroup/cpu,cpuacct type cgroup (rw,nosuid,nodev,noexec,relatime,cpu,cpuacct)
cgroup on /sys/fs/cgroup/devices type cgroup (rw,nosuid,nodev,noexec,relatime,devices)
cgroup on /sys/fs/cgroup/freezer type cgroup (rw,nosuid,nodev,noexec,relatime,freezer)
cgroup on /sys/fs/cgroup/net_cls,net_prio type cgroup (rw,nosuid,nodev,noexec,relatime,net_cls,net_prio)
cgroup on /sys/fs/cgroup/perf_event type cgroup (rw,nosuid,nodev,noexec,relatime,perf_event)
cgroup on /sys/fs/cgroup/cpuset type cgroup (rw,nosuid,nodev,noexec,relatime,cpuset)
cgroup on /sys/fs/cgroup/pids type cgroup (rw,nosuid,nodev,noexec,relatime,pids)
cgroup on /sys/fs/cgroup/blkio type cgroup (rw,nosuid,nodev,noexec,relatime,blkio)
debugfs on /sys/kernel/debug type debugfs (rw,relatime)
sunrpc on /run/rpc_pipefs type rpc_pipefs (rw,relatime)
systemd-1 on /proc/sys/fs/binfmt_misc type autofs (rw,relatime,fd=44,pgrp=1,timeout=0,minproto=5,maxproto=5,direct)
mqueue on /dev/mqueue type mqueue (rw,relatime)
configfs on /sys/kernel/config type configfs (rw,relatime)
/dev/mmcblk0p2 on /srv/openhab2-addons type ext4 (rw,noatime)
/dev/mmcblk0p2 on /srv/openhab2-conf type ext4 (rw,noatime)
/dev/mmcblk0p2 on /srv/openhab2-sys type ext4 (rw,noatime)
/dev/mmcblk0p2 on /srv/openhab2-userdata type ext4 (rw,noatime)
/dev/mmcblk0p2 on /srv/openhab2-logs type ext4 (rw,noatime)
/dev/mmcblk0p1 on /boot type vfat (rw,relatime,fmask=0022,dmask=0022,codepage=437,iocharset=ascii,shortname=mixed,errors=remount-ro)
tmpfs on /run/user/1000 type tmpfs (rw,nosuid,nodev,relatime,size=397036k,mode=700,uid=1000,gid=1000)
192.169.179.165:/volume1/Sicherungen/rsync on /backup type nfs4 (rw,relatime,vers=4.1,rsize=131072,wsize=131072,namlen=255,soft,proto=tcp,timeo=600,retrans=2,sec=sys,clientaddr=192.169.179.166,local_lock=none,addr=192.169.179.165)
```

raspiBackup Version war bisher die 0.6.5-beta (lief lange problemlos), habe ich jetzt nach dem Fehler aber upgegraded zur 0.6.6 (leider gleicher Fehler). Seitdem ist auf dem System aber auch ein `apt-get update && apt-get upgrade`gelaufen. Will nicht ausschließen, dass da was schief gegangen ist.

Während das Backup läuft, kann ich auf dem Zielsystem (Synology DS216+ii) sehen, dass ein Ordner erstellt wird, der wird später aber wie gesagt wieder aufgeräumt.

Kannst Du mir hier aushelfen? Ich weiß nicht wo ich ansetzen soll

Herzlichen Dank und viele Grüße
Patrick

framps, 2020-12-30:
```
dd: Fehler beim Lesen von '/dev/mmcblk0p1': Eingabe-/Ausgabefehler
```
Das bedeutet dass dd Lese- oder Schreibfehler beim Sichern von /dev/mmcblk0p1 bekommt. Dummerweise sagt dd nicht ob es ein Schreib- oder Lesefehler ist.

Rufe mal folgenden Befehl auf:

`sudo dd if=/dev/mmcblk0p1 of=/tmp/dd.img bs=2M`

Damit erzeugst Du eine Kopie von /dev/mmcblk0p1. Wenn Du wieder einen RC 1 bekommst ist es ein Lesefehler. Das bedeutet dann das offensichtlich Deine SD Karte defekt ist und ersetzt werden sollte. Wenn kein fehler RC 1 auftritt gibt es Schreibprobleme auf den Backupspace. Warum es die gibt musst Du genauer untersuchen. ich vermute aber dass Deine SD Karte das Zeitliche gesegnet hat.

PatrickGlatz, 2021-01-01:
Hey @framps,

das dachte ich mir schon fast. Habe ein dd auf tmp gemacht und auch hier bekomme ich einen Lesefehler. Ein Ausfall der Karte ist mittlerweile auch wahrscheinlich geworden, sie läuft seit 2 Jahren 24/7 mit relativ viel logging (dämlicher Weise). Werde versuchen das letzte erfolgreiche Backup zu restoren.

Vielen Dank für deine Hilfe.

P.S.: Gibt es eine Möglichkeit, raspiBackup ohne das backup des root file systems zu starten, sodass ich den letzten Zustand der Files nochmal wegspeichere?

Viele Grüße

framps, 2021-01-01:
> P.S.: Gibt es eine Möglichkeit, raspiBackup ohne das backup des root file systems zu starten, sodass ich den letzten Zustand der Files nochmal wegspeichere?

Nicht direkt. Du kannst Dir aber aus deinem letzten Backuplauf aus dem Debuglog die Zeile rausholen die Du benutzen musst um Dein /root zu sichern. Bei mir sieht es z.B. wie folgt aus:

```
--> executeCmd Command: rsync --exclude="/backup/backup/rsync"                   --exclude="/backup/backup/rsync/@HOSTNAME@/@HOSTNAME@-rsync-backup-20201229-050001/raspiBackup.log"     
                --exclude="/backup/backup/rsync/@HOSTNAME@/@HOSTNAME@-rsync-backup-20201229-050001/raspiBackup.msg"                     --exclude='.gvfs'                       --exclude=/proc/*                       --exclude=/lost+found/*                         --exclude=/sys/*                        --exclude=/dev/*                        --exclude=/boot/*                       --exclude=/tmp/*                        --exclude=/run/*                        --exclude=/var/cache/*                  --exclude=/@HOSTNAME@-backup.img --exclude=/@HOSTNAME@-backup.tmg --exclude=/@HOSTNAME@-backup.sfdisk --exclude=/@HOSTNAME@-backup.mbr --exclude=/raspiBackup.log --exclude=/raspiBackup.msg                     
                       --link-dest="/backup/backup/rsync/@HOSTNAME@/@HOSTNAME@-rsync-backup-20201222-102208"                   --numeric-ids                   -aHAx --delete                                                          
        /                       "/backup/backup/rsync/@HOSTNAME@/@HOSTNAME@-rsync-backup-20201229-050001"     
```
Dort musst Du `executeCmd Command: ` loeschen und die notwendigen Verzeichnisse anlegen bzw anpassen. Beim Restore musst Du dann auch etwas trichsen: Du musst das /root Verzeichnis mit einem Backup von /dev/mmcblk0p1 ergaenzen. Ist aber etwas Frickelarbeit.

Viel Glueck !

framps, 2021-01-15:
@PatrickGlatz Hast Du Erfolg gehabt? Brauchst Du noch Support? Sonst bitte den Issue closen :wink:

PatrickGlatz, 2021-01-19:
Hi @framps,

sorry, wollte den issue nicht von mir aus closeness, aber der restore hat funktioniert, die alte SD Karte hab ich weggehauen und die Backups laufen wieder 👍

Hab dir eine kleine Aufmerksamkeit via Paypal für dein großartiges Tool und deinen Mega Support zukommen lassen.

Vielen Dank für alles und viele Grüße

closed 2021-01-19 by PatrickGlatz

framps, 2021-01-19:
> wollte den issue nicht von mir aus closeness

Ich mache sie nur zu wenn keine Reaktionen mehr kommen. Ansonsten closed fuer mich derjenige der in geoeffnethat :wink: 

> Hab dir eine kleine Aufmerksamkeit via Paypal

Ist angekommen. Vielen Dank :smiley: