framps/raspiBackup #297: Telegram message - better highlight when backup failed

Steinardo, 2020-12-29:
Hello Framp

I use raspiBackup in compination with the Telegram messenger. I make every week a backup but the last three backups failed (not a problem from raspiBackup). I didn't checked the Telegram message well and i miss read the text. I know that you already send the text in bold and underlined but this was not enough for me to see that something is wrong. The text "Backup erfolgreich beendet" and " __**Backup nicht erfolgreich!!**__ " was too similar for my eyes when spending 1 second to read the message.

Therefore i would like to raise a change in the message. If possible please add the ⚠️ emoji and the prefix warning. For example "⚠️ Warung: Backup nicht erfolgreich!!". The emoji will catch directly the eye that something is different.

Best regards and keep up the great work.

Steinardo

framps, 2020-12-30:
Thank you very much for your issue. I think it's a useful enhancement request for next release.

sikercito, 2021-01-11:
Hi!

Thank you very much for the useful tool. I know that my request is a lot of work but I would like to know if in future releases could be possible to edit the message that is sent to Telegram through the config file. I think that I can do it modifying code lines 464 and 467 but it won't persist when an update comes... Thanks!

framps, 2021-01-11:
I plan to change the message in next release. But I don't plan to make the message configurable. Why don't you just let me know which message you would like to have :wink:

sikercito, 2021-01-11:
I'm afraid that it would be useless for the other users: I have a private channel in telegram where I receive notifications from the different services that I have running in my Raspberry (openVpn connections, changes of my public ip...) so each one has my raspberry and the service names in bold, then a jump line, and the message. But don't worry, i think I can make an script to automatize the message when you update the app :)

framps, 2021-06-10:
I created a fix for your issue. You now get an exclamation mark when the backup failed. 

Please verfify the fix and grab this file and copy it into your home directory, e.g. /home/pi. Execute chmod +x raspiBackup.sh. Then call sudo ./raspiBackup.sh (note the leading period !) .

Steinardo, 2021-06-10:
Check it and I like it. The visibility when a backup is not successful is better. Thank you for implementing this.

Here is a screenshot from before and after.
[image]

closed 2021-06-10 by Steinardo

framps, 2021-06-10:
Thank you very much you verified the issue :+1: