framps/raspiBackup #298: RBK0077E: Restore failed

Jasimo, 2021-01-07:
Guten Morgen,
ich wollte heute mal mein auf einem Raspi4 erzeugtes Backup testen und auf eine gleich große SD (32GB) restoren.
Den Restore mache ich auf einem anderen Raspberry, die Ziel SD steckt in einem Cardreader am USB Port.
Mittels dem folgenden Aufruf starte ich den Restore:
sudo raspiBackup.sh -d /dev/sda /mnt/backup/LXC-Server/LXC-Server-tgz-backup-20210107-081024

im Log steht folgendes:

^[[1;33m!!! RBK0065W: Device /dev/sda will be repartitioned and all data will be lost.^[[0m
--- RBK0067I: Current partitions on /dev/sda:
Number  Start   End      Size     Type     File system  Flags
 1      4,19MB  62,9MB   58,7MB   primary  fat16        lba
 2      62,9MB  31915MB  31852MB  primary  ext4
--- RBK0066I: Device /dev/sda will be overwritten with the saved boot and root partition.
--- RBK0069I: Bootpartition /dev/sda1 will be formatted and will get the restored Boot partition.
--- RBK0070I: Rootpartition /dev/sda2 will be formatted and will get the restored Root partition.
--- RBK0038I: Are you sure? y/N
--- RBK0050I: Restoring backup from /mnt/backup/LXC-Server/LXC-Server-tgz-backup-20210107-081024.
umount: /dev/sda1: not mounted
umount: /dev/sda2: not mounted
sfdisk: Checking that no-one is using this disk right now ...
sfdisk: OK
sfdisk: unrecognized input: dos

Disk /dev/sda: 30436 cylinders, 64 heads, 32 sectors/track
Old situation:
Units: cylinders of 1048576 bytes, blocks of 1024 bytes, counting from 0

   Device Boot Start     End   #cyls    #blocks   Id  System
/dev/sda1          4      59      56      57344    c  W95 FAT32 (LBA)
/dev/sda2         60   30436-  30377-  31105536   83  Linux
                end: (c,h,s) expected (1023,63,32) found (740,31,32)
/dev/sda3          0       -       0          0    0  Empty
/dev/sda4          0       -       0          0    0  Empty
--- RBK0033I: Please wait until cleanup has finished.
umount: /dev/sda1: not mounted
umount: /dev/sda2: not mounted
--- RBK0010I: LXC-Server: raspiBackup.sh V0.6.6 (7989828) stopped at Do 7. Jan 09:16:06 CET 2021 with rc 112.
^[[1;31m??? RBK0077E: Restore failed. Check previous error messages.^[[0m

Viele Grüße
Jan

Jasimo, 2021-01-07:
Hab mal auf dem Raspi der restoren soll ein downgrade auf V0.6.5 gemacht (vorher V0.6.6) dadurch werden im restore prozess komischwerweise mehr infos ins Log geschrieben. Log hab ich angehängt, irgendwas klappt da beim anlegen der Partitionen durch sfdisk auf sda nicht
[raspiBackup.log](https://github.com/framps/raspiBackup/files/5782459/raspiBackup.log)

framps, 2021-01-07:
Vielen Dank fuer das Log. Das hilft mir sehr :+1: 

>  irgendwas klappt da beim anlegen der Partitionen durch sfdisk auf sda nicht. Version von sfdisk util-linux 2.25.2

Im Log habe ich gesehen dass Du zum Restore Jessie benutzt. Auf Buster ist es `sfdisk from util-linux 2.33.1`. Kann es sein dass Du das Backup auf Buster oder Stretch erstellt hast?

Jasimo, 2021-01-07:
Hallo,

ja korrekt Backup ist von einem Buster System, der Restore findet unter Jessi statt. Ich kann aber ja mal auch den Restore unter Buster probieren.

Jasimo, 2021-01-07:
Restore unter Buster klappt!

framps, 2021-01-07:
Ich erinnerte mich dunkel dass schon mal jemand das Problem hatte. Deshalb habe ich ja gleich im Debuglog nach der OS Release gesehen :smile: . Man sollte eben mit demselben OS Release restoren mit dem man das Backup erzeugt hat :wink: 

Wenn jetzt alles OK fuer Dich ist bitte den Issue closen.

Jasimo, 2021-01-07:
ok, wusste ich nicht.
Danke dir für deine Hilfe und für das tolle Tool!

closed 2021-01-07 by Jasimo

framps, 2021-01-07:
> ok, wusste ich nicht.

Kein Problem.