framps/raspiBackup #299: Request: rclone support

alexschomb, 2021-01-08:
Dear @framps,

first I'd like to thank you for providing raspiBackup to us which is a really easy and reliable tool for Raspberry Pi backups. As I saw that you're in the middle of refactoring the codebase I'd like to add a feature request. It would be really helpful if you could add functionality that implements or even automates the usage of rclone repositories for raspiBackup. It can be done manually (important: you'd need to use tar or dd mode, because soft links are not supported!), but creating the mount service and all the other configuration takes quite a lot time. Looking at ``rclone-sync.sh`` of https://github.com/pageauc/rclone4pi I think it should be possible to raspiBackup directly onto a rclone repository without even mounting it: ``/usr/bin/rclone $rcloneParam -v $localDir $rcloneName:$remoteDir``

Best regards

Alex

framps, 2021-01-09:
raspiBackup is very extensible (See [here](https://www.linux-tips-and-tricks.de/en/backup/#wrapper)). You can wrap raspiBackup in another script and do any pre- and postprocessing before the backup starts and after the backup finished. In you case all the required mount stuff can be executed there. In addition pre- and post plugins can be written and plugged in into raspiBackup (See [here](https://www.linux-tips-and-tricks.de/en/13-raspberry/443-raspibackup-extensions/))

If you create plugins or a wrapper for rclone-sync just create an issue here and of course I will answer any questions related to  plugins/wrappers and help to get this code out of the door. But I will not write and test any code and integrate it into raspiBackup. If requested i will add the code to the raspiBackup repo such that other raspiBackup users can also use rclone-sync.

framps, 2021-01-23:
Are you fine with my answer? If yes please close the issue :wink:

alexschomb, 2021-01-26:
Hi, thanks. Yes, I am fine. Unfortuantely I don't have enough time to implement a full plugin, but thanks for your hints. When there is more time I will try to come back to your suggestion!

closed 2021-01-26 by alexschomb