framps/raspiBackup #300: Is it possible to use Freenas zfs filesystem for backups with hardlinks?

AnduriI, 2021-01-17:
Hey there,

I have a setup running using 6 or 7 raspi and one of them with an usb-hdd attached. This is running fine for now, but I don't like the idea of having all my backups only on a single drive.
As I have setup a Freenas for all my private data I don't want to be lost somehow I also thought of chaning the backup to this location. Is it possible to use hardlinks on that type of filesystem? Maybe that would be an improvement anyone could benefit.
Having them on a raid system cancels out the risk of drive failure...

Thank you in advance,
Anduril

framps, 2021-01-17:
Hello Anduril,

as far as I understand a Freenas can export an EXT3/4 filesystem via nfs. That's all what you need. :wink: 

Cu framp

framps, 2021-02-05:
Any further questions? Otherwise please close the issue.

AnduriI, 2021-02-07:
Hey framps,

sorry for the delay, I was testing this for different RPis. Still it does not work. I think the fs that Freenas exports is not compatible with hardlinks.
I get several times this `df: /media/nfs: Eingabe-/Ausgabefehler` during raspiBackup. I added the raspiBackup.log and .msg (had to rename it to .msg.log for upload), maybe you can see a reason there...

[raspiBackup.msg.log](https://github.com/framps/raspiBackup/files/5938969/raspiBackup.msg.log)
[raspiBackup.log](https://github.com/framps/raspiBackup/files/5938968/raspiBackup.log)

framps, 2021-02-07:
```
20210204-122041 DBG 2630:              --> getFsType /mnt/Backup2
df: /media/nfs: Eingabe-/Ausgabefehler
20210204-122041 DBG 2636:              <-- getFsType nfs
20210204-122041 DBG 5537:          --- Found partitions on /dev/mmcblk0: 2
20210204-122041 DBG 5571:          --- SMART_RECYCLE_PARMS: 2 8 18 3
20210204-122041 DBG 5572:          --- smart recycle parms: 4
20210204-122041 DBG 2574:              --> supportsHardlinks /mnt/Backup2
20210204-122041 DBG 2582:              --- Links: 2
20210204-122041 DBG 2587:              <-- supportsHardlinks 0
20210204-122042 DBG 2630:              --> getFsType /mnt/Backup2
df: /media/nfs: Eingabe-/Ausgabefehler
20210204-122042 DBG 2636:              <-- getFsType nfs
20210204-122042 DBG 5637:          --- Filesystem: nfs
20210204-122042 DBG 2537:              --> supportsFileAttributes /mnt/Backup2
20210204-122042 DBG 2551:              --- ----r-xrwx # nobody # nogroup
20210204-122042 DBG 2557:              --- ----r-xrwx # nobody # nogroup
20210204-122042 DBG 2565:              <-- supportsFileAttributes 0
20210204-122043 DBG 2596:              --> supportsSymlinks /mnt/Backup2
20210204-122043 DBG 2605:              <-- supportsSymlinks 0
```
As you can see in he debuglog hardlinks are supported.

```
rsync: readlink_stat("/media/nfs") failed: Input/output error (5)
IO error encountered -- skipping file deletion
rsync error: some files/attrs were not transferred (see previous errors) (code 23) at main.c(1196) [sender=3.1.2]
```
Looks like user root is not allowed to delete files in the nas box. As far as I can see you have an authorization issue with your nas. if you fix this raspiBackup will be able to create a backup successfully. Unfortunately nfs access configuration sometimes creates and issue. Please check whether you use nfs3, nfs4.1 or nfs 4.2.

framps, 2021-03-11:
Any Update? Otherwise pleased close the issue.

AnduriI, 2021-03-12:
Hey framps,

well I got it working. So exporting a nfs share works with hardlinks for raspiBackup. I was testing this for a little more than a week. I wanted to see if all my RPis can backup to there and see the size of the backups (doing weekly backup).
Only thing that I'm not happy about is I had to set the permissions to 777 for the share, as I had problems writing files. But that's not a problem of raspiBackup but rather me not completely understanding permission system on nfs shares...

closed 2021-03-13 by framps