framps/raspiBackup #303: raspiBackup (tar) only backs up boot partition

SJMarty, 2021-01-27:
Hi Framp,

I have a Pi 4B running Buster with a USB drive as the backup path.  I have the backup type set to tar.  The backup completes successfully but it is only backing up the boot partition.  I have attached the log file.

Thanks.

framps, 2021-01-27:
Hello SJMarty,

this shoudln't happen :cry: 

I need the debug log for further analysis ... but it looks like you missed to attach the log :wink:

SJMarty, 2021-01-27:
[raspiBackup.log](https://github.com/framps/raspiBackup/files/5883360/raspiBackup.log)

SJMarty, 2021-01-27:
Sorry...log is attached now.  :)

framps, 2021-01-28:
Thank you very much for the debug log.

According the debug log a tar file should have been created. Unfortunately there is a debug statement missing which lists the contents of the backup directory to prove this :cry: 
```
20210127-160233 DBG 3019:                      --> colorAnnotation 1
20210127-160233 DBG 3043:                      <-- colorAnnotation 
--- RBK0085I: Backup of type tar started. Please be patient.
20210127-160233 DBG 1474:                      --> executeCmd Command: tar 		-cpi --one-file-system 		 		 		 		-f "/media/pi/sp-usb/@HOSTNAME@/@HOSTNAME@-tar-backup-20210127-160222/@HOSTNAME@-tar-backup-20210127-160222.tar" 		--warning=no-xdev 		--numeric-owner 		--exclude="/*" 		--exclude="/media/pi/sp-usb/@HOSTNAME@/@HOSTNAME@-tar-backup-20210127-160222/raspiBackup.log" 		--exclude="/media/pi/sp-usb/@HOSTNAME@/@HOSTNAME@-tar-backup-20210127-160222/raspiBackup.msg" 		--exclude='.gvfs' 		--exclude=/proc/* 		--exclude=/lost+found/* 		--exclude=/sys/* 		--exclude=/dev/* 		--exclude=/tmp/* 		--exclude=/boot/* 		--exclude=/run/* 		--exclude=/var/cache/* 		 		/
20210127-160233 DBG 1475:                      --- Redirect: & - Skips: 
20210127-160233 DBG 1494:                      <-- executeCmd 0
20210127-160233 DBG 4193:                  <-- tarBackup 0
```

Woud you please execute following command `sudo ls -la /media/pi/sp-usb/@HOSTNAME@/@HOSTNAME@-tar-backup-20210127-160222/@HOSTNAME@-tar-backup-20210127-160222.tar` and show the result? Please update the masqueraded parts of the directory accordingly :wink:

SJMarty, 2021-01-28:
It appears there is a tar file present but it's only 10 KB.

`-rw-r--r-- 1 root root 10240 Jan 27 16:02 /media/pi/sp-usb/HOSTNAME/HOSTNAME-tar-backup-20210127-160222/HOSTNAME-tar-backup-20210127-160222.tar`

framps, 2021-01-28:
I found the root cause for the issue. 

```
20210127-160233 DBG 1474:                      --> executeCmd Command: tar 		-cpi --one-file-system 		 		 		 		-f "/media/pi/sp-usb/@HOSTNAME@/@HOSTNAME@-tar-backup-20210127-160222/@HOSTNAME@-tar-backup-20210127-160222.tar" 		--warning=no-xdev 		--numeric-owner 		--exclude="/*" 		--exclude="/media/pi/sp-usb/@HOSTNAME@/@HOSTNAME@-tar-backup-20210127-160222/raspiBackup.log" 		--exclude="/media/pi/sp-usb/@HOSTNAME@/@HOSTNAME@-tar-backup-20210127-160222/raspiBackup.msg" 		--exclude='.gvfs' 		--exclude=/proc/* 		--exclude=/lost+found/* 		--exclude=/sys/* 		--exclude=/dev/* 		--exclude=/tmp/* 		--exclude=/boot/* 		--exclude=/run/* 		
--exclude=/var/cache/* 		 		/
```

`--exclude="/*"` is wrong and causes everything to be excluded from the backup. 3 days ago I fixed a minor issue but for some reasons one important line in the code was deleted which causes the invalid exclude statement. I added the deleted line again and also the additional debug line I mentioned above.

Please use `sudo raspiBackup.sh -U -S` to grab the latest raspiBackup code which includes the fix, verify the fix and close the issue.

Thank you very much you created the issue ... it's a serious bug.

SJMarty, 2021-01-28:
That did the trick.  Thanks for a very quick turn around on identifying and fixing the issue.

closed 2021-01-28 by SJMarty

framps, 2021-01-28:
> That did the trick. Thanks for a very quick turn around on identifying and fixing the issue.

As I wrote - it's a serious bug which had to be fixed asap and I'm glad you detected and reported the issue.