framps/raspiBackup #304: berryboot on iscsi - backup returns error RBK0001E

owl770, 2021-01-28:
I am running berryboot with iscsi as the destination for RPiOS.

raspiBackup fails with:
RBK0001E: Unexpected program error occured. (a9c8b83), Linenumber: 5296, Error: Unable to find root mountpoint in /proc/cmdline.

cat /proc/cmdline contains:
coherent_pool=1M 8250.nr_uarts=0 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 video=HDMI-A-1:1920x1080M@60,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  elevator=deadline quiet bootmenutimeout=10 datadev=UUID=a2139f94-ab75-4ed7-8ddf-f9f25345dad2 datadev=mmcblk0p2 quiet splash plymouth.ignore-serial-consoles

Is there a way I can still do a raspiBackup on a RPi running berryboot on iscsi?

framps, 2021-01-28:
raspiBackup right now scans for either `imgpart=....` or `root=...` to find the root partition. I don't know berryboot but given your /proc/cmdline I assume  `datadev=...`is the definition of the root partition. If this is correct

try to change
`	if [[ $cmdline =~ .*(imgpart|root)=([^ ]+) ]]; then`
into
`	if [[ $cmdline =~ .*(imgpart|root|datadev)=([^ ]+) ]]; then`

Frankly I'm surprised to see two datadev definitions :confused: . Are you sure both are required?

Please note: berryboot is not officially supported by raspiBackup. But let's try to find the root cause for the issue. If minor changes in raspiBackup are required in order to fix the issue I'll do this. But major changes will not be done :wink:

owl770, 2021-01-29:
Thanks for the prompt reply.
Yes, making that change fixes the issue. The backup completes successfully - a dd backup of /dev/mmcblk0p1 and a compressed tar backup of / (root).
I don't know why berryboot has inserted two datadev parameters.

framps, 2021-01-29:
Next release of raspiBackup will include the update for berryboot. please close the issue.

owl770, 2021-01-29:
Thanks for your help and a great product.

closed 2021-01-29 by owl770

fockr, 2021-10-03:
Hey framps,

I encountered a similar problem on my Raspberry Pi 3b+ with Freedombox installed.

```bash
--- RBK0009I: freedombox: raspiBackup.sh V0.6.6 (262862c) So 3. Okt 17:24:53 UTC 2021 gestartet.
--- RBK0128I: Logdatei ist /media/root/USB-STICK/freedombox/freedombox-dd-backup-20211003-172451/raspiBackup.log.
--- RBK0116I: Konfigurationsdatei /usr/local/etc/raspiBackup.conf wird benutzt.
??? RBK0001E: Unerwarteter Programmfehler trat auf. (262862c), Zeile: 5322, Fehler: Unable to find root mountpoint in /proc/cmdline.
??? RBK0148E: @@@@@@@@@@@@@@@@@@@@ Stacktrace @@@@@@@@@@@@@@@@@@@@
  File "/usr/local/bin/raspiBackup.sh", line 8063, in main
    doit
  File "/usr/local/bin/raspiBackup.sh", line 5064, in doit
    		doitBackup
  File "/usr/local/bin/raspiBackup.sh", line 5561, in doitBackup
    	getRootPartition
  File "/usr/local/bin/raspiBackup.sh", line 5322, in getRootPartition
    		assertionFailed $LINENO "Unable to find root mountpoint in /proc/cmdline"
  File "/usr/local/bin/raspiBackup.sh", line 1433, in assertionFailed
    	logStack
--- RBK0033I: Bitte warten bis aufgeräumt wurde.
--- RBK0043I: Unvollständiges Backup /media/root/USB-STICK/freedombox/freedombox-dd-backup-20211003-172451 in wird gelöscht. Das kann etwas dauern. Bitte Geduld.
--- RBK0049I: Meldungen wurden in /home/jonny/raspiBackup.msg gesichert.
--- RBK0026I: Debug Logdatei wurde in /home/jonny/raspiBackup.log gesichert.
--- RBK0010I: freedombox: raspiBackup.sh V0.6.6 (262862c) So 3. Okt 17:24:57 UTC 2021 beendet mit Returncode 101.
??? RBK0005E: Backup fehlerhaft beendet. Siehe vorhergehende Fehlermeldungen.
```
The `/proc/cmdline` is empty except the word `quiet`.

I'm not sure if raspiBackup supports Freedombox. I thought I'd give it a try since it is basically a Debian distrbution with a few enhancements.

Thanks in advance and thank you for your work,
Jon

framps, 2021-10-03:
> ```
> "Unable to find root mountpoint in /proc/cmdline"
> ```
raspiBackup expects `root=<rootpartition>` in /boot/cmdline.txt.

> I'm not sure if raspiBackup supports Freedombox. I thought I'd give it a try

raspiBackup is supported for RaspberryOS only. I know of raspiBackup users running other distros. But your distro misses a required information and thus cannot use raspiBackup - unfortunately.

fockr, 2021-10-04:
A pitty... I thought my search for a hassle-free live backup solution came to an end. Thanks anyways!