framps/raspiBackup #306: Question About rsync Backup Options

SJMarty, 2021-01-28:
When running an rsync backup to an NFS share on my Synology NAS, I received the error message...

`??? RBK0024E: Backup tool rsync received an error
rsync: set_acl: sys_acl_set_file(media/pi, ACL_TYPE_ACCESS): Operation not supported (95)`

In the FAQ, one of the options to fix this issue states...

Define DEFAULT_RSYNC_BACKUP_OPTIONS="-aHx --delete" (remove A from the default) which will disable rsync to save ACLs.

However, in the User's Guide, it lists the default option for this setting as...

DEFAULT_RSYNC_BACKUP_OPTIONS="**--delete** -aHA**X**x"

There is an additional "X" present and the "--delete" is at the beginning vs. the end.

For clarification, to disable rsync/ACLs to avoid the RBK0024E error message, what should I set the DEFAULT_RSYNC_BACKUP_OPTIONS value to?

framps, 2021-01-28:
You're getting familiar with github.I like this :+1: And you are one of the raspiBackup users which read the doc carefully :+1: .

All these flags are rsync flags and you can lookup them with `man rsync`. The order of theses options really doesn't matter. Key point is to remove 'A' . 'X' was included in former times You need this option in special environments and I decided to remove it. I will update the User's guide page.

SJMarty, 2021-01-28:
Great.  Thank you!

closed 2021-01-28 by SJMarty